Koenkk/zigbee2mqtt #24900: [New device support]: I have tried to set this up using the guide with no luck

jamiegalbreath, 2024-11-24:
### Link

https://www.amazon.co.uk/dp/B0DG8PJC5G?ref=ppx_yo2ov_dt_b_fed_asin_title

### Database entry

{"id":2,"type":"EndDevice","ieeeAddr":"0xa4c138ce596d0846","nwkAddr":57126,"manufId":4417,"manufName":"_TZ3000_402vrq2i","powerSource":"Battery","modelId":"TS004F","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":260,"inClusterList":[1,3,4,6,4096,0],"outClusterList":[3,4,5,6,8,4096,25,10],"clusters":{"genBasic":{"attributes":{"stackVersion":0,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":66,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1732451631190}

### Zigbee2MQTT version

 1.41.0 commit: unknown

### Comments

I have tried to set this up using the guide with no luck. Help would be greatly appreciated.

### External definition

```shell
const {battery, onOff, commandsOnOff, commandsLevelCtrl} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['TS004F'],
    model: 'TS004F',
    vendor: '_TZ3000_402vrq2i',
    description: 'Automatically generated definition',
    extend: [battery(), onOff({"powerOnBehavior": false}), commandsOnOff(), commandsLevelCtrl()],
    meta: {},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

Does not show any triggers in setting up an automation

codefaux, 2024-11-25:
I should add, after creating the .js file and adding it to configuration.yaml I get the following error on any action performed upon the device aka twist, push, etc:

```
[2024-11-25 14:54:33] error:    z2m: Failed to configure 'ZigDial2', attempt 1 (Error: ZCL command 0xa4c138b4b46ef09f/1 genOnOff.configReport([{"minimumReportInterval":0,"maximumReportInterval":65000,"reportableChange":1,"attribute":"onOff"}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
    at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:350:28)
    at Endpoint.zclCommand (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:956:26)
    at processTicksAndRejections (node:internal/process/task_queues:95:5)
    at Endpoint.configureReporting (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:688:9)
    at setupAttributes (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:162:17)
    at /app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:546:17
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/index.ts:205:21)
    at Configure.configure (/app/lib/extension/configure.ts:130:13)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:206:17))
```

DaeMonSxy, 2024-11-29:
Hi gents,

just got mine also, would be good to have it integrated, isnt it?:)

codefaux, 2024-12-01:
After fiddling around and searching more thoroughly, I've gotten mine working. The answer is in another pending issue;

https://github.com/Koenkk/zigbee2mqtt/issues/23543#issuecomment-2508257550

I'm still not sure how to get to that point with the tutorial, but it's working.

It does support three-tap to switch from Command to Event mode.

In Event mode, it supports Single tap, Double tap (don't tap too fast), Hold (4sec), Rotate Left and Rotate Right

In command mode, it supports Single, no Double, no Hold, brightness_step_up/_down and distance, color_temperature_step_up/_down and distance.

djurny, 2024-12-02:
[1.42.0](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.42.0) supports the [Smart Knob Dimmer](https://www.aliexpress.com/item/3256807534367914.html).
From the release notes:

- [#8257](https://github.com/Koenkk/zigbee-herdsman-converters/issues/8257) Detect _TZ3000_402vrq2i as Tuya ERS-10TZBVK-AA (@Velomeloman)

Groetjes,

codefaux, 2024-12-02:
As this release just pushed 14 hours ago, I'll go ahead and say they had support for our device already in the pipeline. Nice to see it released, after @jamiegalbreath verifies support they can probably close this Issue.

Can someone weigh in on best practices moving forward? Should we nuke the external definition file after moving to this version? Is the external definition from the Issue I linked essentially the same as the internal definition will be?

If it is best to remove the external definition, will this impact Home Assistant devices?

djurny, 2024-12-02:
The external converter was made by copy/pasting parts from what is now available in [1.42.0](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.42.0). I would remove the external definition and just use what is in [1.42.0](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.42.0).
I did not see anything bork after doing so.
Groetjes,

Velomeloman, 2024-12-02:
I also made an automatic converter. After it didn't work, I found a similar device in herdsman converters and copied the code into my js. After checking the converter's operability, I made a pull request to detect this dimmer.
[white_knob.js.txt](https://github.com/user-attachments/files/17978768/white_knob.js.txt)

chupacabra71, 2024-12-05:
i purchased a similar device from [aliexpress](https://www.aliexpress.us/item/3256805640427007.html?spm=a2g0o.productlist.main.27.7978SbGySbGyoO&algo_pvid=686018c4-6190-4bfe-9e4d-826151656cd3&aem_p4p_detail=2024120505431615949096285037280002409988&algo_exp_id=686018c4-6190-4bfe-9e4d-826151656cd3-13&pdp_npi=4%40dis%21USD%2117.13%2111.99%21%21%2117.13%2111.99%21%402103244b17334061966973691e73ea%2112000034491954011%21sea%21US%213349649559%21X&curPageLogUid=bDe2Ai9wyDJU&utparam-url=scene%3Asearch%7Cquery_from%3A&search_p4p_id=2024120505431615949096285037280002409988_4#nav-specification) and i am on 1.42.0, but i cannot get the device to get any commands.  The device documentation shows the model as `RSH-SC20`.  when i look at what the device exposes it shows all as NULL.  the definition shows as:
```
const {battery, onOff, commandsOnOff, commandsLevelCtrl} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['TS004F'],
    model: 'TS004F',
    vendor: '_TZ3000_abrsvsou',
    description: 'Automatically generated definition',
    extend: [battery(), onOff({"powerOnBehavior":false}), commandsOnOff(), commandsLevelCtrl()],
    meta: {},
};

module.exports = definition;
```
and this is what the exposes shows up as:

[image]

I am a bit new to Z2M, so any pointers would be greatly appreciated.

codefaux, 2024-12-11:
> but i cannot get the device to get any commands

Your setup may be working properly here.

When these dial devices update, they send multiple, nearly-simultaneous updates.
When in Command mode, this will normally be a step size, an action, a null action state, and a null step size -- within milliseconds of each other. In the UI, you -might- get to see text flicker between N/A and brightness_step_up but most of the time it won't show at all. Plug the device into HASS or monitor your MQTT topic to see if this is the case for you. It shows updates via Device panel in HASS;
[image]

But the Zigbee2MQTT dash shows N/A the whole time;
[image]

When in Event mode they drop the step_size feature and the push-and-hold-and-turn feature but gain the capacity to measure a double-click. I use Command mode, so I can capture step size.

Also worth noting, I've found that these report step size in multiples of 12.5, rounded up. AKA -- 13, 25, 38, 50, 63, etc... So, in Home Assistant, you can use an automation to capture step_size to an input_number any time its state changes from Unavailable, divide it by 12.5, and filter that to an int. You'll also need an automation to capture when the Action changes, and so on.

chupacabra71, 2024-12-11:
> Also worth noting, I've found that these report step size in multiples of 12.5, rounded up. AKA -- 13, 25, 38, 50, 63, etc... So, in Home Assistant, you can use an automation to capture step_size to [...]

Thank you!  just checked via MQTT Explorer and they are indeed sending messages.  very confusing, but they seem to be working as you mentioned.

codefaux, 2024-12-11:
> Thank you!

Quite welcome! I'm glad I could help. If you need help working with this sort of event in Home Automation I'm pretty familiar and willing to help.

> very confusing, but they seem to be working as you mentioned.

It seems confusing at first, but consider -- how would a service know that an event had already been handled? How would it differentiate between the same event being sent twice in a row? By sending an event and returning to 'null' it ensures subsequent triggers will be able to fire regardless of condition. It does make catching things a bit tricky; if you try to read the value after the trigger, it's probably already gone, so you have to capture it with the state event itself.

FWIW I have two Home Assistant automations related to these dials -- one triggered by the step_size event and store it in an input_number, one to catch the action trigger (aka brightness_up or brightness_down) and apply the caught step_size to the light.

djurny, 2024-12-12:
Hi there,
I found a great example online somewhere of a "peak-hold" type of sensor. This will allow your automation to use the  `action_step_size` properly. You can then have your automation trigger by `rotate_left`, `brightness_step_down`, `rotate_right`,`brightness_step_up`, `toggle` or `single`. (Or any of the other triggers sent by the device.)
```
# in configuration.yaml:
template:
- trigger:
    - platform: state
      entity_id: sensor.<name>_action_step_size
      not_to:
        - unknown
        - unavailable
    sensor:
    - name: "<name>'s action step size"
      state: '{{trigger.to_state.state}}'
```
Most likely you'd be able to filter out the values `unknown` or `unavailable` in z2m as well to achieve the same effect.

Hope that helps,
Groetjes,

codefaux, 2024-12-12:
Oh - damn, sorry, I had put that into my first message but deleted it to reformat and forgot to re-add it. Good addition. However, you'll want to modify it with the division I mentioned if you want it more easily used. I also added a max() call to ensure it's not set to zero.

I notice you're using template, but I used an automation to store it in a number helper object. I'm curious what the effective difference is, and how one would reference your version.

```
alias: "Kitchen Lighting: (User) Step Size tracking from ZigDial1"
description: ""
triggers:
  - trigger: state
    entity_id:
      - sensor.zigdial1_action_step_size ###EDIT FOR YOURS
    from: unknown
conditions: []
actions:
  - action: input_number.set_value
    metadata: {}
    data:
      value: "{{ max(trigger.to_state.state | float / 12.5, 1) | int }}"
    target:
      entity_id: input_number.kitchen_lighting_brightness_step_size
mode: single
```

djurny, 2024-12-12:
Hi @codefaux,
Functionally both will do the job, but I would think the template sensor will be more compact and will not require an additional 'helper' variable.
Combined it will look as follows for your application:
```
# in configuration.yaml:
template:
- trigger:
    - platform: state
      entity_id: sensor.<name>_action_step_size
      not_to:
        - unknown
        - unavailable
    sensor:
    - name: "<name>'s action step size"
      state: '{{[1, trigger.to_state.state|float / 12.5]|max|int}}'
```
In my automation I use the templates sensor as follows:
```
alias: Smart Knob Dimmer
description: "Yes"
triggers:
  - domain: mqtt
    device_id: ***
    type: action
    subtype: rotate_left
    trigger: device
    id: left
  - domain: mqtt
    device_id:  ***
    type: action
    subtype: brightness_step_down
    trigger: device
    id: left
  - domain: mqtt
    device_id:  ***
    type: action
    subtype: rotate_right
    trigger: device
    id: right
  - domain: mqtt
    device_id:  ***
    type: action
    subtype: brightness_step_up
    trigger: device
    id: right
  - domain: mqtt
    device_id:  ***
    type: action
    subtype: toggle
    trigger: device
    id: press
  - domain: mqtt
    device_id:  ***
    type: action
    subtype: single
    trigger: device
    id: press
conditions: []
actions:
  - choose:
      - conditions:
          - condition: trigger
            id:
              - left
        sequence:
          - action: light.turn_on
            target:
              entity_id: light.work_room
            data:
              brightness_step_pct: >-
                {%set _now=now()%}
                {%set _last_updated=states['sensor.dimmer_work_room_accel'].last_updated%}
                {%set _elapsed=_now-_last_updated%}
                {%if _elapsed.total_seconds() < 1.5%}
                  {%set _accel=states('sensor.dimmer_work_room_accel')|int%}
                  {%if _accel is number%}
                    -{{_accel|float*100/255}}
                  {%else%}
                    -10
                  {%endif%}
                {%else%}
                  -10
                {%endif%}
      - conditions:
          - condition: trigger
            id:
              - right
        sequence:
          - action: light.turn_on
            target:
              entity_id: light.work_room
            data:
              brightness_step_pct: >-
                {%set _now=now()%}
                {%set _last_updated=states['sensor.dimmer_work_room_accel'].last_updated%}
                {%set _elapsed=_now-_last_updated%}
                {%if _elapsed.total_seconds() < 1.5%}
                  {%set _accel=states('sensor.dimmer_work_room_accel')|int%}
                  {%if _accel is number%}
                    {{_accel|float*100/255}}
                  {%else%}
                    10
                  {%endif%}
                {%else%}
                  10
                {%endif%}
      - conditions:
          - condition: trigger
            id:
              - press
        sequence:
          - action: adaptive_lighting.apply
            data:
              entity_id: switch.adaptive_lighting_work_room
              adapt_brightness: true
              adapt_color: true
            enabled: false
          - action: adaptive_lighting.set_manual_control
            data:
              entity_id: switch.adaptive_lighting_work_room
              manual_control: false
mode: single
```
Some remarks though:
- The example uses a different calculation for the `action_step_size` - the 'more' you jog the dial, the bigger the adjustment.
- The automation supports both operation modes, so if someone accidentally triple-clicks the button, it still sort-of works.
- Added a check to make sure that the template sensor value is valid. After some testing, I found that the events do not always come in in order. I prefer to have the automation ignore the `action_step_size` in case the last value was sent later on or way earlier than the rotation/click events came in.
- There is room for further improvement by moving the timestamp checking in an automation variable, so I might start reworking it at some point in time. For now it's working adequately for my usecase (dimming my work room lighting with a fancy round dimmer know on my desk). 

Groetjes,

IamFloFlo, 2024-12-17:
Hello everyone!

Before I start: I am also learning everything that is possible with Zigbee2MQTT and MQTT. 
I saw this and thought it would be perfect for a spotify controller! Play, pause, skip song, previous song etc. 

Unfortunately when I tried adding the knob it didn't work. I get the following message: 

[image]

Any idea on how to fix this? I read above that it was supported but it doesn't seem to be the case.

I am on Zigbee2MQTT Firmware: 1.40.2.

Thanks in advance for any help :)

chupacabra71, 2024-12-17:
> Any idea on how to fix this? I read above that it was supported but it doesn't seem to be the case.
> 
> I am on Zigbee2MQTT Firmware: 1.40.2.
> 
> Thanks in advance for any help :)

yeah, they are a bit finicky.  Remove the device and re-add it, being close to the coordinator helps.  When adding it, keep pressing the button every second until it is recognized.  If you don't press the button, it seems to time out during the interview.

IamFloFlo, 2024-12-17:
> yeah, they are a bit finicky. Remove the device and re-add it, being close to the coordinator helps. When adding it, keep pressing the button every second until it is recognized. If you don't press [...]

first of all, thank you for the quick reply! I tried adding it again, around 10 times, while pressing the button and turning it. Unfortunately it is really quick to be found and it still says unsupported. I reset it and pulled out the battery but unfortunately it didn't seem to work. I will try it again tomorrow

djurny, 2024-12-19:
Hi @IamFloFlo,
Zigbee2MQTT v1.42.0 should have support for your device. Looking at the Zigbee model/manufacturer codes, it seems you have the exact same device as I got. 
Groetjes,

buz-ch, 2025-02-21:
I have mine in event mode but neither rotate directions ever show any more than rotate_  - no action step, no quasi simultaneous other events - neither in the z2m log nor in MQTT Explorer. 

Log only shows (for press and then rotate to show two)
```
info 2025-02-21 13:55:04z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Smart-Knob-2', payload '{"action":"single","action_rate":null,"action_step_size":null,"action_transition_time":null,"battery":100,"linkquality":172,"operation_mode":"event","voltage":3000}'
info 2025-02-21 13:55:04z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Smart-Knob-2/action', payload 'single'
info 2025-02-21 13:55:09z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Smart-Knob-2', payload '{"action":"rotate_right","action_rate":null,"action_step_size":null,"action_transition_time":null,"battery":100,"linkquality":108,"operation_mode":"event","voltage":3000}'
info 2025-02-21 13:55:09z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Smart-Knob-2/action', payload 'rotate_right'
```
And that is it. Like this it seems kind of pointless to have a dial. What am I missing?

codefaux, 2025-02-26:
I decided to check my own knobs to attempt to answer @buz-ch but it turns out mine aren't working either. I have multiple dials, and they all behave the same.

@buz-ch -- "in the log" -- which log? Try to avoid statements which assume people know what you're talking about. Also I agree, without step size the dial is hardly useful for dimming lights. It could control locks, window shutters, power on/off, or a myriad other things with just rotate left/right.

More generally though;

My dials have all stopped reporting Action to the Home Assistant integration. In Command mode, they send ONLY step size changes when turned; they do not report Action at all. When tapped, in Command mode, they clear the Step Size value and set it Unknown, but do not send a 'toggle' Action. In Event mode, they send nothing.

[image]

[image]

The dials, when deleted from Home Assistant and re-added via restarting Zigbee2Mqtt, still do not offer Action as a Sensor, and still behave the same.

[image]

Directly within the Zigbee2Mqtt webUI, the dials DO properly show and update Action, Action step size, Action transition time, etc the entire time while in Command mode, and only populate Action with rotate_right/_left, single, double in event mode. (As expected.)  Previous behavior of instantly dropping state to Null is gone (unexpected but good) and other functions seem proper.

[image]

I have tried the Interview and Reconfigure options, nothing seems to change.

This integration needs attention.

buz-ch, 2025-02-26:
In the Z2M log. In event mode, mine work (like others):
```
info 2025-02-26 20:04:55z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Smart-Knob-1', payload '{"action":"single","action_rate":null,"action_step_size":null,"action_transition_time":null,"battery":82,"linkquality":156,"operation_mode":"event","voltage":2900}'
info 2025-02-26 20:04:55z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Smart-Knob-1/action', payload 'single'
info 2025-02-26 20:05:01z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Smart-Knob-1', payload '{"action":"rotate_left","action_rate":null,"action_step_size":null,"action_transition_time":null,"battery":82,"linkquality":156,"operation_mode":"event","voltage":2900}'
info 2025-02-26 20:05:01z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Smart-Knob-1/action', payload 'rotate_left'
info 2025-02-26 20:05:04z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Smart-Knob-1', payload '{"action":"single","action_rate":null,"action_step_size":null,"action_transition_time":null,"battery":82,"linkquality":156,"operation_mode":"event","voltage":2900}'
info 2025-02-26 20:05:04z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Smart-Knob-1/action', payload 'single'
info 2025-02-26 20:05:04z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Smart-Knob-1', payload '{"action":"single","action_rate":null,"action_step_size":null,"action_transition_time":null,"battery":82,"linkquality":156,"operation_mode":"event","voltage":2900}'
info 2025-02-26 20:05:08z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Smart-Knob-1/action', payload 'double'
info 2025-02-26 20:05:30z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Smart-Knob-1', payload '{"action":"rotate_right","action_rate":null,"action_step_size":null,"action_transition_time":null,"battery":82,"linkquality":156,"operation_mode":"event","voltage":2900}'
info 2025-02-26 20:05:30z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Smart-Knob-1/action', payload 'rotate_right'
info 2025-02-26 20:05:35z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Smart-Knob-1', payload 
```
In command mode I do not get any Z2M log entries at all (aside of the initial one confirming they switched to command mode) . I do care too much what HA does with that, I can always just use mqtt triggers (usually do so anyhow) but without any data going to mqtt, that will not work, either. 

They also seem to run through battery quickly, since my post above I have not used them and they dropped from 100 to 82... Maybe time to recycle them, really.

codefaux, 2025-02-26:
@buz-ch It seems we're having different issues with the same integration. I've checked my Z2M logs, and I do show complete messages when I turn the knob;

[image]

My issue appears to be the Home Assistant integration's detection isn't keeping the Action part of the message, and I'm pretty sure I didn't manually touch anything there.

[image]

It seems several very different vendors use almost the same identification values. I have three knobs which are vastly different in design, reliability, and battery life. Ironically the most expensive one is the one which works the least reliably. The less-than-$5 knob from Aliexpress is actually the best.

Regarding battery life versus use -- the device is alive the entire time it has a battery. The device regularly handshakes with its coordinator to exchange messages. Occasionally using a device like a button or knob only increases overall power use by a small fraction. If you want something 100% dormant until it is triggered, I suggest picking up some 433mhz hardware; that's the newest it gets for no-battery idle.

buz-ch, 2025-02-26:
I am aware they handshake but they have CR2450 for chrissakes,  Hue Dimmers live forever on puny CR2032... I suspect I have a different device then you. Altho mine are of the 5 USD AE variety, go figure.

onlyann, 2025-04-16:
@codefaux I am running in the exact same issue than you with the Hue Dial Switch integration.

The `action` shows in Z2M when I view the web interface but no action sensor was created in HA Core.

Did you find a solution?

codefaux, 2025-04-16:
@onlyann Nope -- still broken. I pulled two of them to test various things just now, can't get the HASS side to work without manual stuff probably, and I'm just not motivated -- they didn't work reliably enough for me in the first place, lol. Broke shortly after I added it, and it has never worked since. They're collecting dust, complaining about low battery. I went ahead and pulled the batteries now lol. Glad I only paid like $5 each from Ali. Sorry I can't help, but I'm listening if anyone figures it out.

onlyann, 2025-04-16:
@codefaux I eventually discovered that Z2M 2.0 release removed all action sensors and replaced them with Automation Triggers.

I found out in the release notes.
They mention you can re-enable those sensors by enabling an option. Not that I tried since triggers are good enough for me combined with a helper to store the last pressed action.

codefaux, 2025-04-16:
@onlyann Thank you, I'll look into that if I can find it. I've never used Automation Triggers before, guess it's time to learn.

Some apps pop a changelog for you, especially for breaking changes. Others break things and mention it in the notes. I guess that clarifies how careful I'll have to be with handling z2m.

closed 2025-08-17 by github-actions[bot]