Koenkk/zigbee2mqtt #24905: Sonoff trvzb: valve opening percentage

vindaalex, 2024-11-24:
### What happened?

The new version 1.2.1 seems to allow direct control of the valve [changelog 1.2.1](https://forum.ewelink.cc/t/trvzb-firmware-changelog/74377). Would it be possible to add this feature also to z2m?

z2m: 1.41.0-1
ZBDongle-E: 7.4.3
Sonoff trvzb: 1.2.1

Background: I use a thermostat controller within Home Assistant which directly controls the valve positions. Currently using Eurotronic trv's but these are really noisy and loose valve calibration after a while. I have two sonoff's whick I would like to use as replacements.

### What did you expect to happen?

Current changelog to 1.2.1 shows new option to control valve position

### Zigbee2MQTT version

1.41.0-1

### Adapter firmware version

7.4.3

### Adapter

zbdongle-e ember

### Setup

add-on in HA

Vlada-M9o, 2024-12-08:
Hello, TRVZ is a good valve price/quality ratio. However, the ability to control the valve position in percentages or degrees is very important and lacks functionality. For example, a temperature change of 1 degree down, opening a valve by 5%. The ON and OFF function is a bad solution.

vindaalex, 2024-12-08:
I think I've found a workaround by controlling `<entity_id>_valve_closing_degree` and `<entity_id>_valve_opening_degree` together via a Hoe Assistant automation to the required opening percentage. You could for instance use a [Home Assistant PID thermostat](https://community.home-assistant.io/t/thermostat-with-pid-controller/55834) or more complex [multi-room](https://community.home-assistant.io/t/multizone-thermostat-incl-various-control-options/308898) to calculate the opening percentage.

```
- alias: set_sl4_trv
  id: set_sl4_trv
  trigger:
    - platform: state
      entity_id: input_number.sl4_pid
  action:
    - service: number.set_value
      data:
        value: "{{100 - states('input_number.sl4_pid')|int}}"
      target:
        entity_id:
          - number.sl4_trv_valve_closing_degree
    - service: number.set_value
      data:
        value: "{{states('input_number.sl4_pid')|int}}"
      target:
        entity_id:
          - number.sl4_trv_valve_opening_degree
```

closed 2025-02-15 by github-actions[bot]