Koenkk/zigbee2mqtt #24908: Operation of multiple ROBB (ROB_200-008-0) zigbee wall switches fails

sbruckmueller, 2024-11-24:
### What happened?

When setting up a Zigbee network with Zigbee2MQTT (addon in  homeassistant), only a limited number of wall switches (ROBB ROB_200-008-0) can be used. After twelve buttons have been configured, the following appears during configurring button number 13 ERROR:  zh:ember: Failed to register group '35105' in multicast table with status=INVALID_STATE. 
Each additional button produces a similar error , but with different group number.

### What did you expect to happen?

I expexcted to get back a zigbee-message with valid payload  e.g. '"action":"on_1".  

### How to reproduce it (minimal and precise)

zigbee2mqtt config: port: >-
  /dev/serial/by-id/usb-Itead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_V2_30ebbe8738d9ee119522b54c37b89984-if00-port0
adapter: ember
rtscts: false

### Zigbee2MQTT version

1.41.0

### Adapter firmware version

7.4.1.0 , 7.4.3.0, 7.4.4.0

### Adapter

ZBDongle-E 3.0 USB Dongle Plus,EFR32MG21 

### Setup

PI 5B homeassistant on sd-card

sbruckmueller, 2024-11-27:
This solved my problem: Flashing ZBDongle-E with firmware https://github.com/Nerivec/silabs-firmware-builder/releases/download/v2024.6.2/sonoff_zbdonglee_ncp-uart-sw_115200_8.0.2.0.gbl

closed 2024-11-27 by sbruckmueller

Bram-V, 2025-01-04:
I have the exact same problem. Only with a SkyConnect_v1.0, using Zigbee2MQTT 2.0.0. 

I have two ROBB ROB_200-008-0 wall switches. It seems that after a restart of Z2M, I can choose which 4 pairs (On and Off) of buttons I want to use, because the first 4 pairs clicked will work. And the first time I do it, I need to click a button twice for it to really show the data. 

Adapter: SkyConnect v1.0 from Nabu Casa
Firmware version: sdk_version='4.4.4', ezsp_version='7.4.4.0'
Zigbee-herdsman version: 3.2.1

```
[2025-01-04 20:48:03] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/SchakelaarGang', payload '{"action":"brightness_stop_1","action_group":63425,"battery":100,"linkquality":104}'
[2025-01-04 20:48:03] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/SchakelaarGang/action', payload 'brightness_stop_1'
[2025-01-04 20:48:23] error: 	zh:ember: Failed to register group '48017' in multicast table with status=INVALID_STATE.
```

CerielRoland, 2025-07-31:
Got a solution for this Bram-V?