Koenkk/zigbee2mqtt #24912: [New device support]: ??? TS0601 (_TZE200_kccdzaeo) not supported

Chipatusik, 2024-11-25:
### Link

https://aliexpress.ru/item/1005008046609585.html

### Database entry

{"id":27,"type":"EndDevice","ieeeAddr":"0xa4c138fb37411e11","nwkAddr":22376,"manufId":4742,"manufName":"_TZE200_kccdzaeo","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,6,1280,10],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"TS0601","manufacturerName":"_TZE200_kccdzaeo","powerSource":1,"zclVersion":3,"appVersion":145,"stackVersion":2,"hwVersion":1,"swBuildId":"0122052017"}},"ssIasZone":{"attributes":{"iasCieAddr":"0x84b4dbfffec7affb","zoneState":0}},"genOnOff":{"attributes":{"onOff":0,"moesStartUpOnOff":2,"tuyaBacklightSwitch":1}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x84b4dbfffec7affb","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0}],"meta":{}}},"appVersion":145,"stackVersion":2,"hwVersion":1,"swBuildId":"0122052017","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1466202208},"lastSeen":1732505351156}

### Zigbee2MQTT version

1.41.0-1

### Comments

_TZE200_kccdzaeo
[image]

### External definition

```shell
const { } = require("zigbee-herdsman-converters/lib/modernExtend");
const exposes = require("zigbee-herdsman-converters/lib/exposes");
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    fingerprint: tuya.fingerprint('TS0601', ['_TZE200_kccdzaeo']),
    model: 'Haozee-HZ',
    vendor: 'TuYa',
    description: 'Haozee switch 1 gang with 10G human presence sensor',
    configure: tuya.configureMagicPacket,
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    exposes: [
        e.presence(),
        e.numeric('radar_sensitivity', ea.STATE_SET).withValueMin(0).withValueMax(19).withValueStep(1).withDescription('Sensitivity of the radar'),
        e.numeric('detection_range', ea.STATE_SET).withValueMin(0).withValueMax(1000).withValueStep(1).withDescription('Detection distance').withUnit('cm'),
        e.switch().setAccess('state', ea.STATE_SET),
        //tuya.exposes.countdown(),
        //e.enum('all_on', ea.STATE_SET, ['ON']).withDescription('All On'),
        //e.binary('switch_reversal', ea.STATE_SET, 'off', 'on').withDescription('Switch reversal'),
        //e.enum('all_off', ea.STATE_SET, ['ON']).withDescription('All OFF'),
        //e.text('stay_sensitivity', ea.STATE_SET).withDescription('Stay ensitivity'),
        e.enum('backlight_switch', ea.STATE_SET, ['off', 'on']).withDescription('Backlight Switch'),
        e.enum('relay_status', ea.STATE_SET, ['off', 'on', 'previous']).withDescription('Power-on behavior'),
        e.enum('induction_lighting', ea.STATE_SET, ['None', 'All channels']).withDescription('Induction lighting'),
        e.numeric('pir_delay', ea.STATE_SET).withPreset('default', 10, 'Default value').withUnit('s').withDescription('Pir Delay'),
        e.enum('induction_lighting_off', ea.STATE_SET, ['None', 'All channels']).withDescription('Induction lighting off'),
    ],
    meta: {
        tuyaDatapoints: [
            [1, 'presence', tuya.valueConverter.trueFalse1],
            [2, 'radar_sensitivity', tuya.valueConverter.raw],
            [4, 'detection_range', tuya.valueConverter.raw],
            [101, 'state', tuya.valueConverter.onOff],
            [102, 'countdown', tuya.valueConverter.countdown],
            [107, 'all_on', tuya.valueConverter.onOff],
            [108, 'switch_reversal', tuya.valueConverterBasic.lookup({ 'off': tuya.enum(0), 'on': tuya.enum(1) })],
            [109, 'all_off', tuya.valueConverter.onOff],
            [110, 'stay_sensitivity', tuya.valueConverter.raw],
            [111, 'backlight_switch', tuya.valueConverterBasic.lookup({ 'off': false, 'on': true })],
            [112, 'relay_status', tuya.valueConverterBasic.lookup({ 'off': tuya.enum(0), 'on': tuya.enum(1), 'previous': tuya.enum(3) })],
            [113, 'induction_lighting', tuya.valueConverterBasic.lookup({ 'None': tuya.enum(0), 'All channels': tuya.enum(1) })],
            [114, 'pir_delay', tuya.valueConverter.countdown],
            [115, 'induction_lighting_off', tuya.valueConverterBasic.lookup({ 'None': tuya.enum(0), 'All channels': tuya.enum(1) })],
        ]
    }
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

stay_sensitivity

CronexSK, 2024-12-13:
I join in

ControlCircuit, 2025-01-06:
also interested

kaushal1976, 2025-01-17:
Great. At the time of buying I thought this has a lux sensor as well. How does the Tuya app implement motion (presence) light without a lux sensor?

kaushal1976, 2025-01-17:
What is "induction lighting"? Did you mean "indicator light"?

ITMarco, 2025-02-20:
Hi, I bought this to use with my home assistant but the IAS zone is never off. It goes in "off" mode but immediately goes back on without any motion being detected.

kaushal1976, 2025-02-20:
> Hi, I bought this to use with my home assistant but the IAS zone is never off. It goes in "off" mode but immediately goes back on without any motion being detected. I use zigbee mode btw, not zigbee2mqtt and wondering if the behavior is the same for everyone

Hi,

As you have explained, there is a "bouncy effect" in the presence sensor, especially when there is no presence. Adjusting the sensitivity helped to overcome this issue. I set mine to 12, and works OK

ITMarco, 2025-02-20:
Thanks for the quick response. How do I set the sensitivity then?

kaushal1976, 2025-02-20:
The settings are there under "Zigbee2MQTT" section on the HA (If you have used the external definition given above. Please see the screenshot below.

[image]

ITMarco, 2025-02-20:
okay thanks. will have to get the mqtt thing going then. I've used the ZHA support until now (that was what my question was about: If everyone experiences the same thing). thanks again!

kaushal1976, 2025-02-20:
Ah, OK. Sorry, I misread the question.

closed 2025-05-13 by github-actions[bot]