Koenkk/zigbee2mqtt #24916: [New device support]: New Support for Aliexpress Smoke Detector

markro76, 2024-11-25:
### Link

https://de.aliexpress.com/item/1005006862684253.html?spm=a2g0o.order_list.order_list_main.48.21ef5c5fXBHDNu&gatewayAdapt=glo2deu

### Database entry

{"id":41,"type":"EndDevice","ieeeAddr":"0xa4c138ffd7a8e9b2","nwkAddr":47773,"manufId":4417,"manufName":"_TZE284_rccxox8p","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0,60672],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"stackVersion":0,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":77,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1732536980570}

### Zigbee2MQTT version

1.41.0-1

### Comments

Please Help

Device '0xa4c138ffd7a8e9b2' with Zigbee model 'TS0601' and manufacturer name '_TZE284_rccxox8p' is NOT supported, please follow https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html

### External definition

```shell
none
```

### What does/doesn't work with the external definition?

all

oyooyo, 2024-11-29:
I just received the same device and realized it's not yet supported by zb2mqtt.
But it appears to be more or less identical to a smoke detector called "PA-44Z":
https://www.zigbee2mqtt.io/devices/PA-44Z.html

"PA-44Z" not only looks identical, it contains manufacturer ID "_TZE200_rccxox8p", which is very similar to "_TZE284_rccxox8p".

So I tried to simply add manufacturer id '_TZE284_rccxox8p' to the "PA-44Z" definition and at first sight it seems to work fine.

markro76, 2024-11-29:
I have no plan to write an new Config. Can you please give me the .js File from your Config.

Many Thanks

oyooyo, 2024-11-29:
I haven't written a new config either as I have no experience with that. All I did was opening the file "devices/tuya.js" from the "zigbee-herdsman-converters" package (just search for a file named tuya.js) with a text editor and then simply changing this line:
```
        fingerprint: tuya.fingerprint('TS0601', ['_TZE200_m9skfctm', '_TZE200_rccxox8p']),
```
to
```
        fingerprint: tuya.fingerprint('TS0601', ['_TZE200_m9skfctm', '_TZE200_rccxox8p', '_TZE284_rccxox8p']),
```
That's probably not the proper way to do it, as I assume those changes will get overwritten when updating zigbee2mqtt.
But by that time the device will be supported out of the box, so I hope it will be okay.

markro76, 2024-11-29:
I can´t find the file. Can you tell me were i can find the File ?

atsoller, 2024-11-30:
Any Idea how to make these changes permanent in a docker container. I am running zigbee2mqtt in home assistant. I can change the file in the container, but aber a restart ist is changed back.

markro76, 2024-11-30:
I have the same Problem. I Hope the device will have support in the next Version of zigbee2mqtt :-)

atsoller, 2024-12-01:
If you have Terminal installed in HA you need to switch off save mode. Than you an switch into the z2mqtt conternair with this command.
docker exec -it $(docker ps | grep zigbee2mqtt | cut -d" " -f 1) /bin/sh
Change to the directory
cd node_modules/zigbee-herdsman-converters/devices/
and you can edit the file tuya.js, You need to install a editor or work wirh the installed Vi editor. So its good if you know a bit about linux.. But as I wrote before, i can not get it to work. I guess the fille is always updated when restarting the container.
Maybe someone who knows more about containers can help with that..

guener, 2024-12-17:
It looks like the configuration of PA-44Z works fine. So we probably don't need a new converter. The manufacturer id could be added to the PA-44Z converter. I will do further testing and will report any findings here. Thanks @oyooyo for your hint.

QempZor, 2025-01-06:
The manufacture id is added in 20.13. 
I am on the latest Z2M release, wich has 20.12. 

Is there anyway to update to 20.13?
Iam running the HA Addon

BOFH90, 2025-01-08:
> Is there anyway to update to 21.13? Iam running the HA Addon

The Version 21.13 was already merged. We need to wait for the next Z2M Release.

m-lindemann, 2025-01-17:
While waiting for the Update, I was able to bring the device up with the following external converter. To be clear here, I don´t understand what I´m doing. My ability is limited to google, copy and paste!

`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
        fingerprint: tuya.fingerprint('TS0601', ['_TZE284_rccxox8p', '_TZE222_rccxox8p']),
        model: 'PA-44Z',
        vendor: 'Tuya',
        description: 'Photoelectric smoke detector',
        fromZigbee: [tuya.fz.datapoints],
        toZigbee: [tuya.tz.datapoints],
        onEvent: tuya.onEventSetTime,
        configure: tuya.configureMagicPacket,
        exposes: [
            e.smoke(),
            e.battery(),
            tuya.exposes.silence(),
            e.test(),
            e.numeric('smoke_concentration', ea.STATE).withUnit('ppm').withDescription('Parts per million of smoke detected'),
            e.binary('device_fault', ea.STATE, true, false).withDescription('Indicates a fault with the device'),
        ],
        meta: {
            tuyaDatapoints: [
                [1, 'smoke', tuya.valueConverter.trueFalse0],
                [2, 'smoke_concentration', tuya.valueConverter.divideBy10],
                [11, 'device_fault', tuya.valueConverter.raw],
                [15, 'battery', tuya.valueConverter.raw],
                [16, 'silence', tuya.valueConverter.raw],
                [101, 'test', tuya.valueConverter.raw],
            ],
        },
    };

module.exports = definition;`

markro76, 2025-01-20:
You must restart the Zigbee2mqtt Addon. In the LogFiles of the Addon you can see the loaded Files of the folder. See Screenshot

[image]

withdemon1, 2025-01-20:
> You must restart the Zigbee2mqtt Addon. In the LogFiles of the Addon you can see the loaded Files of the folder. See Screenshot
> 
> [image]

Thank you very much, I see it correctly.

alvarokrn, 2025-02-12:
I have bought recently this device thinking it was PA-44Z. I would love that Zigbee2mqtt supports it.

alvarokrn, 2025-02-25:
Taking in consideration that this model is like a clone of the [Tuya PA-44Z](https://www.zigbee2mqtt.io/devices/PA-44Z.html) and  [looks like it works pretty similar](https://github.com/Koenkk/zigbee2mqtt/issues/24916#issuecomment-2598116216). This should be pretty straightforward to support it, isn't it?

Fograd, 2025-03-20:
> While waiting for the Update, I was able to bring the device up with the following external converter. To be clear here, I don´t understand what I´m doing. My ability is limited to google, copy and paste! Code for converter was found here: [#23250](https://github.com/Koenkk/zigbee2mqtt/issues/23250)
> 
> const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
> const tz = require('zigbee-herdsman-converters/converters/toZigbee');
> const exposes = require('zigbee-herdsman-converters/lib/exposes');
> const reporting = require('zigbee-herdsman-converters/lib/reporting');
> const ota = require('zigbee-herdsman-converters/lib/ota');
> const tuya = require('zigbee-herdsman-converters/lib/tuya');
> const utils = require('zigbee-herdsman-converters/lib/utils');
> const globalStore = require('zigbee-herdsman-converters/lib/store');
> const e = exposes.presets;
> const ea = exposes.access;
> 
> const definition = {
>         fingerprint: tuya.fingerprint('TS0601', ['_TZE284_rccxox8p', '_TZE222_rccxox8p']),
>         model: 'PA-44Z',
>         vendor: 'Tuya',
>         description: 'Photoelectric smoke detector',
>         fromZigbee: [tuya.fz.datapoints],
>         toZigbee: [tuya.tz.datapoints],
>         onEvent: tuya.onEventSetTime,
>         configure: tuya.configureMagicPacket,
>         exposes: [
>             e.smoke(),
>             e.battery(),
>             tuya.exposes.silence(),
>             e.test(),
>             e.numeric('smoke_concentration', ea.STATE).withUnit('ppm').withDescription('Parts per million of smoke detected'),
>             e.binary('device_fault', ea.STATE, true, false).withDescription('Indicates a fault with the device'),
>         ],
>         meta: {
>             tuyaDatapoints: [
>                 [1, 'smoke', tuya.valueConverter.trueFalse0],
>                 [2, 'smoke_concentration', tuya.valueConverter.divideBy10],
>                 [11, 'device_fault', tuya.valueConverter.raw],
>                 [15, 'battery', tuya.valueConverter.raw],
>                 [16, 'silence', tuya.valueConverter.raw],
>                 [101, 'test', tuya.valueConverter.raw],
>             ],
>         },
>     };
> 
> module.exports = definition;

This worked like a charm, thank you!

closed 2025-05-28 by github-actions[bot]