Koenkk/zigbee2mqtt #24918: [SONOFF TRVZB] Wrong name for attribute Local Temperature Calibration

gbasile, 2024-11-25:
### Link

https://sonoff.tech/product/home-appliances/trvzb/

### Model

TRVZB

### Description

Local Temperature Calibration

### Vendor

-

### Picture (link)

-

### Database entry

-

### Notes

Please forgive me if I'm reporting this to the wrong place 🙏 

This issue is related to how Home Assistant names an attribute that belongs to a device integrated with Zigbee2Mqtt

This device has an attribute called `Local Temperature Calibration` that is named correctly in zigbee2mqtt
[CorreosImpresoAdmision.pdf](https://github.com/user-attachments/files/17906645/CorreosImpresoAdmision.pdf)

The same attributes is called `Temperature` on the Home Assistant counterpart, potentially creating some confusion to Home Assistant users 
[image]

acommuni, 2024-12-24:
I have the same issue, 

Here is the current state of the TRVZB 

{
    "battery": 100,
    "child_lock": "UNLOCK",
    "closing_steps": 554,
    "frost_protection_temperature": 7,
    "idle_steps": 30,
    "linkquality": 188,
    "local_temperature": 17.3,
    "local_temperature_calibration": -2.4,
    "occupied_heating_setpoint": 16,
    "open_window": "ON",
    "running_state": "idle",
    "system_mode": "auto",
    "update": {
        "installed_version": 4609,
        "latest_version": 4609,
        "state": "idle"
    },
    "valve_closing_degree": 100,
    "valve_closing_limit_voltage": 2620,
    "valve_motor_running_voltage": 1484,
    "valve_opening_degree": 100,
    "valve_opening_limit_voltage": 2590,
    "weekly_schedule": {
        "friday": "00:00/16 07:00/19 10:00/16 17:00/19 22:00/16 22:30/16",
        "monday": "00:00/16 07:00/19 08:00/16 18:00/19 22:00/19 22:30/16",
        "saturday": "00:00/16 08:00/19 12:30/16 14:00/19 22:00/19 22:30/16",
        "sunday": "00:00/16 08:00/19 12:30/16 14:00/19 22:00/19 22:30/16",
        "thursday": "00:00/16 07:00/19 10:00/16 17:00/19 22:00/16 22:30/16",
        "tuesday": "00:00/16 07:00/19 10:00/16 17:00/19 22:00/16 22:30/16",
        "wednesday": "00:00/16 07:00/19 08:00/16 18:00/19 22:00/19 22:30/16"
    },
    "update_available": null
}

Perhaps it could be link to hvacThermostat report which has a different name : local_temp (vs local_temperature)

[image]

acommuni, 2025-02-26:
Up

stone1609, 2025-04-28:
up