Koenkk/zigbee2mqtt #24924: Z2M is very unforgiving if the ZigbeeAdapter is connected over a network: Its completly crashes the HA-Z2M and is not waiting for a reconnect

bachmarc, 2024-11-26:
### What happened?

If I connect a Network2Zigbee to Z2M then it works fine if the network is rock solid.

If the Adapter is disconnected for a short time or network unstable for a few seconds the whole home assistance add-on container crashes/stops

This is with a provoked IP conflict: 
`06:58:10] INFO: Preparing to start...
[06:58:10] INFO: Socat not enabled
[06:58:10] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2024-11-26 07:00:59] error: 	zh:zstack:znp: Socket error Error: read ECONNRESET
[2024-11-26 07:00:59] error: 	z2m: Adapter disconnected, stopping`

Z2M is stopping .... 

### What did you expect to happen?

I would expect that Z2M is catching the event and retries to reconnect to Network2Zigbee.
The chances are good that it will be available again soon and is only a hick-up because VPN reconnects or what ever.
Even an IP conflict shutdown Z2M... 

Network is not as reliable as a plugged USB that should be considered in error handling.

### How to reproduce it (minimal and precise)

Connect a SLZB-06 net2zigbee adapter to Z2M and disrupt the connection for short or provoke a IP conflict

### Zigbee2MQTT version

1.41.0-1

### Adapter firmware version

20221226

### Adapter

SLZB-06

### Setup

Home assistant Add-on

### Debug log

06:58:10] INFO: Preparing to start...
[06:58:10] INFO: Socat not enabled
[06:58:10] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2024-11-26 07:00:59] error: 	zh:zstack:znp: Socket error Error: read ECONNRESET
[2024-11-26 07:00:59] error: 	z2m: Adapter disconnected, stopping

schtack, 2024-11-26:
I think, I have the same kind of problem.
I'm using a SLZB-06 adapter over cellular 4g link and z2m stop working when adapter is disconected.

Arn0uDz, 2024-11-27:
Also have this problem and would love a wait for reconnect option.

Koenkk, 2024-11-27:
You can use the watchdog option to automatically restart z2m

schtack, 2024-11-27:
Hi @Koenkk, thanks for your reply.
From my understanding, using the watchdog is not a reliable solution, my adapter disconnects at least once every 2 to 6 hours.

Or can you explain to us how to configure the watchdog reliably please?

kind regards.

Koenkk, 2024-11-27:
@schtack then you should look into making the connection more stable.

bachmarc, 2024-11-29:
In my case that was easy. But how would one stabilize a Wireguard connection to beach house?
If my USB stick crashes due to OTA attempt Z2M reports it and reconnects it.

I would consider a network the classical use case for error tolerance. I am not talking about ages just a small timeout that it waits for a re-connection like 60s.

peschee, 2025-01-08:
@bachmarc what did you end up using? I’m running z2m in a LXC on proxmox. As soon as my SLZB disappears for a short time, z2m crashes. 

@Koenkk is the watchdog setting also the way to go when using LXC? This just starts the node process afaik.

Koenkk, 2025-01-09:
@peschee yes watchdog is the way to go, it just restarts the process after some backoff time.

peschee, 2025-01-12:
@Koenkk this does not seem to work. I've setup the sysctl service to use the watchdog setting. as soon as the coordinator goes away, the process is exited and sysctl restarts it:

```shell
Jan 12 09:40:34 zigbee2mqtt pnpm[4901]: Starting Zigbee2MQTT with watchdog (2000,60000,300000,900000,1800000,3600000).
Jan 12 09:40:35 zigbee2mqtt pnpm[4901]: [2025-01-12 09:40:35] info:         z2m: Logging to console, file (filename: log.log)
Jan 12 09:40:35 zigbee2mqtt pnpm[4901]: [2025-01-12 09:40:35] info:         z2m: Starting Zigbee2MQTT version 2.0.0 (commit #unknown)
Jan 12 09:40:35 zigbee2mqtt pnpm[4901]: [2025-01-12 09:40:35] info:         z2m: Starting zigbee-herdsman (3.2.1)
Jan 12 09:40:35 zigbee2mqtt pnpm[4901]: [2025-01-12 09:40:35] info:         zh:zstack:znp: Opening TCP socket with 10.46.0.63:6638
Jan 12 09:40:35 zigbee2mqtt pnpm[4901]: [2025-01-12 09:40:35] error:         zh:zstack:znp: Socket error Error: connect ECONNREFUSED 10.46.0.63:6638
Jan 12 09:40:35 zigbee2mqtt pnpm[4901]: [2025-01-12 09:40:35] error:         z2m: Error while starting zigbee-herdsman
Jan 12 09:40:35 zigbee2mqtt pnpm[4901]: [2025-01-12 09:40:35] error:         z2m: Failed to start zigbee-herdsman
Jan 12 09:40:35 zigbee2mqtt pnpm[4901]: [2025-01-12 09:40:35] error:         z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
Jan 12 09:40:35 zigbee2mqtt pnpm[4901]: [2025-01-12 09:40:35] error:         z2m: Exiting...
Jan 12 09:40:35 zigbee2mqtt pnpm[4901]: [2025-01-12 09:40:35] error:         z2m: Error: Error while opening socket
Jan 12 09:40:35 zigbee2mqtt pnpm[4901]:     at Socket.<anonymous> (/opt/zigbee2mqtt/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:158:24)
Jan 12 09:40:35 zigbee2mqtt pnpm[4901]:     at Socket.emit (node:events:536:35)
Jan 12 09:40:35 zigbee2mqtt pnpm[4901]:     at emitErrorNT (node:internal/streams/destroy:170:8)
Jan 12 09:40:35 zigbee2mqtt pnpm[4901]:     at emitErrorCloseNT (node:internal/streams/destroy:129:3)
Jan 12 09:40:35 zigbee2mqtt pnpm[4901]:     at processTicksAndRejections (node:internal/process/task_queues:90:21)
Jan 12 09:40:35 zigbee2mqtt pnpm[4901]: /opt/zigbee2mqtt/node_modules/.pnpm/readable-stream@3.6.2/node_modules/readable-stream/lib/_stream_writable.js:264
Jan 12 09:40:35 zigbee2mqtt pnpm[4901]:   var er = new ERR_STREAM_WRITE_AFTER_END();
Jan 12 09:40:35 zigbee2mqtt pnpm[4901]:            ^
Jan 12 09:40:35 zigbee2mqtt pnpm[4901]: Error: write after end
Jan 12 09:40:35 zigbee2mqtt pnpm[4901]:     at writeAfterEnd (/opt/zigbee2mqtt/node_modules/.pnpm/readable-stream@3.6.2/node_modules/readable-stream/lib/_stream_writable.js:264:12)
Jan 12 09:40:35 zigbee2mqtt pnpm[4901]:     at DerivedLogger.Writable.write (/opt/zigbee2mqtt/node_modules/.pnpm/readable-stream@3.6.2/node_modules/readable-stream/lib/_stream_writable.js:300:21)
Jan 12 09:40:35 zigbee2mqtt pnpm[4901]:     at DerivedLogger.log (/opt/zigbee2mqtt/node_modules/.pnpm/winston@3.17.0/node_modules/winston/lib/winston/logger.js:231:12)
Jan 12 09:40:35 zigbee2mqtt pnpm[4901]:     at Logger.log (/opt/zigbee2mqtt/lib/util/logger.ts:198:25)
Jan 12 09:40:35 zigbee2mqtt pnpm[4901]:     at Logger.info (/opt/zigbee2mqtt/lib/util/logger.ts:211:14)
Jan 12 09:40:35 zigbee2mqtt pnpm[4901]:     at Znp.onPortClose (/opt/zigbee2mqtt/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:88:16)
Jan 12 09:40:35 zigbee2mqtt pnpm[4901]:     at Object.onceWrapper (node:events:639:26)
Jan 12 09:40:35 zigbee2mqtt pnpm[4901]:     at Socket.emit (node:events:524:28)
Jan 12 09:40:35 zigbee2mqtt pnpm[4901]:     at TCP.<anonymous> (node:net:351:12)
Jan 12 09:40:35 zigbee2mqtt pnpm[4889]:  ELIFECYCLE  Command failed with exit code 1.
```

shouldn't the process be kept running to the watchdog can coordinate the restarts? or do I need to setup the sysctl service differently?

Koenkk, 2025-01-12:
You shouldn't, but the watchdog fails due to the `write after end` error which has been fixed in https://github.com/Koenkk/zigbee2mqtt/pull/25737 

The fix is available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2025-03-21 by github-actions[bot]