Koenkk/zigbee2mqtt #24927: [New device support]: Zigbee Tuya Smart Dual Water Timer

mrreebdoog, 2024-11-26:
### Link

https://www.aliexpress.com/item/1005007549776847.html?spm=a2g0o.order_list.order_list_main.21.2bd51802BEBPeT

### Database entry

Zigbee Model TS0601 Zigbee Manufacturer _TZE284_eaet5qt5 IEEE Address 0x30fb10fffe181013 Network address 0xD2A7 / 53927

### Zigbee2MQTT version

1.41.0-1

### Comments

Created a new definition for the AliExpress dual water meters TS0601 _TZE284_eaet5qt5
Can control it ok, but cannot get the battery to report correctly at all, and cannot get a configure line to work at all so the data values were a lot of guess work

### External definition

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');

const definition = {
    fingerprint: [
        {
            modelID: 'TS0601',
            manufacturerName: '_TZE284_eaet5qt5',
        },
    ],
    model: 'SGW08Z',
    vendor: 'Tuya',
    description: 'Smart Dual Water Timer',
    fromZigbee: [tuya.fz.datapoints, tuya.fz.battery],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime,
    exposes: [
        e.battery(), //Doesnt appear to give real values - Can't find the Voltage either
        e.binary('state_l1', ea.STATE_SET, true, false).withLabel('Valve 1'),
        e.binary('state_l2', ea.STATE_SET, true, false).withLabel('Valve 2'),
        e.numeric('Countdown_Timer_1', ea.STATE_SET).withValueMin(0).withValueMax(255).withUnit('min').withDescription('Countdown timer until shutoff'),
        e.numeric('Countdown_Timer_2', ea.STATE_SET).withValueMin(0).withValueMax(255).withUnit('min').withDescription('Countdown timer until shutoff'),
        e.binary('Valve_State_1', ea.STATE, 'open', 'closed').withDescription('Binary Valve 1 State'),
        e.binary('Valve_State_2', ea.STATE, 'open', 'closed').withDescription('Binary Valve 2 State'),
    ],
    meta: {
        tuyaDatapoints: [
            [1, 'state_l1', tuya.valueConverter.raw],   //Can get and set the left valve
            [2, 'state_l2', tuya.valueConverter.raw],   //Can get and set the right valve         
            [13, 'Countdown_Timer_1', tuya.valueConverter.countdown],   //Can get and set the left valve countdown timer
            [14, 'Countdown_Timer_2', tuya.valueConverter.countdown],   //Can get and set the right valve countdown timer
            [104, 'Valve_State_1', tuya.valueConverterBasic.lookup({    //Only get left valve value
                        open: tuya.enum(0),
                        closed: tuya.enum(2),
                    })], //0 when on, 2 when off
            [105, 'Valve_State_2', tuya.valueConverterBasic.lookup({    //Only get right valve value
                        open: tuya.enum(0),
                        closed: tuya.enum(2),
                    })], //0 when on, 2 when off
            ],
    },
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

Data points all work, can control the timer and turn on and off.
Cannot get a configure line to work, and it doesn't appear to want to act like a switch.
Battery value is very wrong (reports 10 when mostly full)

koshisan, 2024-12-06:
Unfortunately, this does not work for me:

[2024-12-06 12:13:59] error: 	z2m: Failed to load external converter file 'TS0601.js' (Cannot read properties of undefined (reading 'options'))
[2024-12-06 12:13:59] error: 	z2m: Probably there is a syntax error in the file or the external converter is not compatible with the current Zigbee2MQTT version

I am on 1.42.0-2 - is this already outdated?

mrreebdoog, 2024-12-06:
Unsure, I haven't updated yet.
This is my exact file as of right now and it appears to work, just no battery info (But rename to .js)

[tuyaDualWaterTimer.txt](https://github.com/user-attachments/files/18032368/tuyaDualWaterTimer.txt)

koshisan, 2024-12-06:
Thanks! That one is working! 
(Except for the battery status, but that is expected)

Chengxuyuan0, 2024-12-11:
> Unfortunately, this does not work for me:
>
> [2024-12-06 12:13:59] error: z2m: Failed to load external converter file 'TS0601.js' (Cannot read properties of undefined (reading 'options')) [2024-12-06 [...]

Hello there! I understand you're facing the same issue. How did you resolve it?

mrreebdoog, 2024-12-11:
> Hello there! I understand you're facing the same issue. How did you resolve it?

Try the file I posted just above

Aralox, 2024-12-12:
@mrreebdoog thank you for this, I've got one of these timers and your converter works well!
I also have a Tuya zigbee single tap timer (TS0049 _TZ3210_0jxeoadc), and that works fine with z2m but it doesn't have an explicit 'valve state' property like you have for the dual timer. I really like this property, because when I was working out the flakiness of my zigbee mesh, my timer would sometimes not respond to commands and I'd have no idea what its state was without physically checking it.
Anyway thanks, hope you/someone figures out the battery thing and we get this merged into the product. :)

dantol1, 2024-12-28:
Awesome work, I just bought this too and since I'm a beginner in zigbee2mqtt was a little overwhelmed by trying to write a custom converter for it, you saved me the trouble 😁

mgsol2004, 2025-02-23:
despues de efecctuar pequeñas modificaciones, me termino funcionando, pero el registro de temperatura no hay caso.

[tuyaDualWaterTimer (1).txt](https://github.com/user-attachments/files/18926658/tuyaDualWaterTimer.1.txt)

mgsol2004, 2025-02-23:
He corregido el código reemplazando tuya.valueConverter.lookup por tuya.valueConverterBasic.lookup, que es la función correcta en versiones recientes de Zigbee2MQTT.

Pasos a seguir
Reemplaza el archivo tuyaDualWaterTimer.js con este código corregido.
Asegúrate de que configuration.yaml contenga:
yaml
Copiar
Editar
external_converters:
  - tuyaDualWaterTimer.js
Reinicia Zigbee2MQTT y revisa si el dispositivo es detectado correctamente

mrestivill, 2025-02-26:
I've created another external converted based on this dp endpoints: https://github.com/prairiesnpr/zha-device-handlers/blob/418803768b4bb75c9fe9c083333961dbfe9aeb31/zhaquirks/tuya/ts0601_valve.py#L488

```javascript
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    fingerprint: tuya.fingerprint('TS0601', ['_TZE284_eaet5qt5']),
    model: 'TS0601_water_switch',
    vendor: 'Tuya',
    description: 'Dual water valve',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime,
    configure: tuya.configureMagicPacket,
    exposes: [
        e.enum('valve_status_1', ea.STATE, ['manual', 'auto', 'idle'])
            .withDescription('Valve 1 status (manual, auto, idle)').withLabel('Valve 1 Status'),
        e.enum('valve_status_2', ea.STATE, ['manual', 'auto', 'idle'])
            .withDescription('Valve 2 status (manual, auto, idle)').withLabel('Valve 2 Status'),
        e.switch().withEndpoint('l1').withDescription('Valve 1 on/off').withLabel('Valve 1'),
        e.switch().withEndpoint('l2').withDescription('Valve 2 on/off').withLabel('Valve 2'),
        e.numeric('valve_countdown_1', ea.STATE_SET)
            .withUnit('min')
            .withDescription('Valve 1 max countdown in minutes')
            .withValueMin(0)
            .withValueMax(1440),
        e.numeric('valve_countdown_2', ea.STATE_SET)
            .withUnit('min')
            .withDescription('Valve 2 max countdown in minutes')
            .withValueMin(0)
            .withValueMax(1440),
        e.numeric('valve_duration_1', ea.STATE)
            .withUnit('s')
            .withDescription('Valve 1 irrigation last duration in seconds'),
        e.numeric('valve_duration_2', ea.STATE)
            .withUnit('s')
            .withDescription('Valve 2 irrigation last duration in seconds'),
        e.battery(),
    ],
    meta: {
        tuyaDatapoints: [
            [1, 'state_l1', tuya.valueConverter.onOff], // Valve 1 on/off
            [2, 'state_l2', tuya.valueConverter.onOff], // Valve 2 on/off
            [13, 'valve_countdown_1', tuya.valueConverter.raw], // Valve 1 countdown
            [14, 'valve_countdown_2', tuya.valueConverter.raw], // Valve 2 countdown
            [25, 'valve_duration_1', tuya.valueConverter.raw], // Valve 1 duration
            [26, 'valve_duration_2', tuya.valueConverter.raw], // Valve 2 duration
            [59, 'battery', tuya.valueConverter.batteryPercentage], // Battery 
            [104, 'valve_status_1', tuya.valueConverterBasic.lookup({'manual': 0, 'auto': 1, 'idle': 2})], // Valve 1 status
            [105, 'valve_status_2', tuya.valueConverterBasic.lookup({'manual': 0, 'auto': 1, 'idle': 2})], // Valve 2 status
        ],
        multiEndpoint: true, // Enable multi-endpoint support
    },
    endpoint: (device) => {
        return {
            l1: 1,
            l2: 1,
        };
    },
};

module.exports = definition;
```

mrreebdoog, 2025-02-28:
@mrestivill Thanks, but for some reason mine will not work at all with _configure: tuya.configureMagicPacket_ in the file, and cant use multiendpoint either.

mrestivill, 2025-02-28:
I've created a PR: https://github.com/Koenkk/zigbee-herdsman-converters/pull/8898

mrestivill, 2025-02-28:
I'm using 2.0 version of zigbee2mqtt, maybe this is the difference. I've seen at the issue that you were using 1.41 version.

> [@mrestivill](https://github.com/mrestivill) Thanks, but for some reason mine will not work at all with _configure: tuya.configureMagicPacket_ in the file, and cant use multiendpoint either.

gbilic, 2025-04-21:
The battery state seems correctly retreived using raw value converter. 
`[59, 'battery', tuya.valueConverter.raw], // Battery`

Quokka1979, 2025-04-25:
zigbeeModel: TS0601
    model: TS0601
    vendor: _TZE284_fhvpaltk

I can confirm this: [59, 'battery', tuya.valueConverter.raw], // Battery

@Koenkk can you please add this to the commit

Koenkk, 2025-04-26:
This device is not supported yet, can someone make a PR?

Could you make a PR by clicking [here](https://github.com/Koenkk/zigbee-herdsman-converters/edit/master/src/devices/tuya.ts)?

Quokka1979, 2025-04-27:
I don’t know how to do this, I’m sorry

Jaegerstefan, 2025-04-29:
The same device is also sold with the manufactor id _TZE284_fhvpaltk. The converter   @mgsol2004 worked here too in Z2M 2.1.1 with an external converter.

Koenkk, 2025-05-01:
Added, could someone submit a picture for the docs?

Could you also submit a picture for the docs? This can be done by:
1. Create a fork by clicking [here](https://github.com/Koenkk/zigbee2mqtt.io/fork)
2. Go to the `public/images/devices` directory, *Add file* -> *Upload files*
3. Upload the files and press *Commit changes*
4. Press *Contribute* -> *Open pull request* -> update title/description -> *Create pull request*

Make sure that:
- The filename is `TS0601_water_switch.png`
- The size is 512x512
- The background is transparent (use e.g. [Adobe remove background](https://new.express.adobe.com/tools/remove-background))

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.

Jaegerstefan, 2025-05-02:
Done in [https://github.com/Koenkk/zigbee2mqtt.io/pull/3755](https://github.com/Koenkk/zigbee2mqtt.io/pull/3755)

closed 2025-05-03 by Koenkk

MOLLYMAN01, 2025-06-26:
Leider funktioniert es nicht richtig.
Es wird lediglichlich der Status vom Gerät zu z2m übertragen. Umgekehrt,  also eine Schaltsnforderung wird nicht umgesetzt.

Koenkk, 2025-07-20:
Valve control is fixed in the latest dev.

> Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.

handfreezer, 2025-08-13:
I confirm on 2.6.0 of z2m : my _TZE284_fhvpaltk seems working fine.