Koenkk/zigbee2mqtt #24935: Aqara (RTCGQ11LM) Motion Sensor: No temperature and battery value since Z-Stack_3.x.0_coordinator_20240710

jtig2019, 2024-11-27:
### What happened?

- updated my Sonoff Dongle Plus to the firmware version Z-Stack_3.x.0_coordinator_20240710
- by changing the pan id and removing the coordinator_backup.json I had to repair all my devices
- the aqara motion sensor repairing was successful, but the temperature and battery value are not exposed
- I tried repairing again and again, but the values were still not exposed

Used pairing steps: https://www.zigbee2mqtt.io/devices/RTCGQ11LM.html

### What did you expect to happen?

 Battery and temperature should be exposed.

### Zigbee2MQTT version

1.41.0

### Adapter firmware version

20240710

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus ZBDongle-P

### Setup

docker

### Debug log

-

lux73, 2024-11-27:
no Problem here, all of my RTCGQ11LM exposing all Values with FW20240710 @ CC2652RB Stick

mrmaximas, 2024-11-27:
no problem with 1.41.0-dev commit: [5ca4cf4a] on HMG-01 POE Plus with CC2652P7 20240710

jtig2019, 2024-11-28:
After several hours of runtime without any further attempts to repair the devices, the temperature and battery level are now displayed.

Thx for your feedback!

closed 2024-11-28 by jtig2019