Koenkk/zigbee2mqtt #24936: [Wrong device]: Tuya eWeLink ZG-204ZL indeftifies as eWeLink SQ510A

Janne112x, 2024-11-27:
### Link

https://de.aliexpress.com/item/1005007370984412.html?spm=a2g0o.order_list.order_list_main.5.34005c5fd85f53&gatewayAdapt=glo2deu

### Model

https://de.aliexpress.com/item/1005007370984412.html?spm=a2g0o.order_list.order_list_main.5.34005c5fd85f53&gatewayAdapt=glo2deu

### Description

Motion Sensor ZG-204ZL

### Vendor

Tuya

### Picture (link)

https://www.zigbee2mqtt.io/images/devices/ZG-204ZL.png

### Database entry

dont know where to find

### Notes

Tuya eWeLink ZG-204ZL indeftifies as eWeLink SQ510A

mariogppcom, 2024-11-28:
Hey! Before applying the update - 1.41.0-1 - my "ZMS-102 PIR sensors" worked correctly. Now they are being identified as "eWeLink - Water leak sensor - SQ510A".

mariogppcom, 2024-11-28:
To confirm, I rolled back to version 1.40.2-1 and the PIR sensors started working correctly again (Zigbee Model: SNZB-03, Zigbee Manufacturer: eWeLink, Description: motion sensor).

Procedure performed by restoring the backup made by Home Assistant (Configuration-> System -> Backups -> file "addon_9336c2b0_zigbee2mqtt_1.40.2-1").

icagn, 2024-11-29:
Bonjour à tous, j'ai le même soucis

Janne112x, 2024-12-02:
It is fixed, with the new update.

closed 2025-02-08 by github-actions[bot]