Koenkk/zigbee2mqtt #24941: OTA Update: Cannot read properties of undefined (reading 'data')

CrazYoshi, 2024-11-27:
### What happened?

I'm trying to update the Philips Light Bulb [Philips 8718696449691](https://www.zigbee2mqtt.io/devices/8718696449691.html#philips-8718696449691) firmware using the OTA section in Z2M.
A new firmware is found but the update fail logging this error: Update of '0x0017880104b9944e' failed (Cannot read properties of undefined (reading 'data'))

### What did you expect to happen?

I expected the update of the device should start and the firmware is updated at the end of the process

### How to reproduce it (minimal and precise)

Have a Philips 8718696449691 bulb with an older firmware and try to update that to the latest firmware

### Zigbee2MQTT version

1.41.0

### Adapter firmware version

20240710

### Adapter

zStack3x0

### Setup

Docker container

### Debug log

```
[2024-11-27 21:54:38] info: 	z2m: Updating '0x0017880104b9944e' to latest firmware

[2024-11-27 21:54:39] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x0017880104b9944e', payload '{"brightness":254,"linkquality":80,"power_on_behavior":null,"state":"OFF","update":{"installed_version":16782598,"latest_version":16783874,"state":"available"},"update_available":null}'

[2024-11-27 21:54:39] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{"id":"0x0017880104b9944e"},"error":"Update of '0x0017880104b9944e' failed (Cannot read properties of undefined (reading 'data'))","status":"error","transaction":"3k86p-16"}'

[2024-11-27 21:54:39] error: 	z2m: Update of '0x0017880104b9944e' failed (Cannot read properties of undefined (reading 'data'))
```

dieugab, 2024-11-28:
Having the same issue with Philips [8718696695203](https://www.zigbee2mqtt.io/devices/8718696695203.html#philips-8718696695203)

abort, 2024-11-28:
Same occurs for Philips [8718696548738](https://www.zigbee2mqtt.io/devices/8718696548738.html)

olivercjung, 2024-11-29:
Same problem with Philips [9290012573A](https://www.zigbee2mqtt.io/devices/9290012573A.html#philips-9290012573a).

Koenkk, 2024-11-30:
Please test again after updating to the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) on 2 December, this contains quite some refactoring to the OTA code.

CrazYoshi, 2024-12-01:
Sorry, my bad. I’ll test again in few days

Koenkk, 2024-12-01:
It's in dev now!

CrazYoshi, 2024-12-02:
I've tested again, the previous error is solved but the update cannot find the image:
```
[2024-12-02 20:45:46] info: 	z2m: Updating 'luci_scrivaniastudio' to latest firmware

[2024-12-02 20:45:48] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/luci_scrivaniastudio', payload '{"brightness":254,"linkquality":72,"power_on_behavior":null,"state":"OFF","update":{"installed_version":16782598,"latest_version":16783874,"state":"available"}}'

[2024-12-02 20:45:48] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{},"error":"Update of 'luci_scrivaniastudio' failed (Invalid response from https://github.com/Koenkk/zigbee-OTA/raw/master/images1/Hue/100B-010C-01001A02-ConfLight-Lamps_0012.zigbee status=404.)","status":"error","transaction":"t13in-1"}'

[2024-12-02 20:45:48] error: 	z2m: Update of 'luci_scrivaniastudio' failed (Invalid response from https://github.com/Koenkk/zigbee-OTA/raw/master/images1/Hue/100B-010C-01001A02-ConfLight-Lamps_0012.zigbee status=404.)

[2024-12-02 20:45:58] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/luci_scrivaniastudio', payload '{"brightness":254,"linkquality":72,"power_on_behavior":null,"state":"OFF","update":{"installed_version":16782598,"latest_version":16783874,"state":"available"}}'

[2024-12-02 20:45:58] info: 	z2m: Update available for 'luci_scrivaniastudio'
```
I see that that image is present on https://github.com/Koenkk/zigbee-OTA/raw/master/images/Hue/100B-010C-01001A02-ConfLight-Lamps_0012.zigbee instead. Could it be just a typo?

Koenkk, 2024-12-03:
Should be fixed now, can you try again?

CrazYoshi, 2024-12-03:
Yes, I confirm it is fixed now. I was able to successfully upgrade the firmware using OTA.

closed 2024-12-04 by Koenkk