Koenkk/zigbee2mqtt #24942: Brennenstuhl Thermostat not supported but officially supported

David-Glom, 2024-11-27:
### What happened?

I bought a brennenstuhl Thermostat( HT CZ 01), which is officially supported, is declined by Zigbee2MQTT

“ z2m: Device '0x3c2ef5fffe9af791' with Zigbee model 'TS0601' and manufacturer name '_TZE200_lrznf59v' is NOT supported, please follow…” 

Maybe it is just a new manufacturer name? 

By the way, thank you very much for your fantastic work :smile

### What did you expect to happen?

I expected that it gets connected to Z2M

### How to reproduce it (minimal and precise)

Buy a new Brennenstuhl HT CZ 01 and try to join with Z2M

### Zigbee2MQTT version

1.41.0

### Adapter firmware version

7.4.4 [GA]

### Adapter

EmberZNet

### Setup

Addon on Homeassistant OS on Pi 5

lux73, 2024-11-28:
the Manufacturer name indicates that is a TuYA Re-brand

TuYA always change '_TZE200_lrznf59v' and every time it has to be added to z2m before recognize

David-Glom, 2024-11-28:
Thanks for your fast answer! Is there anywhere a short description, how to add it?

lux73, 2024-11-28:
sure -> https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html

David-Glom, 2024-11-28:
So I found out that it is a [Siterwell GS361A-H04](https://www.zigbee2mqtt.io/devices/TS0601_thermostat.html#siterwell-gs361a-h04) just rebranded, so probably just the fingerprint changed? Is it possible to test locally? The "problem" testing it locally is, that the siterwell uses a lot of legacy configs, which doesn't really resolve well in the external definition I think. Is it possible to modify the original siterwell.ts (src/devices/siterwell.ts)? Thanks for your help so far!

David-Glom, 2024-11-29:
Solved It, using this script the thermostat works as expected:

```
//const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const legacy = require('zigbee-herdsman-converters/lib/legacy');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const {DefinitionWithExtend} = require('zigbee-herdsman-converters/lib/types')
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE200_lrznf59v',
    description: 'Radiator valve with thermostat',
    fromZigbee: [legacy.fz.tuya_thermostat, fz.ignore_basic_report],
    toZigbee: [
        legacy.tz.tuya_thermostat_child_lock,
        legacy.tz.siterwell_thermostat_window_detection,
        legacy.tz.tuya_thermostat_valve_detection,
        legacy.tz.tuya_thermostat_current_heating_setpoint,
        legacy.tz.tuya_thermostat_system_mode,
        legacy.tz.tuya_thermostat_auto_lock,
        legacy.tz.tuya_thermostat_calibration,
        legacy.tz.tuya_thermostat_min_temp,
        legacy.tz.tuya_thermostat_max_temp,
        legacy.tz.tuya_thermostat_comfort_temp,
        legacy.tz.tuya_thermostat_eco_temp,
        legacy.tz.tuya_thermostat_force,
        legacy.tz.tuya_thermostat_preset,
        legacy.tz.tuya_thermostat_boost_time,
        ],
    exposes: [
            e.child_lock(),
            e.window_detection(),
            e.battery(),
            e.valve_detection(),
            e
                .climate()
                .withSetpoint('current_heating_setpoint', 5, 30, 0.5, ea.STATE_SET)
                .withLocalTemperature(ea.STATE)
                .withSystemMode(['off', 'auto', 'heat'], ea.STATE_SET)
                .withRunningState(['idle', 'heat'], ea.STATE),
        ],
    meta: {
        tuyaThermostatSystemMode: legacy.thermostatSystemModes4,
        tuyaThermostatPreset: legacy.thermostatPresets,
        tuyaThermostatPresetToSystemMode: legacy.thermostatSystemModes4,
        },
    extend: [],
};

module.exports = definition;
```

So it really is a [Siterwell GS361A-H04](https://www.zigbee2mqtt.io/supported-devices/#siterwell-gs361a-h04) inside. So maybe it would be more easier to just add this model info to the Sitterwell file @Koenkk ?

closed 2024-11-29 by David-Glom

Koenkk, 2024-11-30:
Added, will be supported with tomorrows release.

arteck, 2024-12-01:
@Koenkk now Siterwell GS361A-H04 does not work . i can't set the temperetaure... plz check

Koenkk, 2024-12-01:
@arteck what doesn't work?

arteck, 2024-12-02:
this version works 
[image]