Koenkk/zigbee2mqtt #24943: ZWT198 thermostat sending wrong 'local_temperature_calibration'

marcospm92, 2024-11-27:
### What happened?

I installed a thermostat (Avatto ZWT198), and I am finding a strange behavior with the 'local_temperature_calibration' value sent to my Home Assistant instance.

If I set the value to, for example, -1°C, it is automatically changed after a short period to a way bigger value, in my case 6552.6.
Looks like it could be a mistake in how negative values are managed.

Find here an example of the payload:
info 2024-11-27 23:35:56z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/termostato_salon', payload '{"backlight_mode":"medium","child_lock":"UNLOCK","current_heating_setpoint":20,"deadzone_temperature":1,"factory_reset":"OFF","frost_protection":"ON","linkquality":163,"local_temperature":20,"local_temperature_calibration":6552.6,"max_temperature_limit":60,"preset":"manual","running_state":"idle","schedule_holiday":"08:00/22.0°C 23:00/16.0°C","schedule_weekday":"06:00/20.0°C 08:00/16.0°C 11:30/16.0°C 12:30/16.0°C 17:00/22.0°C 22:00/16.0°C","sensor":"internal","system_mode":"heat","working_day":"5-2"}'

This is causing a lot of error messages in Home Assistant, because the value should be between -9.9 and 9.9.

### What did you expect to happen?

If I put a negative value in the allowed range, it should be sent like that I assume

### How to reproduce it (minimal and precise)

Just set a negative temperature, ie -1.0 in the 'local_temperature_calibration' and wait for a couple of minutes

### Zigbee2MQTT version

1.41.0

### Adapter firmware version

20230507

### Adapter

Electrolama ZZH!

### Setup

Docker container in a Ubuntu 22.04 i3 laptop

alexsotoaguilera, 2024-11-29:
same here https://github.com/Koenkk/zigbee2mqtt/issues/24758

majkers, 2025-02-03:
Same here https://github.com/Koenkk/zigbee2mqtt/issues/26157

OpaApo, 2025-02-19:
I have exactly the same problem. The value is first set correctly but after a random time it transforms to this big invalid number.

majkers, 2025-04-22:
still exists

closed 2025-06-29 by github-actions[bot]