Koenkk/zigbee2mqtt #24945: Tuya PA-44Z Smoke Det Fails to reset Alarm and PPM State

ckglobalroaming, 2024-11-28:
### What happened?

When Test or trigger Tuya PA-44Z the device reports SMOKE:{TRUE} and PPM:{VALUE} however when the device returns to normal state, Z2M does not refect the normal state values. both SMOKE and PPM remain at the alarmed states indefinitley.

### What did you expect to happen?

When Test or trigger Tuya PA-44Z the device reports SMOKE:{TRUE} and PPM:{VALUE}. When the device returns to normal state, Z2M reports the normal state values  SMOKE:{FALSE} and PPM:0

### How to reproduce it (minimal and precise)

Spray test smoke, get alarm, observe alarm states in Z2M. Clear smoke, values remain in Z2M hours later.
I would expect to be able to resolve this with Zigbee Endpoint/Cluster/repintervals but this is TUYA notZigbee.
Any assistance appreciated. I purchased 20 of these damn things so i could hook them to Homeassistant and they are no value if they stay in alarm...

### Zigbee2MQTT version

1.41.0

### Adapter firmware version

20221226 (best stability)

### Adapter

CC2652P SLZB-06 Ethernet

### Setup

Homeassistant

ckglobalroaming, 2024-11-28:
Device is normal for hours but still reports

```
[2024-11-28 10:49:59] debug:    z2m: Received Zigbee message from 'Smoke Detector Garage', type 'attributeReport', cluster 'genBasic', data '{"modelId":"TS0601"}' from endpoint 1 with groupID 0
[2024-11-28 10:49:59] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Smoke Detector Garage', payload '{"battery":100,"device_fault":0,"last_seen":"2024-11-28T10:49:59+10:00","linkquality":131,"silence":false,"smoke":true,"smoke_concentration":100,"test":false}'
```

alex1702, 2024-11-29:
I have exactly the same problem and don't know whether it's due to zigbee2mqtt or the smoke detectors themselves. If it is due to the smoke detectors, is it possible to reset the status after a certain time?

ckglobalroaming, 2024-12-13:
Bump. Smoke detectors. Assistance please

Poci, 2024-12-22:
Same problem here, both SMOKE and PPM remain at the alarmed states indefinitely.

Poci, 2024-12-22:
> It seems to be due to the smoke detectors, someone else has also bought such models but from a different dealer and they work. I then ordered these too and complained to aliexpress about the others. The status of the new ones is reset correctly. But I have only tested one out of 8 so far.
> 
> Purchased from: https://a.aliexpress.com/_ExTZED6
> 
> But I still think it would be good to have a workaround for these strangely behaving ones.

Mine detector is this one https://vi.aliexpress.com/item/1005005855776130.html

mjenda, 2024-12-25:
> It seems to be due to the smoke detectors, someone else has also bought such models but from a different dealer and they work. I then ordered these too and complained to aliexpress about the others. [...]

I've recently received a device from the same seller, and encountered the same issue. The only way to fix it was to reconnect it to the network.

During testing (by putting out the match near the sensor), I discovered that the device gets stuck only if I press the mute button, which is inevitable because the buzzer is very loud. However, if I tolerate the noise and avoid pressing the mute button, it eventually resets to normal operation on its own.

ckglobalroaming, 2024-12-25:
Pulled them all out sent them back. Lesson learnt.

Gorgue, 2025-01-11:
I'm the same issue

efranulic, 2025-02-06:
Yes, I have this issue too

efranulic, 2025-02-07:
> It seems to be due to the smoke detectors, someone else has also bought such models but from a different dealer and they work. I then ordered these too and complained to aliexpress about the others. [...]

I think this could be tested connecting the smoke detector directly to Tuya app and checking if it behaves like when connected to z2m.

henkiejan1, 2025-02-22:
Have 4 of this sensors and the concentration wil still  report a value after smoke is detected. But when i removed the battery press few times on the test button and installed the battery again it's back to zero. If i press the test button just for test it will test correctly and press quicly again and it will mute. But test and it will also correctly back to normal. But the silence button in Z2M does  nothing and does not work...

HolgerReimers, 2025-02-23:
> Have 4 of this sensors and the concentration wil still report a value after smoke is detected. But when i removed the battery press few times on the test button and installed the battery again it's back to zero. 

That helped and worked for me as well. Thank you!

cypherbits, 2025-03-12:
Same issue on the latest version. Plus the self-test action not working with error message 'Error: No datapoint defined for 'self_test''

HeinMelissen, 2025-03-20:
I've got 9 of these smoke detectors. Use normal 9v (alkaline) batteries, but they drain within 2 months.
Is this a problem with these smoke detectors, or do I need an update (OTA or something)

alvarokrn, 2025-03-25:
I have the same problem. I did a test with real smoke and it still show smoke concentration 20ppm

alvarokrn, 2025-05-25:
Is this solved?

alvarokrn, 2025-07-31:
Commenting just to keep the ticket open

SmOeL-swdev, 2025-08-01:
Same issue here with this smoke detector (_TZE284_rccxox8p), via MQTT there is a set topic which can be used to reset and let the device reports its entire status, however for such a device this should be done automaticly. Otherwise the workaround would need to be that the report is requested upon a certain interval

cypherbits, 2025-10-01:
No activity does not mean it is fixed.

ckglobalroaming, 2025-10-01:
Not resolved.

alvarokrn, 2025-12-16:
No activity does not mean it is fixed.

SmOeL-swdev, 2026-02-17:
> No issue is still remaining, same as from I started this thread. But learned to life with it for now. As long as smoke is detected, and they don’t falsely trigger them selves I’m happy (enough) with them. Resetting them is 5 minutes and my house does not burn down that often luckily. Still would be nice if Zigbee2Mqtt could solve this issue on there end?

You've reported the issue and multiple others facing the same issue, as you can see in this thread. You figured a workable solution for you while the others don't so that incentive enough to get this fixed

alvarokrn, 2026-04-19:
It still isn't fixed

closed 2026-09-21 by github-actions[bot]

TheOneOgre, 2026-11-11:
### Workaround: manually clear the cached alarm without touching the detector

I'm having the same issue with a PA-44Z (`TS0601`, manufacturer `_TZE284_rccxox8p`) on Zigbee2MQTT **2.14.2**. The audible alarm stops, but Z2M continues showing smoke detected and a concentration of 100.

I was able to clear it by stopping Z2M and editing `state.json`, but doing that every time is pretty clunky. I now have an external extension that clears those cached values while Z2M is running, triggered through MQTT or a Home Assistant script. I've confirmed the manual reset works on my setup.

**This is a workaround, not a fix for the underlying reporting issue.** It doesn't query or reset the physical detector, and it doesn't establish that smoke has cleared. Only run it after you've confirmed that yourself.

It uses Z2M's `publishEntityState()` function, which updates the internal cache and publishes the updated state to MQTT. It changes only:

```json
{
  "smoke": false,
  "smoke_concentration": 0
}
```

Battery, fault, silence, and other properties are left alone. The extension doesn't block future device reports, although I haven't established reliable behavior through subsequent full alarm/clear cycles yet.

### 1. Install the extension

Create an `external_extensions` folder inside your Zigbee2MQTT data directory, alongside `configuration.yaml`, if it doesn't already exist.

Save the following as:

```text
external_extensions/smoke-cache-reset.mjs
```

For Docker installations, this is normally `/app/data/external_extensions/` **inside the container**. Use the corresponding mapped directory on the host.

Make sure external JavaScript is enabled. If needed, add this under your existing `advanced` section in `configuration.yaml`—don't create a second `advanced` section:

```yaml
advanced:
  enable_external_js: true
```

<details>
<summary>Click to expand the extension code</summary>

```javascript
export default class SmokeCacheReset {
    constructor(zigbee, mqtt, state, publishEntityState, eventBus,
        enableDisableExtension, restartCallback, addExtension, settings, logger) {
        Object.assign(this, {
            zigbee, mqtt, state, publishEntityState,
            eventBus, settings, logger
        });
        this.queue = Promise.resolve();
    }

    eligible(device) {
        return device?.isDevice?.() === true &&
            device.definition?.model === 'PA-44Z';
    }

    describe(device) {
        const cached = this.state.get(device);
        return {
            name: device.name,
            ieee_address: device.ieeeAddr,
            model: device.definition?.model,
            smoke: cached.smoke,
            smoke_concentration: cached.smoke_concentration
        };
    }

    async start() {
        this.requestTopic =
[... 208 lines cut ...]
```

These are **Zigbee2MQTT names**, not Home Assistant entity IDs.

### Checking the result

For each reset, Z2M should log:

```text
Manually cleared cached smoke state: Detector 2 (...)
```

The extension also publishes a response to:

```text
zigbee2mqtt/smoke_cache_reset/response
```

That response includes the before/after values, or an error if something failed. A successful HA script execution by itself only confirms that HA sent the MQTT message, so check the response or the actual detector states.

Keep `retain: false` on commands. The timestamp rejects requests older than two minutes, and HA and the Z2M host need reasonably synchronized clocks.

There is no automatic timed clearing in this extension. This is just a manual way to acknowledge the stale readings after confirming the smoke is gone, without climbing up to the detector or restarting Z2M each time. Z2M handles saving the updated cache through its normal persistence mechanism.

Hopefully this helps anyone else stuck with these things. I'd still be interested in an actual converter/reporting fix if someone figures out why the clear state isn't coming through.