Koenkk/zigbee2mqtt #24959: Power on behavoiur "UNSUPPORTED_ATTRIBUTE" for Enbrighten 43076 devices

movieguy2001, 2024-11-29:
### What happened?

Setting the power on behavior in the device causes the following alerts:

z2m: Publish 'set' 'power_on_behavior' to 'Powder Room Fan' failed: 'Error: Got `UNSUPPORTED_ATTRIBUTE` error, device does not support power on behaviour'

### What did you expect to happen?

No error

### How to reproduce it (minimal and precise)

Attempt to change power on behavior for a 43076 device in the devices interface of Zigbee2MQTT

### Zigbee2MQTT version

1.41.0

### Adapter firmware version

20240316

### Adapter

SLZB-06P7

### Setup

Add-On on Home Assistant Blue

### Debug log

Info 2024-11-29 09:57:29z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/options', payload '{"data":{"restart_required":false},"status":"ok","transaction":"4o42t-2"}'
Error 2024-11-29 09:57:53z2m: Publish 'set' 'power_on_behavior' to 'Powder Room Fan' failed: 'Error: Got `UNSUPPORTED_ATTRIBUTE` error, device does not support power on behaviour'
Info 2024-11-29 09:57:53z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'power_on_behavior' to 'Powder Room Fan' failed: 'Error: Got `UNSUPPORTED_ATTRIBUTE` error, device does not support power on behaviour'","meta":{"friendly_name":"Powder Room Fan"},"type":"zigbee_publish_error"}'

deviantintegral, 2024-12-12:
Yep, this has been an issue for a long while, and I never bothered to file a bug report. Thank you!

jeffbl, 2025-01-26:
On the assumption that a fix for this might fix other Enbrighten (Jasco) devices, this exact issue also happens on my [43095 zigbee smart plug-in switch](https://www.zigbee2mqtt.io/devices/43095.html):

Zigbee2MQTT version 2.0.0-2
Adapter SLZB-06M
Home Assistant running in VM under Proxmox

```
error 2025-01-26 15:00:00z2m: Publish 'set' 'power_on_behavior' to 'Switch refrigerator' failed: 'Error: Got `UNSUPPORTED_ATTRIBUTE` error, device does not support power on behaviour'
```

closed 2025-04-04 by github-actions[bot]