Koenkk/zigbee2mqtt #24960: CC1352P2_CC2652P_launchpad_coordinator_20240710.zip contain hex file type 3

AndreasRM, 2024-11-29:
### What happened?

The flash file is written/compiled as "intel hex type 3". This version is not supported 
by TI Flash Programmer 2.

So could you be so kind and provide the new firmware in the former format?

### What did you expect to happen?

So could you be so kind and provide the new firmware in the former format so I could flash it
with TI Flash Programmer?

### Zigbee2MQTT version

1.41.0 

### Adapter firmware version

20220219

### Adapter

Sonoff ZBDongle-P

### Setup

Home Assistant

meks007, 2024-11-29:
https://zig-star.com/radio-docs/zigstar-multi-tool/
just use this. TI Flash Programmer has a bug that renders it useless.

AndreasRM, 2024-12-02:
Thanks for your tip - the tool is a bit unstable but works  :-)

closed 2025-02-08 by github-actions[bot]