Koenkk/zigbee2mqtt #24964: [New device support]: Tuya floor thermostat TS0601 _TZE200_spyvfeti

jjjchmielewski, 2024-11-29:
### Link

https://heatdecor.com/sklep/termostat-programowalny-hd-t1000-w-z-komunikacja-zigbee-3-0-bialy/

### Database entry

{"id":2,"type":"Router","ieeeAddr":"0x048727fffe55a804","nwkAddr":45424,"manufId":4098,"manufName":"_TZE200_spyvfeti","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\u0000\u0000\u0000\u0000\u0007\u0000\u0000\u0000\u0000\u0007\u0000\u0000\u0000\u0000\u0007\u0000\u0000\u0000\u0000\u0007","65506":31,"65508":0,"modelId":"TS0601","manufacturerName":"_TZE200_spyvfeti","powerSource":1,"zclVersion":3,"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1732919709750}

### Zigbee2MQTT version

1.41.0-1

### Comments

I tried to force this device to use config for this one already supported https://www.zigbee2mqtt.io/devices/TS0601_floor_thermostat.html
Unfortunately still described as not supported. External definition also failed due to problem with data points mapping. 

### External definition

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const ea = exposes.access;

module.exports = {
    zigbeeModel: ['_TZE200_spyvfeti'],  // Zamień '_TZE200_abcdefgh' na odpowiedni model twojego urządzenia
    model: 'TS0601',
    vendor: 'Tuya',
    description: 'Electric underfloor heating thermostat',
    fromZigbee: [
        fz.thermostat,            // Konwerter odczytu danych
        fz.ignore_basic_report,   // Ignoruje niepotrzebne raporty
    ],
    toZigbee: [
        tz.thermostat_occupied_heating_setpoint,  // Ustawienie temperatury
        tz.thermostat_local_temperature,          // Odczyt bieżącej temperatury
        tz.thermostat_running_state,              // Status pracy
    ],
    exposes: [
        exposes.climate()
            .withLocalTemperature()
            .withSetpoint('occupied_heating_setpoint', 5, 35, 0.5)  // Zakres ustawienia temperatury: 5-35°C
            .withSystemMode(['off', 'heat', 'auto'])                // Tryby: wyłączony, ogrzewanie, automatyczny
            .withRunningState(['idle', 'heat']),                    // Stan pracy: bezczynność, ogrzewanie
    ],
};
```

### What does/doesn't work with the external definition?

I used GPT to help generate the external definition, but unfortunately nothing works

damo8353, 2025-01-02:
Try this:

const exposes = require('zigbee-herdsman-converters/lib/exposes');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['TS0601'],
    model: 'BHT-002-GCLZB',
    vendor: 'Moes',
    description: 'Smart knob thermostat',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    exposes: [
        e.binary('child_lock', ea.STATE_SET, 'ON', 'OFF').withLabel('Child lock'),
        e.binary('rgb_backlight', ea.STATE_SET, 'OFF', 'ON').withLabel('RGB backlight').withDescription('Default: Off'),
        e.binary('eco_mode', ea.STATE_SET, 'OFF', 'ON').withLabel('ECO mode NOT SET').withDescription('Default: Off'),
        e.numeric('eco_temperature', ea.STATE_SET)
            .withValueMin(5)
            .withValueMax(30)
            .withValueStep(1)
            .withUnit('°C')
            .withDescription('Max temperature in ECO mode. Default: 20'),
        e.binary('valve_state', ea.STATE, false, true).withLabel('Heating in process'),
        e.climate()
            .withSystemMode(['off', 'heat'], ea.STATE_SET)
            .withPreset(['manual', 'temporary manual', 'schedule', 'eco'])
            .withSetpoint('current_heating_setpoint', 5, 35, 0.5, ea.STATE_SET)
            .withLocalTemperature(ea.STATE)
            .withLocalTemperatureCalibration(-9, 9, 0.1, ea.STATE_SET)
            .withDescription('Default: -3'),
        e.numeric('deadzone_temperature', ea.STATE_SET)
            .withValueMin(1)
            .withValueMax(5)
            .withValueStep(1)
            .withUnit('°C')
            .withDescription('Hysteresis. Default: 1'),
        e.numeric('min_temperature', ea.STATE_SET)
            .withValueMin(5)
            .withValueMax(15)
            .withValueStep(1)
            .withUnit('°C')
            .withDescription('Default: 5'),
        e.numeric('max_temperature', ea.STATE_SET)
            .withValueMin(15)
            .withValueMax(45)
            .withValueStep(1)
            .withUnit('°C')
            .withDescription('Default: 35'),
        e.numeric('min_temperature_limit', ea.STATE_SET)
            .withValueMin(0)
            .withValueMax(10)
            .withValueStep(1)
            .withUnit('°C')
            .withLabel('Low temperature protection')
            .withDescription('Default: 0'),
        e.numeric('lower_limit_of_temperature', ea.STATE_SET)
            .withValueMin(5)
            .withValueMax(15)
            .withValueStep(1)
            .withUnit('°C')
            .withLabel('The lower limit of temperature')
            .withDescription('Default: 5'),
        e.enum('sensor', ea.STATE_SET, ['IN', 'OU', 'AL']).withLabel('Sensor')
            .withDescription('Choose which sensor to use. Default: AL'),
        e.numeric('external_temperature_input', ea.STATE)
            .withLabel('Floor temperature')
            .withUnit('°C')
            .withDescription('Temperature from floor sensor'),
[... 50 lines cut ...]
            [102, 'schedule_monday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(1)],
            [103, 'schedule_tuesday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(2)],
            [104, 'schedule_wednesday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(3)],
            [105, 'schedule_thursday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(4)],
            [106, 'schedule_friday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(5)],
            [107, 'schedule_saturday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(6)],
            [108, 'schedule_sunday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(7)],
            [109, 'external_temperature_input', tuya.valueConverter.divideBy10],
            [110, 'deadzone_temperature', tuya.valueConverter.raw],
            [111, 'max_temperature_limit', tuya.valueConverter.divideBy10],
            [112, 'min_temperature_limit', tuya.valueConverter.divideBy10],
            [113, 'eco_temperature', tuya.valueConverter.divideBy10],
            [114, 'screen_time_set', tuya.valueConverterBasic.lookup({
                off: tuya.enum(0), 
                '10s': tuya.enum(1), 
                '30s': tuya.enum(2), 
                '60s': tuya.enum(3)
            })],
            [115, 'rgb_backlight', tuya.valueConverterBasic.lookup({ON: true, OFF: false})],
        ],
    },
    onEvent: tuya.onEventSetLocalTime,
};

module.exports = definition;

jjjchmielewski, 2025-01-21:
@damo8353 I've tested it and it doesn't function as intended. Most of the exposed parameters are mismatched, for example set temperature is shown as current temp, and floor temp is showing some weir value like 0.6 Celcius. So something in configuration must not be right.

jjjchmielewski, 2025-03-23:
Eventually I've bought zigbee gateway to use the thermostats. I'll try to find some time to use it to debug tuya codes as in tutorial