Koenkk/zigbee2mqtt #24965: Phillips Hue Twilight - resulting with "Not Supported"  (Z2MQTT compatability states it is supported)

wardwolfram, 2024-11-30:
### What happened?

I purchased the Phillips Hue Twilight lamp as it was compatible with Z2MQTT.

HA, Z2MQTT and the Twilight lamp are all up to date with FW.

After repeatedly pairing the Twilight, HA/Z2MQTT states "Not Supported"

The Twilight that I have identifies as:
Model: LGT003

The Z2MQTT compatibility website states the model supported = 929003711201
The box the Twilight came in has an unlabeled similar number of = 929003711401

Pairing is successful, but I have no control of any feature of the lamp.

Let me know how I can help.

### What did you expect to happen?

Ability to control the Twilight via Z2MQTT

### How to reproduce it (minimal and precise)

100% of the time each time that I tried to pair.

### Zigbee2MQTT version

1.41.0-1

### Adapter firmware version

6.10.3.0 build 297

### Adapter

Sonoff 3.0 USB Dongle Plus - EZSP v8 6.10.3.0 build 297

### Setup

Home Assistant on a NUC

### Debug log

[log2.log](https://github.com/user-attachments/files/17963606/log2.log)

I am not sure how to 'refresh' logs, etc. so the error codes are captured.   Hope this is the correct one. 

[45df7312_zigbee2mqtt_2024-11-30T00-24-30.993Z.log](https://github.com/user-attachments/files/17963614/45df7312_zigbee2mqtt_2024-11-30T00-24-30.993Z.log)

Koenkk, 2024-11-30:
Added! will be supported in tomorrows release.

closed 2024-11-30 by Koenkk

wardwolfram, 2024-12-01:
Hello Koenkk,  I updated Z2MQTT this morning and repaired the Twilight.  I continue to get the status of 'Unsupported'. Is their any other information that I can help provide?

[image]

wardwolfram, 2024-12-02:
Is their anything I can help with to get the Hue Twilight light supported?  Let me know. Thanks!

Koenkk, 2024-12-02:
Whoops, found the issue and pushed a fix, will be included in the next release (1 Jan)

wardwolfram, 2024-12-03:
Thanks for looking into this.