Koenkk/zigbee2mqtt #24967: [Wrong device]: NOUS A6Z is identified as Tuya TS011F_plug_1

echicor, 2024-11-30:
### Link

https://nous.technology/product/a6z.html

### Model

A6Z

### Description

Outdoor Smart Socket

### Vendor

NOUS

### Picture (link)

https://nous.technology/share/resize/thumbs/448x495/A6Z.jpg

### Database entry

{"id":67,"type":"Router","ieeeAddr":"0xa4c138b360f590f9","nwkAddr":57738,"manufId":4417,"manufName":"_TZ3000_266azbg3","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[3,4,5,6,1794,2820,57344,57345,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"C��.\u0003","65506":54,"65508":0,"65534":0,"modelId":"TS011F","manufacturerName":"_TZ3000_266azbg3","powerSource":1,"zclVersion":3,"appVersion":192,"stackVersion":0,"hwVersion":1,"dateCode":""}},"haElectricalMeasurement":{"attributes":{"acCurrentDivisor":1000,"acCurrentMultiplier":1,"rmsVoltage":230,"rmsCurrent":0,"activePower":0}},"seMetering":{"attributes":{"divisor":100,"multiplier":1,"currentSummDelivered":112}},"genOnOff":{"attributes":{"32768":0,"onOff":0,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":1,"moesStartUpOnOff":2,"tuyaBacklightSwitch":1}},"manuSpecificTuya_3":{"attributes":{"53248":1,"53249":0,"53250":2928329,"53251":125,"53252":10338,"53253":1229,"powerOnBehavior":2,"switchType":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b00259087ef","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00124b00259087ef","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x00124b00259087ef","endpointID":1}],"configuredReportings":[{"cluster":2820,"attrId":1285,"minRepIntval":5,"maxRepIntval":3600,"repChange":5},{"cluster":2820,"attrId":1288,"minRepIntval":5,"maxRepIntval":3600,"repChange":50},{"cluster":2820,"attrId":1291,"minRepIntval":5,"maxRepIntval":3600,"repChange":10},{"cluster":1794,"attrId":0,"minRepIntval":5,"maxRepIntval":3600,"repChange":257}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":192,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1732959756727}

Koenkk, 2024-12-01:
added!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) tomorrow.

Could you also submit a picture for the docs? This can be done by:
1. Create a fork by clicking [here](https://github.com/Koenkk/zigbee2mqtt.io/fork)
2. Go to the `public/images/devices` directory, *Add file* -> *Upload files*
3. Upload the files and press *Commit changes*
4. Press *Contribute* -> *Open pull request* -> update title/description -> *Create pull request*

Make sure that:
- The filename is `A6Z.png`
- The size is 512x512
- The background is transparent (use e.g. [Adobe remove background](https://new.express.adobe.com/tools/remove-background))

closed 2024-12-01 by Koenkk

imLeeat, 2025-01-07:
Same problem with Nous D2Z. it is identified as Tuya TS01F-plug_3

bummzulu, 2025-02-19:
Same problem here:

	•	Zigbee2MQTT Version: 2.1.1 (commit #1accb8b66b95b99658c50e4c0209edcb946a5fe2)
	•	zigbee-herdsman Version: 3.2.5
	•	Coordinator Adapter:
	•	Firmware version: {"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20221226,"transportrev":2},"type":"zStack3x0"}

AlexanderLeporiday, 2025-02-20:
> Same problem with Nous D2Z. it is identified as Tuya TS01F-plug_3

I've just had two pull requests approved that will resolve the issue for the Nous D2Z.
It'll give 'enough' functionality for general use as the icon will display correctly.
These changes aren't fully live yet so look for them in the next release.