Koenkk/zigbee2mqtt #24968: [Wrong device]:

echicor, 2024-11-30:
### Link

https://de.paulmann.com/p/standard-230v-smart-home-zigbee-3.0-led-reflektor-gu10-350lm-4-8w-rgbw-dimmbar-chrom-matt/29149

### Model

152561

### Description

Standard 230V Smart Home Zigbee 3.0 LED Reflektor GU10 350lm 4,8W RGBW+ dimmbar

### Vendor

Paulmann

### Picture (link)

https://content.paulmann.com/thumbnail/7c/95/64/1694182507/29149fr01_1160x1160.jpg

### Database entry

{"id":61,"type":"Router","ieeeAddr":"0xa4c13800ac4b258c","nwkAddr":50921,"manufId":4632,"manufName":"Paulmann Licht GmbH","powerSource":"Mains (single phase)","modelId":"RGBWW","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[0,3,4,5,6,8,768,4096,4111],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"RGBWW","manufacturerName":"Paulmann Licht GmbH","powerSource":1,"zclVersion":3,"appVersion":0,"stackVersion":2,"hwVersion":112,"swBuildId":"PIIC5401"}},"lightingColorCtrl":{"attributes":{"colorCapabilities":31,"colorTempPhysicalMin":153,"colorTempPhysicalMax":454,"currentHue":243,"colorMode":2,"currentSaturation":210,"colorTemperature":333}},"genOnOff":{"attributes":{"onOff":1}},"genLevelCtrl":{"attributes":{"currentLevel":84}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":0,"stackVersion":2,"hwVersion":112,"swBuildId":"PIIC5401","zclVersion":3,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1732959718096}

dr-waterstorm, 2024-12-05:
I bought the same model and have the same issue, so yesterday I tried adding this devices to the database so it would show up as a separate device.
Therefore I created an external definition as suggested in the docs for adding a new devices. This worked out well in general. However I'm not able to control the `color_temp` on my model.

I have the 291.48 "RGBWW" as well. The my modified definition is:

```
const {light} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    fingerprint: [
        {modelID: 'RGBWW', manufacturerName: 'Paulmann Licht GmbH'},
    ],
    zigbeeModel: ['RGBWW'],
    model: '291.48',
    vendor: 'Paulmann',
    description: 'GU10 LED RGBW 350 lm',
    extend: [
        light({
            "colorTemp":{"range":[153,454]},
            "color":{
                "modes":["xy","hs"],
                "enhancedHue":true
            }
        })
    ],
    meta: {},
};

module.exports = definition;
```

@echicor does changing the color temperature work for you? I can click it in the interface but it only changes the brightness. Even though the package says "2200 - 6500K" I cannot seem to get that working.
Changing the color itself and the brightness works just fine. 

Anyone has any ideas how to fix the color temperature issues?

echicor, 2025-02-04:
i can change the color temperature ... homeassistant change this for a group of this lights with an automation while the lights are on

closed 2025-04-13 by github-actions[bot]