Koenkk/zigbee2mqtt #24973: [New device support]: Add support for Hive SLR2d

mwhapples, 2024-11-30:
### Link

https://www.hivehome.com/guides/hive-heating

### Database entry

{"id":9,"type":"Router","ieeeAddr":"0x001e5e0902a62be5","nwkAddr":51122,"manufId":4153,"manufName":"Computime","powerSource":"Mains (single phase)","modelId":"SLR2d","epList":[5,6,242],"endpoints":{"5":{"profId":260,"epId":5,"devId":769,"inClusterList":[0,3,9,10,513,64768],"outClusterList":[3,25,1026],"clusters":{"genBasic":{"attributes":{"modelId":"SLR2d","manufacturerName":"Computime","powerSource":1,"zclVersion":8,"appVersion":3,"stackVersion":2,"hwVersion":1,"dateCode":"20220819"}},"hvacThermostat":{"attributes":{"localTemp":2102,"tempSetpointHold":1,"tempSetpointHoldDuration":65535,"systemMode":4,"occupiedHeatingSetpoint":650,"runningState":0}}},"binds":[{"cluster":0,"type":"endpoint","deviceIeeeAddress":"0x8c65a3fffe548e38","endpointID":1},{"cluster":3,"type":"endpoint","deviceIeeeAddress":"0x8c65a3fffe548e38","endpointID":1},{"cluster":9,"type":"endpoint","deviceIeeeAddress":"0x8c65a3fffe548e38","endpointID":1},{"cluster":10,"type":"endpoint","deviceIeeeAddress":"0x8c65a3fffe548e38","endpointID":1},{"cluster":513,"type":"endpoint","deviceIeeeAddress":"0x8c65a3fffe548e38","endpointID":1}],"configuredReportings":[{"cluster":513,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":10},{"cluster":513,"attrId":41,"minRepIntval":0,"maxRepIntval":3600,"repChange":0},{"cluster":513,"attrId":28,"minRepIntval":10,"maxRepIntval":3600,"repChange":null},{"cluster":513,"attrId":18,"minRepIntval":0,"maxRepIntval":3600,"repChange":10},{"cluster":513,"attrId":35,"minRepIntval":0,"maxRepIntval":3600,"repChange":0},{"cluster":513,"attrId":36,"minRepIntval":0,"maxRepIntval":3600,"repChange":10}],"meta":{}},"6":{"profId":260,"epId":6,"devId":769,"inClusterList":[513],"outClusterList":[],"clusters":{"hvacThermostat":{"attributes":{"systemMode":5,"runningState":1,"tempSetpointHoldDuration":15,"tempSetpointHold":1}}},"binds":[{"cluster":0,"type":"endpoint","deviceIeeeAddress":"0x8c65a3fffe548e38","endpointID":1},{"cluster":3,"type":"endpoint","deviceIeeeAddress":"0x8c65a3fffe548e38","endpointID":1},{"cluster":9,"type":"endpoint","deviceIeeeAddress":"0x8c65a3fffe548e38","endpointID":1},{"cluster":10,"type":"endpoint","deviceIeeeAddress":"0x8c65a3fffe548e38","endpointID":1},{"cluster":513,"type":"endpoint","deviceIeeeAddress":"0x8c65a3fffe548e38","endpointID":1}],"configuredReportings":[{"cluster":513,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":10},{"cluster":513,"attrId":41,"minRepIntval":0,"maxRepIntval":3600,"repChange":0},{"cluster":513,"attrId":28,"minRepIntval":10,"maxRepIntval":3600,"repChange":null},{"cluster":513,"attrId":18,"minRepIntval":0,"maxRepIntval":3600,"repChange":10},{"cluster":513,"attrId":35,"minRepIntval":0,"maxRepIntval":3600,"repChange":0},{"cluster":513,"attrId":36,"minRepIntval":0,"maxRepIntval":3600,"repChange":10}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":3,"stackVersion":2,"hwVersion":1,"dateCode":"20220819","zclVersion":8,"interviewCompleted":true,"meta":{"configured":-399547537},"lastSeen":1732972982980}

### Zigbee2MQTT version

1.41.0 commit: cc31cea

### Comments

This seems to be another minor revision of the Hive thermostat recievers. I followed the guide for adding support for new devices and when using the definitions for the SLR2c it all seems to work. I am not sure how to go about modifying the original repository files to add this for a merge request.

### External definition

```shell
const {deviceEndpoints} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['SLR2d'],
    model: 'SLR2d',
    vendor: 'Computime',
    description: 'Automatically generated definition',
    extend: [deviceEndpoints({"endpoints":{"5":5,"6":6}})],
    meta: {"multiEndpoint":true},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

I based the definition on the SLR2c definition. All seems to work.

mwhapples, 2024-11-30:
The heating endpoint should also have emergency_heating in the list of system modes as it supports Hive's boost. So the new definition is like:
```
const {deviceEndpoints} = require('zigbee-herdsman-converters/lib/modernExtend');
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const ota = require('zigbee-herdsman-converters/lib/ota');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition =     {
    zigbeeModel: ['SLR2d'],
    model: 'SLR2d',
    vendor: 'Hive',
    description: 'Dual channel heating and hot water thermostat',
    fromZigbee: [fz.thermostat, fz.thermostat_weekly_schedule],
    toZigbee: [
        tz.thermostat_local_temperature,
        tz.thermostat_system_mode,
        tz.thermostat_running_state,
        tz.thermostat_occupied_heating_setpoint,
        tz.thermostat_control_sequence_of_operation,
        tz.thermostat_weekly_schedule,
        tz.thermostat_clear_weekly_schedule,
        tz.thermostat_temperature_setpoint_hold,
        tz.thermostat_temperature_setpoint_hold_duration,
    ],
    endpoint: (device) => {
        return {heat: 5, water: 6};
    },
    meta: {disableDefaultResponse: true, multiEndpoint: true},
    configure: async (device, coordinatorEndpoint) => {
        const heatEndpoint = device.getEndpoint(5);
        const waterEndpoint = device.getEndpoint(6);
        const binds = ['genBasic', 'genIdentify', 'genAlarms', 'genTime', 'hvacThermostat'];
        await reporting.bind(heatEndpoint, coordinatorEndpoint, binds);
        await reporting.thermostatTemperature(heatEndpoint);
        await reporting.thermostatRunningState(heatEndpoint);
        await reporting.thermostatSystemMode(heatEndpoint);
        await reporting.thermostatOccupiedHeatingSetpoint(heatEndpoint);
        await reporting.thermostatTemperatureSetpointHold(heatEndpoint);
        await reporting.thermostatTemperatureSetpointHoldDuration(heatEndpoint);
        await reporting.bind(waterEndpoint, coordinatorEndpoint, binds);
        await reporting.thermostatTemperature(waterEndpoint);
        await reporting.thermostatRunningState(waterEndpoint);
        await reporting.thermostatSystemMode(waterEndpoint);
        await reporting.thermostatOccupiedHeatingSetpoint(waterEndpoint);
        await reporting.thermostatTemperatureSetpointHold(waterEndpoint);
        await reporting.thermostatTemperatureSetpointHoldDuration(waterEndpoint);
    },
    exposes: [
        e
            .climate()
            .withSetpoint('occupied_heating_setpoint', 5, 32, 0.5)
            .withLocalTemperature()
            .withSystemMode(['off', 'auto', 'heat', 'emergency_heating'])
            .withRunningState(['idle', 'heat'])
            .withEndpoint('heat'),
        e
            .binary('temperature_setpoint_hold', ea.ALL, true, false)
            .withDescription(
                'Prevent changes. `false` = run normally. `true` = prevent from making changes.' +
                    ' Must be set to `false` when system_mode = off or `true` for heat',
            )
            .withEndpoint('heat'),
        e
            .numeric('temperature_setpoint_hold_duration', ea.ALL)
            .withValueMin(0)
            .withValueMax(65535)
            .withDescription(
                'Period in minutes for which the setpoint hold will be active. 65535 = attribute not' +
                    ' used. 0 to 360 to match the remote display',
            )
            .withEndpoint('heat'),
        e
            .climate()
            .withSetpoint('occupied_heating_setpoint', 22, 22, 1)
            .withLocalTemperature()
            .withSystemMode(['off', 'auto', 'heat', 'emergency_heating'])
            .withRunningState(['idle', 'heat'])
            .withEndpoint('water'),
        e
            .binary('temperature_setpoint_hold', ea.ALL, true, false)
            .withDescription(
                'Prevent changes. `false` = run normally. `true` = prevent from making changes.' +
                    ' Must be set to `false` when system_mode = off or `true` for heat',
            )
            .withEndpoint('water'),
        e
            .numeric('temperature_setpoint_hold_duration', ea.ALL)
            .withValueMin(0)
            .withValueMax(65535)
            .withDescription(
                'Period in minutes for which the setpoint hold will be active. 65535 = attribute not' +
                    ' used. 0 to 360 to match the remote display',
            )
            .withEndpoint('water'),
    ],
};

module.exports = definition;
```

mwhapples, 2025-01-30:
What will it take to get this in, after all I have posted the external definition needed. I just need some advice on how to go about making a suitable pull request. I am prepared to do the work if someone could just help me with that.

DanDon, 2025-01-30:
I'm having the issue where I am using the extension to allow z2m to recognise the device instead of saying upsupported but it will not use it?

The logs have the extension loaded correctly, with the correct model and vendor but when I join the device z2m says that the vendor and model are not supported, quoting the exact vendor and model I have in the extension??

DanDon, 2025-01-30:
It's okay. Got this to work in the end.

For anyone else. Just browse to your /homeassistant/zigbee2mqtt/external_converters folder (create the external_converters folder if you don't have it in there). Create a file in there. Call it what you want (like hive_slr2d.js). Then put this code in the file:

`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['SLR2d'],  // Ensure this matches the actual Zigbee model reported in Zigbee2MQTT
    model: 'SLR2d',
    vendor: 'Computime', // This must match Zigbee2MQTT logs
    description: 'Hive dual-channel heating and hot water thermostat',
    
    fromZigbee: [
        fz.thermostat, 
        fz.thermostat_weekly_schedule
    ],
    
    toZigbee: [
        tz.thermostat_local_temperature,
        tz.thermostat_system_mode,
        tz.thermostat_running_state,
        tz.thermostat_occupied_heating_setpoint,
        tz.thermostat_control_sequence_of_operation,
        tz.thermostat_weekly_schedule,
        tz.thermostat_clear_weekly_schedule,
        tz.thermostat_temperature_setpoint_hold,
        tz.thermostat_temperature_setpoint_hold_duration,
    ],

    endpoint: (device) => {
        return { heat: 5, water: 6 }; // Correct endpoints for heating & hot water
    },

    meta: { disableDefaultResponse: true, multiEndpoint: true },

    configure: async (device, coordinatorEndpoint) => {
        const heatEndpoint = device.getEndpoint(5);
        const waterEndpoint = device.getEndpoint(6);
        const binds = ['genBasic', 'genIdentify', 'genAlarms', 'genTime', 'hvacThermostat'];

        // Configure heating endpoint (5)
        await reporting.bind(heatEndpoint, coordinatorEndpoint, binds);
        await reporting.thermostatTemperature(heatEndpoint);
        await reporting.thermostatRunningState(heatEndpoint);
        await reporting.thermostatSystemMode(heatEndpoint);
        await reporting.thermostatOccupiedHeatingSetpoint(heatEndpoint);
        await reporting.thermostatTemperatureSetpointHold(heatEndpoint);
        await reporting.thermostatTemperatureSetpointHoldDuration(heatEndpoint);

        // Configure water endpoint (6)
        await reporting.bind(waterEndpoint, coordinatorEndpoint, binds);
        await reporting.thermostatTemperature(waterEndpoint);
        await reporting.thermostatRunningState(waterEndpoint);
        await reporting.thermostatSystemMode(waterEndpoint);
        await reporting.thermostatOccupiedHeatingSetpoint(waterEndpoint);
        await reporting.thermostatTemperatureSetpointHold(waterEndpoint);
        await reporting.thermostatTemperatureSetpointHoldDuration(waterEndpoint);
    },

    exposes: [
        // Heating Controls
        e.climate()
            .withSetpoint('occupied_heating_setpoint', 5, 32, 0.5)
            .withLocalTemperature()
            .withSystemMode(['off', 'auto', 'heat', 'emergency_heating'])
            .withRunningState(['idle', 'heat'])
            .withEndpoint('heat'),
        e.binary('temperature_setpoint_hold', ea.ALL, true, false)
            .withDescription('Prevent changes. Must be `false` when system_mode = off or `true` for heat')
            .withEndpoint('heat'),
        e.numeric('temperature_setpoint_hold_duration', ea.ALL)
            .withValueMin(0)
            .withValueMax(65535)
            .withDescription('Period in minutes for which the setpoint hold will be active (0-360 mins)')
            .withEndpoint('heat'),

        // Hot Water Controls
        e.climate()
            .withSetpoint('occupied_heating_setpoint', 22, 22, 1)
            .withLocalTemperature()
            .withSystemMode(['off', 'auto', 'heat', 'emergency_heating'])
            .withRunningState(['idle', 'heat'])
            .withEndpoint('water'),
        e.binary('temperature_setpoint_hold', ea.ALL, true, false)
            .withDescription('Prevent changes. Must be `false` when system_mode = off or `true` for heat')
            .withEndpoint('water'),
        e.numeric('temperature_setpoint_hold_duration', ea.ALL)
            .withValueMin(0)
            .withValueMax(65535)
            .withDescription('Period in minutes for which the setpoint hold will be active (0-360 mins)')
            .withEndpoint('water'),
    ],
};

module.exports = definition;
`

Then restart Z2M and re-add the device if needed but I didn't even need to re-add. The existing 'Unsupported device' just took the converter and now works.

ZackaryH8, 2025-02-06:
The works sort of okay, but the hot water function does not work in Home Assistant. It simply says "Unknown"

[See Here](https://gist.github.com/ZackaryH8/640346883950969b62312c365eacf8bf)

DanDon, 2025-02-13:
This is a better version of the code that works properly with the heating and hot water:

```
const {deviceEndpoints} = require('zigbee-herdsman-converters/lib/modernExtend');
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const ota = require('zigbee-herdsman-converters/lib/ota');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition =     {
    zigbeeModel: ['SLR2d'],
    model: 'SLR2d',
    vendor: 'Computime',
    description: 'Dual channel heating and hot water thermostat',
    fromZigbee: [fz.thermostat, fz.thermostat_weekly_schedule],
    toZigbee: [
        tz.thermostat_local_temperature,
        tz.thermostat_system_mode,
        tz.thermostat_running_state,
        tz.thermostat_occupied_heating_setpoint,
        tz.thermostat_control_sequence_of_operation,
        tz.thermostat_weekly_schedule,
        tz.thermostat_clear_weekly_schedule,
        tz.thermostat_temperature_setpoint_hold,
        tz.thermostat_temperature_setpoint_hold_duration,
    ],
    endpoint: (device) => {
        return {heat: 5, water: 6};
    },
    meta: {disableDefaultResponse: true, multiEndpoint: true},
    configure: async (device, coordinatorEndpoint) => {
        const heatEndpoint = device.getEndpoint(5);
        const waterEndpoint = device.getEndpoint(6);
        const binds = ['genBasic', 'genIdentify', 'genAlarms', 'genTime', 'hvacThermostat'];
        await reporting.bind(heatEndpoint, coordinatorEndpoint, binds);
        await reporting.thermostatTemperature(heatEndpoint);
        await reporting.thermostatRunningState(heatEndpoint);
        await reporting.thermostatSystemMode(heatEndpoint);
        await reporting.thermostatOccupiedHeatingSetpoint(heatEndpoint);
        await reporting.thermostatTemperatureSetpointHold(heatEndpoint);
        await reporting.thermostatTemperatureSetpointHoldDuration(heatEndpoint);
        await reporting.bind(waterEndpoint, coordinatorEndpoint, binds);
        await reporting.thermostatTemperature(waterEndpoint);
        await reporting.thermostatRunningState(waterEndpoint);
        await reporting.thermostatSystemMode(waterEndpoint);
        await reporting.thermostatOccupiedHeatingSetpoint(waterEndpoint);
        await reporting.thermostatTemperatureSetpointHold(waterEndpoint);
        await reporting.thermostatTemperatureSetpointHoldDuration(waterEndpoint);
    },
    exposes: [
        e
            .climate()
            .withSetpoint('occupied_heating_setpoint', 5, 32, 0.5)
            .withLocalTemperature()
            .withSystemMode(['off', 'auto', 'heat', 'emergency_heating'])
            .withRunningState(['idle', 'heat'])
            .withEndpoint('heat'),
        e
            .binary('temperature_setpoint_hold', ea.ALL, true, false)
            .withDescription(
                'Prevent changes. `false` = run normally. `true` = prevent from making changes.' +
                    ' Must be set to `false` when system_mode = off or `true` for heat',
            )
            .withEndpoint('heat'),
        e
[... 95 lines cut ...]
          - action: switch.turn_on
            metadata: {}
            data: {}
            target:
              entity_id: switch.heating_temperature_setpoint_hold_heat
          - action: climate.set_temperature
            metadata: {}
            data:
              temperature: "{{ states('input_number.temperature') | float }}"
            target:
              entity_id: climate.heating_heat
      - conditions:
          - condition: trigger
            id:
              - "4"
        sequence:
          - action: switch.turn_off
            metadata: {}
            data: {}
            target:
              entity_id: switch.heating_temperature_setpoint_hold_heat
mode: single
```

Be aware I have a tracking number helper so that when I turn the heating to 'heat' from 'off', it has remembered the last temperature and sets the heat to that

ZackaryH8, 2025-02-14:
This seems to work perfectly. Thank you for spending the time getting this to work. I think a PR is in order to get this officially supported. :)

samtate, 2025-02-17:
Thanks for this @DanDon . I've got the entities working in Home Assistant now. I'm just trying to work out if it's possible to control the heating and hot water from Home Assistant while keeping the schedules and boosts?

Ideally i'd be able to use the built in schedule for heat and hot water on the thermostat itself, with the option of having HA override it (for example setting a lower temp when we go out), then moving back to the schedule when we're back? Or manually calling a boost for heating or hot water from HA. 

But I can't figure out which values to set to what for any of this to work?

ZackaryH8, 2025-03-25:
[image]
I handled my boosts manually, they are pre-set to an hour at a specific temperature but it works great. (Yes slightly off topic but there is no way to have boosts through ZigBee from what I know)

samtate, 2025-03-25:
https://github.com/andrew-codechimp/HA-Hive-Local-Thermostat

I found this addon that takes the entity that Zigbee2MQTT makes above and gives you full control of your hive including setting it back to schedule, heat and hot water boosts etc. Only thing it can't do is actually modify the schedule as that's only set on the thermostat itself.

mwhapples, 2025-05-25:
I really don't understand what the problem is here in getting this device added. After all there is a definition provided in this ticket, I have offered to do the work if someone could explain what needs to be done to create suitable pull requests, yet nothing seems to be happening. It would be good to get a clarification on why nothing is happening. @Koe as this project is in your user account on GitHub and thus owner, may be you can provide the clarification.

closed 2025-12-05 by github-actions[bot]