Koenkk/zigbee2mqtt #24974: Sonoff ZBMINI R2 Detached relay, External trigger mode, and switch action

thomasvnl, 2024-11-30:
### What happened?

As documented, the ZBMINI R2 exposes the switch action as "toggle" and nothing more. I believe however that it would make more sense for automations if the switch state could be tracked to on/off instead of toggle, especially when the device is in both detached relay mode and the external trigger mode is set to either of the following settings. 

### What did you expect to happen?

I expected the action to return on/off instead of toggle as in detached relay mode and external trigger mode = following(on). For example, in a case where you have this device set to detached relay mode because you want to be able to keep a light with (external) motion detector on during 20:00 and 22:00, and outside these hours trigger based on the motion detector wired onto S1/S2, the "toggle" action would trigger the device on both motion seen and motion lost. But, as the device is in detached relay mode the automations behind this have no idea when the motion = "on". If something else were to set the device to on in the mean time, the external trigger mode would not following(on) and the device could be turned off instead (based on toggle behavior).

### How to reproduce it (minimal and precise)

Configure the ZB Mini R2 as detached relay = true, external trigger mode = following(on). Observe action state (via MQTT) only showing 'toggle' state when flipping switch. Therefor "external trigger mode = following(on)" not guaranteed i.e. not able to follow. 

### Zigbee2MQTT version

1.41.0

### Adapter firmware version

20220219

### Adapter

Sonoff ZB Dongle-P

### Setup

Barebone service running on Ubuntu on RPi

coxtor, 2024-12-03:
In addition to this it does not appear to parse the toggle command correctly, in detached mode the switch does nothing. Debug:

`[2024-12-03 14:14:44] debug:    z2m: No converter available for 'ZBMINIR2' with cluster 'genOnOff' and type 'commandToggle' and data '{}'`

closed 2025-02-09 by github-actions[bot]

robertklep, 2025-02-12:
+1 for having some sort of differentiation possibility.

I have exactly the same problem as @thomasvnl, where I switch the ZBMINIR2 (in detached relay mode) with a mains powered PIR sensor, and just having a "toggle" action is not sufficient.

alexsotoaguilera, 2025-11-21:
> blueprint

That's my blueprint :) It works by hunting the ‘action’ trigger and sending a toggle to the controlled light.