Koenkk/zigbee2mqtt #24982: [New device support]: IH-K009 _TZ3000_8ybe88nf

Namikson, 2024-12-01:
### Link

https://vi.aliexpress.com/item/1005003187008387.html

### Database entry

{"id":3,"type":"EndDevice","ieeeAddr":"0xa4c1380fe8d78e41","nwkAddr":11676,"manufId":4417,"manufName":"_TZ3000_akqdg6g7","powerSource":"Battery","modelId":"TS0201","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":770,"inClusterList":[1,3,1026,1029,0],"outClusterList":[3,25,10],"clusters":{"msTemperatureMeasurement":{"attributes":{"measuredValue":1963}},"msRelativeHumidity":{"attributes":{"measuredValue":3333}},"genBasic":{"attributes":{"65503":"j��.\u0012","65506":51,"65508":0,"65534":0,"stackVersion":0,"dateCode":"","manufacturerName":"_TZ3000_akqdg6g7","zclVersion":3,"appVersion":70,"modelId":"TS0201","powerSource":3}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200,"batteryVoltage":29}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":217189513},"lastSeen":1733023405106}

### Zigbee2MQTT version

1.41.0

### Comments

In Z2M and in Homeassistant it is recognized as:
Zigbee Model: TS0201
Zigbee Manufacturer: _TZ3000_8ybe88nf

Must be model: IH-K009

I'll submit a PR

### External definition

```shell
const {battery, temperature, humidity} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['TS0201'],
    model: 'TS0201',
    vendor: '_TZ3000_akqdg6g7',
    description: 'Automatically generated definition',
    extend: [battery(), temperature(), humidity()],
    meta: {},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

Works fine with temp converter

```
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const e = exposes.presets;

const fzLocal = {
    TS0201_battery: {
        cluster: 'genPowerCfg',
        type: ['attributeReport', 'readResponse'],
        convert: (model, msg, publish, options, meta) => {
            // https://github.com/Koenkk/zigbee2mqtt/issues/11470
            if (msg.data.batteryPercentageRemaining == 200 && msg.data.batteryVoltage < 30) return;
            return fz.battery.convert(model, msg, publish, options, meta);
        },
    },
}

module.exports = [
    {
        fingerprint: [
            {modelID: 'TS0201', manufacturerName: '_TZ3000_akqdg6g7'},
        ],
        model: 'IH-K009',
        vendor: 'Tuya',
        description: 'Temperature & humidity sensor',
        fromZigbee: [fzLocal.TS0201_battery, fz.temperature, fz.humidity],
        toZigbee: [],
        exposes: [e.battery(), e.temperature(), e.humidity(), e.battery_voltage()],
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await endpoint.read('genBasic', ['manufacturerName', 'zclVersion', 'appVersion', 'modelId', 'powerSource', 0xfffe]);
        },
    },
]
```

Namikson, 2024-12-01:
Merged https://github.com/Koenkk/zigbee-herdsman-converters/pull/8407

closed 2024-12-01 by Namikson

sslavtchev, 2025-03-07:
Hi, i have bought recently two sensors reporting same id _TZ3000_akqdg6g7 
but are actually https://www.zigbee2mqtt.io/devices/TH02Z.html#tuya-th02z by the externals looks :-)
is there a way to recognize them as such? air humidity is way off currently

here is the link https://de.aliexpress.com/item/1005008076325713.html?spm=a2g0o.order_detail.order_detail_item.2.665663682QUGg1&gatewayAdapt=glo2deu