Koenkk/zigbee2mqtt #24985: Ikea PARASOLL not reporting contact status

Wallmeier, 2024-12-01:
### What happened?

I have been using an SLZB-06P7 coordinator with 3 Parasoll sensors for some time now. Yesterday I bought two new Parasolls sensors. The first one does not respond to the pairing button at all - I suspect it is hardware problem with that specific sensor. The other sensor can be paired with the coordinator, but does not report the contact status correctly - it is always closed.

I can see a big difference in bind section of the new sensor compared to the existing sensors:
[image]

Example of an existing one:
[image]

### What did you expect to happen?

Transmission of correct contact status

### How to reproduce it (minimal and precise)

It is sufficient to pair a new Parasoll sensor and to open / close the window.

### Zigbee2MQTT version

1.41.0

### Adapter firmware version

20240716

### Adapter

SLZB-06P7

### Setup

Z2M plain on a Raspberry Pi 4

### Debug log

I've deleted the device, changed the log level to Debug and repaired the new Parasoll sensor: [log.log](https://github.com/user-attachments/files/17968860/log.log)
Result is still the same...

Wallmeier, 2024-12-01:
I've just update my Zigbee2Mqtt installation to the new 1.42.0 release - situation did not change (even after deleting and repairing)...

Chartreusito, 2024-12-02:
There is a complete issue about parasoll already ;) https://github.com/Koenkk/zigbee2mqtt/issues/22579

Wallmeier, 2025-01-02:
The problem is due to the installed firmware version on the Parasoll sensor which is 1.0.19 (build 20230406). The working ones all have 1.0.19 (build 20230516).

closed 2025-01-02 by Wallmeier

AdamJohnSwan, 2025-03-28:
@Wallmeier  where did you find the image for this build?

Wallmeier, 2025-03-28:
I did not find the image for the working build. I exchanged the faulty Parasoll at Ikea.