Koenkk/zigbee2mqtt #24986: [Wrong device]: SNZB-03 identified as eWeLink Water leak sensor (SQ510A)

arnonm, 2024-12-01:
### Link

https://www.aliexpress.com/item/1005007792275263.html

### Model

SNZB-03 by report, SZ-P1 printed on box

### Description

Minir PIR Motion Sensor 

### Vendor

Shenzen New Green Energy Technologies

### Picture (link)

https://www.aliexpress.com/item/1005007792275263.html

### Database entry

{"id":2,"type":"EndDevice","ieeeAddr":"0xa4c138db09b2d50c","nwkAddr":63729,"manufId":4742,"manufName":"eWeLink","powerSource":"Battery","modelId":"SNZB-03","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,3,1,1280,32],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"hwVersion":0}},"ssIasZone":{"attributes":{"iasCieAddr":"0x30fb10fffe0409f8","zoneState":1}},"genPollCtrl":{"attributes":{"checkinInterval":14400}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":146}}},"binds":[{"cluster":32,"type":"endpoint","deviceIeeeAddress":"0x30fb10fffe0409f8","endpointID":1},{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x30fb10fffe0409f8","endpointID":1}],"configuredReportings":[{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":65000,"repChange":10}],"meta":{}}},"appVersion":113,"stackVersion":2,"hwVersion":0,"swBuildId":"0122052017","zclVersion":3,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1733054804583,"checkinInterval":3600}

arnonm, 2024-12-01:
Generated code:
const {battery, iasZoneAlarm} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['SNZB-03'],
    model: 'SNZB-03',
    vendor: 'eWeLink',
    description: 'Automatically generated definition',
    extend: [battery(), iasZoneAlarm({"zoneType":"generic","zoneAttributes":["alarm_1","alarm_2","tamper","battery_low"]})],
    meta: {},
};

module.exports = definition;

arnonm, 2024-12-02:
Resolved by https://github.com/Koenkk/zigbee-herdsman-converters/pull/8385

awschallenger, 2024-12-02:
can you also read the illuminance level?

arnonm, 2024-12-02:
{
    "battery": 73,
    "battery_low": false,
    "linkquality": 96,
    "occupancy": false,
    "tamper": false,
    "water_leak": false
}

arnonm, 2024-12-02:
Device is reported correctly, but reporting fields are incorrect:
{
    "battery": 73,
    "battery_low": false,
    "linkquality": 96,
    "occupancy": false,
    "tamper": false,
    "water_leak": false
}

closed 2025-02-08 by github-actions[bot]