Koenkk/zigbee2mqtt #24990: [Feature request]: Add default manufacturer code option for ember adapter.

Kirmas, 2024-12-01:
### Is your feature request related to a problem? Please describe

Currently, we set the manufacturer code in the code as follows:

```
const DEFAULT_MANUFACTURER_CODE = Zcl.ManufacturerCode.SILICON_LABORATORIES;
```
This works logically for most devices. However, some devices like the Aqara D1 require a different manufacturer code:

```
Zcl.ManufacturerCode.LUMI_UNITED_TECHOLOGY_LTD_SHENZHEN
```
There is a relevant discussion about this issue on GitHub: Aqara D1(https://github.com/Koenkk/zigbee2mqtt/issues/19719).

The system has a workaround in place as shown in this [line of code](https://github.com/Koenkk/zigbee-herdsman/blob/9d3546278e2c5e95a3972842e6726b97329fd031/src/adapter/ember/adapter/emberAdapter.ts#L645), However, this workaround sometimes stops working when the number of devices in the network increases, which might be around 50 or more. In my case, I have about 100 devices, but the issue could also be related to a specific device in the network.

I tried changing the default manufacturer code to:
```
const DEFAULT_MANUFACTURER_CODE = Zcl.ManufacturerCode.LUMI_UNITED_TECHOLOGY_LTD_SHENZHEN;
```
This solution worked with 100 devices but caused issues with library updates, which isn't ideal.

### Describe the solution you'd like

I believe having the option to change the default coordinator manufacturer code could serve as a workaround for people experiencing the issue mentioned above. Alternatively, a more ideal solution could be a dropdown menu near the Permit Join 
[image]
menu, allowing users to select different coordinator manufacturers.

I'm not entirely sure if this issue affects other devices or if it's specific to Aqara devices.

### Describe alternatives you've considered

Tried to figuraut why https://github.com/Koenkk/zigbee-herdsman/blob/9d3546278e2c5e95a3972842e6726b97329fd031/src/adapter/ember/adapter/emberAdapter.ts#L645 not always working. But failed.

### Additional context

I understand that this is a complex feature request, and it provides an exception for the Ember protocol (an option that works only with Ember). When I tried figuring out how to implement this on my own, I failed due to a lack of knowledge and skills. However, if someone can do this, please proceed. I believe it will help in situations where a device checks the coordinator's manufacturer.

Best regards, Ihor

Nerivec, 2024-12-02:
Implementing this as a new API is not suitable since it would only be for ember, and require changes in frontend, z2m, etc....

The workaround for Aqara is triggered by a join, so, the number of devices shouldn't matter.
Can you provide a debug log of this happening?

Kirmas, 2025-01-04:
@Nerivec Sorry for the delay. I just tried the new version, but it's still failing. However, I noticed something strange: I don't see the "[WORKAROUND] Setting coordinator manufacturer code to" log at all. My log file:
[45df7312_zigbee2mqtt_2025-01-04T20-20-02.273Z.log](https://github.com/user-attachments/files/18308169/45df7312_zigbee2mqtt_2025-01-04T20-20-02.273Z.log) You can find my device trying to join in the log by its ID "0x04cf8cdf3c7bad25". I really don't understand why it works like this. When I started to build this network, I started with this switch, and it worked fine at the beginning. However, after a week or two, it got disconnected and hasn't connected properly since then.

Thanks for your help!

closed 2025-03-09 by Koenkk