Koenkk/zigbee2mqtt #25000: Unable to start with version 1.42 under docker

mirkochip88, 2024-12-01:
### What happened?

With version 1.42 under docker, the instance does not start. I attach the log. If I go back to version 1.41, it works normally.
I have another instance that I use with the official home assistant addon, which I managed to migrate easily, same coordinator, same firmware and same settings.

### Zigbee2MQTT version

1.42.0

### Adapter firmware version

7.4.1 [GA]

### Adapter

SLZB-06M

### Setup

Docker container in a Proxmox VM, EmberZNet

### Debug log

Starting Zigbee2MQTT with watchdog (2000,60000,300000,900000,1800000,3600000).
/app/node_modules/express-static-gzip/index.js:37
      res.status(400).send(e.message);
          ^
TypeError: res.status is not a function
    at Frontend.expressStaticGzip [as fileServer] (/app/node_modules/express-static-gzip/index.js:37:11)
    at Frontend.onRequest (/app/lib/extension/frontend.ts:144:14)
    at Server.emit (node:events:517:28)
    at parserOnIncoming (node:_http_server:1131:12)
    at HTTPParser.parserOnHeadersComplete (node:_http_common:119:17)
Using '/app/data' as data directory
Starting Zigbee2MQTT with watchdog (2000,60000,300000,900000,1800000,3600000).

mirkochip88, 2024-12-06:
Solved with firmware 2.6.8 on the SLZB-06M

Koenkk, 2024-12-09:
Could you provide the debug log of this?

See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

mirkochip88, 2024-12-10:
With the latest build everything works, both the zigbee network and the frontend. But I had to reconfigure both the frontend section and the zigbee2mqtt proxy addon to make the latter work.

mirkochip88, 2024-12-10:
With the latest build everything works, both the zigbee network and the frontend. But the zigbee2mqtt proxy addon fails to show the frontend, returning error 502. I tried reconfiguring the frontend section with the token and the rest. This appears in the log of z2m:

`[2024-12-10 12:04:19] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2docker/Presa contatori', payload '{"current":0,"energy":0.47,"indicator_mode":"on_off","last_seen":"2024-12-10T11:04:19.145Z","linkquality":104,"power":0,"power_outage_memory":"restore","state":"OFF","update":{"installed_version":69,"latest_version":69,"state":"idle"},"update_available":null,"voltage":238}'
/app/node_modules/express-static-gzip/index.js:37
      res.status(400).send(e.message);
          ^
TypeError: res.status is not a function
    at Frontend.expressStaticGzip [as fileServer] (/app/node_modules/express-static-gzip/index.js:37:11)
    at Frontend.onRequest (/app/lib/extension/frontend.ts:144:14)
    at Server.emit (node:events:517:28)
    at parserOnIncoming (node:_http_server:1131:12)
    at HTTPParser.parserOnHeadersComplete (node:_http_common:119:17)
Using '/app/data' as data directory
Starting Zigbee2MQTT with watchdog (2000,60000,300000,900000,1800000,3600000).
[2024-12-10 12:04:23] info: 	z2m: Logging to console, file (filename: log.log)
[2024-12-10 12:04:23] info: 	z2m: Starting Zigbee2MQTT version 1.42.0 (commit #861cba6)
[2024-12-10 12:04:23] info: 	z2m: Starting zigbee-herdsman (2.1.9)`

and this on z2mproxy:

`2024/12/10 12:04:20 [error] 14#14: *3 upstream prematurely closed connection while reading response header from upstream, client: 172.30.32.2, server: _, request: "GET / HTTP/1.1", upstream: "http://192.168.0.143:9080/?&token=!qknFz!dVzm00mx%b6VyOf#wD@q", host: "home.XXXX.XX", referrer: "https://home.XXXXX.XX/45df7312_zigbee2mqtt_proxy"`

It seems related to the previous problem

mirkochip88, 2024-12-10:
Solved. I had to change Auth token to one without special characters.

closed 2024-12-10 by Koenkk