Koenkk/zigbee2mqtt #25002: Ember Adapter Stopped during an OTA update

dragonlost, 2024-12-01:
### What happened?

When I try to update a device via OTA updates. My adapter, after a certain time, crashes and reboots.

### What did you expect to happen?

I would like to succeed in updating my devices. But impossible always this bug before the end.

### How to reproduce it (minimal and precise)

I launch the update from the OTA Update tab on a connected socket TS011F_plug_1 or IKEA LED2201G8 or LED2103G5 or E2206.
Still the same problem. It does not always happen at the same update percentage level.

### Zigbee2MQTT version

1.42.0

### Adapter firmware version

7.4.4 [GA]

### Adapter

EmberZNet (Sonoff Zigbee Dongle plus)

### Setup

Add on on Home Assitant OS on Virtual machine on Open Media Vault on N100 Computeur

### Debug log

error 2024-12-01 22:06:06zh:ember:uart:ash: Received ERROR from adapter, with code=ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT.
error 2024-12-01 22:06:06zh:ember:uart:ash: ASH disconnected | Adapter status: ASH_NCP_FATAL_ERROR
error 2024-12-01 22:06:06zh:ember:uart:ash: Error while parsing received frame, status=ASH_NCP_FATAL_ERROR.
error 2024-12-01 22:06:06zh:ember: Adapter fatal error: HOST_FATAL_ERROR
info 2024-12-01 22:06:06zh:ember:uart:ash: ASH COUNTERS since last clear:
info 2024-12-01 22:06:06zh:ember:uart:ash: Total frames: RX=3461, TX=4449
info 2024-12-01 22:06:06zh:ember:uart:ash: Cancelled : RX=0, TX=0
info 2024-12-01 22:06:06zh:ember:uart:ash: DATA frames : RX=3447, TX=986
info 2024-12-01 22:06:06zh:ember:uart:ash: DATA bytes : RX=82131, TX=56911
info 2024-12-01 22:06:06zh:ember:uart:ash: Retry frames: RX=13, TX=3
info 2024-12-01 22:06:06zh:ember:uart:ash: ACK frames : RX=0, TX=3460
info 2024-12-01 22:06:06zh:ember:uart:ash: NAK frames : RX=0, TX=0
info 2024-12-01 22:06:06zh:ember:uart:ash: nRdy frames : RX=0, TX=0
info 2024-12-01 22:06:06zh:ember:uart:ash: CRC errors : RX=0
info 2024-12-01 22:06:06zh:ember:uart:ash: Comm errors : RX=0
info 2024-12-01 22:06:06zh:ember:uart:ash: Length < minimum: RX=0
info 2024-12-01 22:06:06zh:ember:uart:ash: Length > maximum: RX=0
info 2024-12-01 22:06:06zh:ember:uart:ash: Bad controls : RX=0
info 2024-12-01 22:06:06zh:ember:uart:ash: Bad lengths : RX=0
info 2024-12-01 22:06:06zh:ember:uart:ash: Bad ACK numbers : RX=0
info 2024-12-01 22:06:06zh:ember:uart:ash: Out of buffers : RX=0
info 2024-12-01 22:06:06zh:ember:uart:ash: Retry dupes : RX=13
info 2024-12-01 22:06:06zh:ember:uart:ash: Out of sequence : RX=0
info 2024-12-01 22:06:06zh:ember:uart:ash: ACK timeouts : RX=3
error 2024-12-01 22:06:15zh:ember:ezsp: ERROR Transaction failure; status=ASH_ERROR_TIMEOUTS. Last Frame: [FRAME: ID=52:"SEND_UNICAST" Seq=203 Len=7].
error 2024-12-01 22:06:15zh:ember:ezsp: ERROR Transaction failure; status=ASH_NCP_FATAL_ERROR. Last Frame: [FRAME: ID=52:"SEND_UNICAST" Seq=203 Len=7].
info 2024-12-01 22:06:36zh:ember:uart:ash: Port closed.
info 2024-12-01 22:06:36zh:ember:uart:ash: ======== ASH stopped ========
info 2024-12-01 22:06:36zh:ember:ezsp: ======== EZSP stopped ========
info 2024-12-01 22:06:36zh:ember: ======== Ember Adapter Stopped ========
error 2024-12-01 22:06:37z2m: Adapter disconnected, stopping

Nerivec, 2024-12-02:
_I've seen a few reports of this recently, seems to be stress-related, a dropped-packet issue, still investigating though..._
Can you try updating to 8.0.2 and see if the issue is still present for you? (You can revert back to 7.4.4 later if you prefer.)

akaSyntaax, 2024-12-03:
Not OP, but I have the same issue. I tried versions 7.4.3 up to 8.0.2 with no luck, although it seems as if the progress went a bit further on 8.0.2 before crashing.

Mikei386, 2024-12-03:
I think that’s the same issue as here: 
[https://github.com/Koenkk/zigbee2mqtt/issues/23861](url)

Nerivec, 2024-12-03:
Indeed, I'm going to close this in favor of #23861 to avoid spreading information too much. Please refer to that one for follow-ups.

closed 2024-12-03 by Nerivec