Koenkk/zigbee2mqtt #25003: Many Actions do not provide Device Conditions in Home Assistant

ejpenney, 2024-12-01:
### What happened?

Saw the note that __action_ entities are going away, went to see how feasible this is in my environment.  For example, I have an automation:

```
description: ""
mode: single
triggers:
  - trigger: state
    entity_id:
      - sensor.ikea_remote_action
    from: null
    to: brightness_up_hold
conditions: []
actions:
  - repeat:
      until:
        - condition: state
          entity_id: sensor.ikea_remote_action
          state: brightness_down_release
      sequence:
        - data:
            brightness_step_pct: 10
            transition: 1
          target: light.upstairs_all
          action: light.turn_on
        - delay: "1"
```

This automation might look something like this after removing the __action_ entities:

```
description: ""
mode: single
triggers:
  - domain: mqtt
    device_id: 91096acf4566eae62076c12c8803ff32
    type: action
    subtype: brightness_up_hold
    trigger: device
conditions: []
actions:
  - repeat:
      until:
        - ???
      sequence:
        - data:
            brightness_step_pct: 10
            transition: 1
          target: light.upstairs_all
          action: light.turn_on
        - delay: "1"
```

But the repeat condition has no equivalent:

[image]

Meaning this automation will break when the __action_ entities go away and I see no equivalent behavior with MQTT Device Triggers.

### What did you expect to happen?

I expected that the events would be available as repeat conditions, but they are not.

### How to reproduce it (minimal and precise)

Try to have an automation that triggers from a long-press event with (any?) Zigbee switch, and then repeats an event until a button-up event occurs on that same device.

I have confirmed this limitation for the following devices:
- TRADFRI remote control (E1524/E1810)
- 3 scene remote and dimmer (ZB-3008)
by RGB Genie
- Wireless switch with 4 buttons (TS0044)
by Tuya
- Hue dimmer switch (324131092621)
by Philips

### Zigbee2MQTT version

1.42.0

### Adapter firmware version

20230507

### Adapter

usb-Silicon_Labs_Sonoff_Zigbee_3.0_USB_Dongle_Plus_0001-if00

### Setup

Proxmox LXC (TTeck)

fir3drag0n, 2024-12-03:
I can report this as well. Several Ikea and aqara devices are missing actions.

ibimslawa, 2024-12-03:
I also have this problem with my TS0044 and TS0043 -- but found the cause! 
In preparation of Z2M v2.0 (since v1.40 or so) the configuration option "homeassistant_legacy_triggers" supposedly was set to false. When this option is set to false the Action entities disappear.
**So for now just set the following in Z2M configuration.yaml:**
```
advanced:
  ...
  homeassistant_legacy_triggers: true
```
But I don't know how these devices will function after the 2.0 update in January...

Koenkk, 2024-12-13:
The legacy action sensor will not be removed with 2.0.0 anymore (https://github.com/Koenkk/zigbee2mqtt/discussions/24198#discussioncomment-11561858). So I will close this.

closed 2024-12-13 by Koenkk