Koenkk/zigbee2mqtt #25004: All devices missing after upgrading from 1.41.0-1 to 1.42.0-1

ki4hrg, 2024-12-02:
### What happened?

After upgrading from 1.41.0-1 to 1.42.0-1, all devices are missing from the device list in z2m as well as from the MQTT integration in Home Assistant.

### Zigbee2MQTT version

1.42.0

### Adapter firmware version

 7.1.1.0 build 273

### Adapter

EZSP v9

### Setup

Add-on on Home Assistant OS using Home Assistant Green

backslashV, 2024-12-02:
Same issue!!!

ki4hrg, 2024-12-02:
So, after a good bit of working on the issue, I finally got it (mostly) working. The issue appears to be with the new location for the config required for z2m to be included in HA’s backups - which, incidentally, made this a nightmare to recover from.

In my case, I attempted - per the instructions for opening an issue - to switch to the edge version, and in the process uninstalled the stable version. Where I had been able to recover from the pre-upgrade backup, I was unable to after trying unsuccessfully to switch to edge and then switching back to stable - still no devices on stable. This tipped me off, though, since the pre-upgrade backup on inspection clearly didn’t have any of the actual data from z2m.

I finally got my devices back after downloading my last **full** HA backup, getting the z2m files from there, and reuploading them to the new `/addon_config/zigbee2mqtt ` directory. I’m missing one device I added in the interim and all the devices are missing any HA metadata such as areas and labels, but they’re at least showing up.

I’m guessing the issue can be resolved in the short-term by changing the add-on config to make set `data_path` to `/config/zigbee2mqtt` which is where the config files used to live and need to be migrated from. You’ll need to migrate eventually, but hopefully that’ll keep you from being dead in the water in the interim.

Leaving this issue open because even if the migration does eventually need to happen, breaking the whole install without a clear warning on upgrade ain’t the way to make that happen.

ThePatricide, 2024-12-02:
Same issue. Wanted to roll back, but half my devices still didn't work in Home Assistant. They are recognized in Z2M, but Home Assistant lists many of them as unavailable. Still figuring out how to restore the working situation. This is quite unconvenient.

adekloet, 2024-12-02:
Indeed this happened, I did copy all the files to the new location, changed the Data path but it didn't bring back the devices. It must be something with the last sentence that the new config file location needs to be mounted. I expect that the mounting is done by the add-on itself, but it seems not to do that. I restarted Home Assistant to see if that would force mounting then ew datapath location, but no.

I left a copy of the config files at the new location, but changed the data path back to the /config location and then all works like normal. I will wait a while to see how this works out for others, but be aware of this.

demey, 2024-12-02:
Same issue.
I performed all necessary steps before update by stopping the addon, coping z2m configuration files  to 
`/addon_configs/45df7312_zigbee2mqtt` and changing `data_path` value. 
Z2m addon started, but all devices disappeared from Web UI. 
No new logs were generated in the new configuration location as well.

I resolved this issue by changing `data_path` value to old one. 
I think z2m works now with configuration from old location.

samumar82, 2024-12-02:
Hello, I tried to move the directories as per instructions but looks like in my installation are placed differently.
I'm using the HA Green device.
Anyone can support?

dpgh947, 2024-12-02:
I also updated without doing anything and everything still works, nothing has moved, files still under /config. I do have the new directory under addon_configs, which seems to have been created at the start of November, and it is empty.

Do we have to move manually, should it have happened automatically on this update, or will it be done automatically at some future point? Why is mine still working but others have their installs broken by this release? Notes are very confusing - "Unreleased" ???

supermarkert, 2024-12-02:
> Do we have to move manually, should it have happened automatically on this update, or will it be done automatically at some future point? Why is mine still working but others have their installs broken by this release? Notes are very confusing - "Unreleased" ???

I've since realized that I believe this specific issue only affects HA Add-on users, and that should be discussed in this issue:
https://github.com/zigbee2mqtt/hassio-zigbee2mqtt/issues/664

ki4hrg, 2024-12-02:
Yep, that definitely seems to be the same issue, though it seems like the issue with the add-on not even starting might be separate from the missing devices one.

MTokarev, 2024-12-02:
I had the same issue.
I made one deviation from the instructions: instead of setting `data_path` to `/addon_config/zigbee2mqtt`, I set it to the actual location `/addon_configs/8c77aaed_zigbee2mqtt`.

It works when the folder is moved under your add-on folder `/addon_configs/8c77aaed_zigbee2mqtt`, but the `data_path` is still set to `/addon_config/zigbee2mqtt`.
This is likely related to the logic that mounts the addons folder into a Docker container.

Nerivec, 2024-12-02:
Please see https://github.com/zigbee2mqtt/hassio-zigbee2mqtt/issues/664#issuecomment-2512758513
Closing this one to avoid too much spread-out information.

closed 2024-12-02 by Nerivec