Koenkk/zigbee2mqtt #25008: [Feature request]: A way to re-establish a better route for a device

henrykuijpers, 2024-12-02:
### Is your feature request related to a problem? Please describe

Problem: Devices are added & removed, sometimes devices are paired in a different location than they will be placed. Consequence: Device will become unreachable, because the route is broken and/or the device is trying to go through an unreachable device to the coordinator.

### Describe the solution you'd like

It would be nice to have a button "reestablish route". It will find the best route available and use that. 

### Describe alternatives you've considered

Go and dig up the device, pair it again. Often you'll find yourself removing all kinds of lamps / wall plates / ... just to do this. 

### Additional context

Device is 99% of the time offline, sometimes online, when it can get a fraction of connection.

Nerivec, 2024-12-02:
This is handled by the device itself, and not something that can be manually controlled.
Proper devices should automatically migrate to the "best available router" when moved (it sometimes takes a while, few minutes, few hours, but eventually, they should).
Note that some devices are not respecting this (like Aqara), and will be stuck to the first router they are paired with; unfortunately, the only recourse for these is to only pair them at their final destination.

closed 2024-12-02 by Nerivec

henrykuijpers, 2024-12-02:
Hm. In my case, they are ecodim (not aqara) and they're 230v (not battery powered). I noticed that one device came back today, but I'm not sure if it will actually stay or go "offline" again soon.

Nerivec, 2024-12-02:
I'm not familiar with that brand, not sure what quirks it might have.
What LQI do you have for that device?
If below 150, you could try putting a router device somewhere in-between it and the closest "good signal" router (or coordinator), to help with the relay of information.
Is it connected through a router, or to the coordinator directly (can check in the network map)? How far?
Devices with bad LQI will sometimes try a bit too hard to find a better match (_that behavior is usually easily visible with battery-powered devices since that consumes battery much faster than normal, but a bit harder to detect for mains-powered devices_).

henrykuijpers, 2024-12-02:
Thanks for that! That message contained a few pieces of information that weren't so clear to me yet! 

That device indeed has a low LQI, 160. Even though it seems to be considered "ok", it's really on the edge of reachability. 

Unfortunately I can't check the network map, as it fails to load. It will load forever and never display a map unfortunately. 

I believe it is connected through a router that isn't there anymore. And I believe it's kind of hard wired into that router, which of course will never provide access (directly or indirectly) to the coordinator anymore. 

It could also be connecting to the coordinator directly, which is not reachable, since the device moved upstairs after pairing. And the coordinator is downstairs. There is a concrete floor of 22 cm thickness with lots of iron in it in-between the 2 floors. 

Maybe I can add the dimmer in the hallway, which will then be a new router perfectly in-between the weak device and the coordinator. 
Will it then automatically discover this new router, though? And automatically update its routing? 
I mean, is a new router device appearing a signal for existing devices, so that they can negotiate with them to see if they provide a better route? (If that is the case, can this negotiating be triggered again for an existing device?)

Nerivec, 2024-12-02:
The mesh should adapt automatically to the presence of a new router, assuming all involved devices are properly following the Zigbee protocol.
If a device seems stuck to a router, it could mean something went wrong with that device, in which case, you may want to do a full reset of that device (checks docs/manual/manufacturer) and re-pair it again, or it could mean that indeed, like Aqara, it is not following the Zigbee protocol properly.
However, based on your previous post, the device "sometimes works", so, it definitely is no longer connected to that router that is no longer there (otherwise, it would _never_ work). It sounds more like it is having troubles reaching back to the coordinator, so, indeed, a good router somewhere in-between should help.

_Note that LQI is always influenced by device+network, for some, 150 might be fine, even 100, and for some, as soon as it starts approaching 150, they start acting up._

You can give the dimmer in-between a try. Just make sure to let the network settle down before making the final call, in some cases, it can take a couple hours for the network to properly adapt (or a couple minutes if the devices are reactive/few 😉).