Koenkk/zigbee2mqtt #25010: Failed to configure (ssIasZone, genPowerCfg, genBinaryInput, msTemperatureMeasurement, ssIasWd)

JoJoJo2211, 2024-12-02:
### What happened?

After installing update 1.42.0, I encountered these errors. I installed several frient smoke detectors several weeks ago. Here are the various error messages:

z2m: Failed to configure 'Rauchmelder Büro', attempt 1 (Error: Bind 0x0015bc00310241a7/35 ssIasZone from '0xe0798dfffefed0a9/1' failed (Delivery failed for '50068'.) at EmberOneWaitress.deliveryFailedFor (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/oneWaitress.ts:100:31) at EmberAdapter.onMessageSent (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:469:42) at Ezsp.emit (node:events:519:28) at Ezsp.ezspMessageSentHandler (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:5032:14) at Ezsp.callbackDispatch (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:735:26) at Ezsp.onAshFrame (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:395:22) at UartAsh.emit (node:events:519:28) at Immediate.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/ember/uart/ash.ts:1169:45) at processImmediate (node:internal/timers:478:21))

z2m: Failed to configure 'Rauchmelder Flur OG', attempt 1 (Error: Bind 0x0015bc003102667b/35 genBinaryInput from '0xe0798dfffefed0a9/1' failed (Delivery failed for '9766'.) at EmberOneWaitress.deliveryFailedFor (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/oneWaitress.ts:100:31) at EmberAdapter.onMessageSent (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:469:42) at Ezsp.emit (node:events:519:28) at Ezsp.ezspMessageSentHandler (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:5032:14) at Ezsp.callbackDispatch (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:735:26) at Ezsp.onAshFrame (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:395:22) at UartAsh.emit (node:events:519:28) at Immediate.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/ember/uart/ash.ts:1169:45) at processImmediate (node:internal/timers:478:21))

z2m: Failed to configure 'Rauchmelder Flur OG', attempt 3 (Error: Bind 0x0015bc003102667b/38 msTemperatureMeasurement from '0xe0798dfffefed0a9/1' failed (Delivery failed for '9766'.) at EmberOneWaitress.deliveryFailedFor (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/oneWaitress.ts:100:31) at EmberAdapter.onMessageSent (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:469:42) at Ezsp.emit (node:events:519:28) at Ezsp.ezspMessageSentHandler (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:5032:14) at Ezsp.callbackDispatch (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:735:26) at Ezsp.onAshFrame (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:395:22) at UartAsh.emit (node:events:519:28) at Immediate.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/ember/uart/ash.ts:1169:45) at processImmediate (node:internal/timers:478:21))

z2m: Failed to configure 'Rauchmelder Flur OG Treppe', attempt 1 (Error: Bind 0x0015bc0031026394/38 msTemperatureMeasurement from '0xe0798dfffefed0a9/1' failed (Delivery failed for '49183'.) at EmberOneWaitress.deliveryFailedFor (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/oneWaitress.ts:100:31) at EmberAdapter.onMessageSent (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:469:42) at Ezsp.emit (node:events:519:28) at Ezsp.ezspMessageSentHandler (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:5032:14) at Ezsp.callbackDispatch (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:735:26) at Ezsp.onAshFrame (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:395:22) at UartAsh.emit (node:events:519:28) at Immediate.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/ember/uart/ash.ts:1169:45) at processImmediate (node:internal/timers:478:21))

z2m: Failed to configure 'Rauchmelder Gästebad', attempt 1 (Error: Bind 0x0015bc00310262f7/35 genBinaryInput from '0xe0798dfffefed0a9/1' failed (Delivery failed for '4478'.) at EmberOneWaitress.deliveryFailedFor (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/oneWaitress.ts:100:31) at EmberAdapter.onMessageSent (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:469:42) at Ezsp.emit (node:events:519:28) at Ezsp.ezspMessageSentHandler (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:5032:14) at Ezsp.callbackDispatch (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:735:26) at Ezsp.onAshFrame (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:395:22) at UartAsh.emit (node:events:519:28) at Immediate.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/ember/uart/ash.ts:1169:45) at processImmediate (node:internal/timers:478:21))

etc.

### Zigbee2MQTT version

1.42.0 commit: unknown

### Adapter firmware version

 7.4.3 [GA]

### Adapter

 EmberZNet

### Setup

Add-On on HA on Generic x86-64

Nerivec, 2024-12-02:
These are battery operated? If yes, try pressing a button on them to wake them up before triggering the reconfigure (might have to press at regular 1-2sec intervals until the configure is done, to keep them from falling asleep too fast).

JoJoJo2211, 2024-12-02:
The smoke detectors are actually fully registered or trained. All entities are also displayed as current and updated regularly.

Nerivec, 2025-01-08:
`TABLE_FULL` is a different error, it means the device can no longer accommodate new bindings. That usually happens when the device doesn't clear previous bindings properly, or when the state of bindings between Z2M and the device get out of sync (Z2M tries to bind, but device already has a previous entry and table is filled). (Can happen with brand new devices too if some tests were run on the device, and they forgot to clear it before shipping it.)

Did you try a factory reset on one of the detectors?

_Could also be a poor firmware implementation on the device, and it's reporting the wrong status, but needs more investigation before we can safely ignore the error._

Smurfsmurfsmurf, 2025-01-09:
Ah ok. Sorry then.
Ofcourse ill provide any informtion u need.
But as its a different error, should I - to not spoil this thread - open a new issue, or use the "discussions" page (as it shouldnt be a z2m problem then)?

Nerivec, 2025-01-09:
Could be related, so let's keep the discussion in here 👍

Smurfsmurfsmurf, 2025-01-10:
OK.
I migratet from ZHA to z2m and was using the smoke detectors in ZHA already (even without problems, but also without firmware updates, and without checking the logs for errors).
For repairing I did a reset on every single device.
As I did purchase my devices over time, I found different firmware versions when migrating (no OTA for this device in ZHA).
It were 2 or 3, and I then upgraded all my devices to newest firmware (z2m sais its .0.8).
So in total the error occured with 3 or 4 different firmware versions.

Because Im having huge trouble with OTA updates in z2m (the only thing tht doesnt work properly), I did install HA in virtual machine under windows and do updates one by one over this system.
So I did reset those guys 3 times (ZHA -> z2m | z2m -> testsys | testsys -> z2m). The behaviour was never different, neither inside the freshly installed testsystem, with z2m running only 1 device)

Smurfsmurfsmurf, 2025-01-16:
I just realized, when restarting z2m every Smoke detector is "configuring". (z2m sais "configuring smoke detector xy).
This only applies for thos Frient smoke detector and for no other device.
Maybe we cn search here?
Why they are configured on startup while other devices are not?

Nerivec, 2025-01-16:
@Smurfsmurfsmurf  About your issues with OTA stability, try the new firmware release https://github.com/Nerivec/silabs-firmware-builder/releases/tag/v2024.6.2-update5 we're getting good feedback so far for users affected with timing issues (looks like it's mostly the Dongle-E with CP chip).

Can you confirm that these are SMSZB-120, and what firmware version? Can you also post a screenshot of the `Bind` page for the device in Z2M frontend?

Smurfsmurfsmurf, 2025-01-17:
Heyhey
Concearning OTA:
Im using mainly 3Reallity Plugs as routers. They recently received a firmware update. In the changelog was something with "Fixing issue that OTA updates fail". So I did setup a test system vi virtulbox on my Windows machine, migrated device by device and updated on my test machine. Having only 1 device connected directly next to the coordinator makes the update run much more stable. And I didnt want to restart my main z2m again and again.
But doing this I updated all my devices. So I cant perform any tests at the moment.
As soon as I see an update I ofcourse will change firmware on my coordinator (Dongle E) and also on my 4 Dongle E Routers.

Concearning the smoke detectors:
These all are Frient SMSZB-120 [these guys here](https://www.zigbee2mqtt.io/devices/SMSZB-120.html#develco-smszb-120)
They all are running on Firmware 4.0.8 - z2m tells me thats the recent firmware version.

[image]

And some offtopic about the coordinator.
I bought the Sonoff Dongle E when starting with Zigbee, becaus internet told me it has the much more modern chipset compred to Dongle P for example. (I atleast understood it that way).
And if im correct the Dongle E Firmare version 8.x.x.x is kinda custom firmware.
If I got a device I need a custom firmware on it, in order to work correctly (dont fail during OTA update), isnt thinking of another coordintor an option?

Nerivec, 2025-01-17:
@mamrai1 regarding the smoke detectors, did you have any more issues after https://github.com/Koenkk/zigbee2mqtt/issues/23684 ?

@Smurfsmurfsmurf about the Silabs coordinator firmware, a few builders ([darkxst](https://github.com/darkxst/silabs-firmware-builder/releases), [nabucasa](https://github.com/NabuCasa/silabs-firmware-builder/releases) & [myself](https://github.com/nerivec/silabs-firmware-builder/releases) mainly) provide firmware built with the same Silabs tool that manufacturers use, and mostly only change configuration to adapt to user feedback. Since the Silabs tool is available, it provides us with the opportunity to keep up with updates (important for a coordinator!), and try to fix common problems, rather than waiting for manufacturers to provide them (which sometimes never happens).
Koenkk does the same thing for [zStack (TI) adapters](https://github.com/Koenkk/Z-Stack-firmware).

mamrai1, 2025-01-17:
No problem fixed. Was definitely a problem with the device’s firmware

Smurfsmurfsmurf, 2025-01-17:
If I remove and readd 3 devices the bindings look like this:

[image]

If I got it correct this is how it should look like?
I didnt see an error on the newly added devices after restarting HomeAssistant, but with all the errors its hard to find specific devices (got 25 smoke detectors). Ill keep readding devices and watch.
Will need some days and report status.

@Nerivec
Thanks for the coordinator explanation!
If I see an OTA update, do u want me to change the coordinator firmware only, or also the firmware of my 4 routers`?
All devices are Sonoff Dongle E.

mamrai1, 2025-01-17:
Yes, no need to remove. Just unbind the genbasic cluster and then reconfigure the device

Smurfsmurfsmurf, 2025-01-17:
Hmm. Sorry again, but I have some devices with bound Cluster "PowerCfg" and ome without.
There should be 4 bounded clusters in the first row. Those are: GenBinaryInput, PowerCfg, sslasWd, sslasZone.
And in the second row only 1 which is Temperature.
So it should look exctly as in my last postet screenshot?

mamrai1, 2025-01-17:
Yes check the first row and unbind any cluster that is different than your screenshot and then press the configure button

Nerivec, 2025-01-17:
@Smurfsmurfsmurf about firmware, you can't OTA the coordinator or an adapter-as-router. You need to flash them using a third-party tool (like [Ember ZLI](https://github.com/Nerivec/ember-zli/wiki), or [universal-silabs-flasher](https://github.com/NabuCasa/universal-silabs-flasher)). But if everything is working for you, you don't necessarily _need_ to (this is definitely not related to your issue here). Feel free to create a discussion [in that repo](https://github.com/Nerivec/ember-zli/discussions) if you have more questions 😉

Thanks @mamrai1 for the feedback!

Smurfsmurfsmurf, 2025-01-24:
Heyho
It took some time to change the clusters, because I got a lot of errors when binding/unbinding.
Retrying some hours later helped.
I managed to correct the clusters and reconfigured all smoke detectors.
I waited another day (in case the mesh needs some time after changing all the clusters) and restartet z2m.
And tada - not a single error!
I love you guys!

@Nerivec concearning OTA and the firmware.
I know about OTA and flashing firmware on my coordinator.
What I ment was - I dont have any DEVICE in z2m that does not use the latest firmware.
So I cant OTA update at the moment. So I cant test flashing my corrdinator with 8.x.x.x firmware.

Can u provide a link to the thread where people are testing the coordinator 8.x.x.x firmware to use OTA updates?

closed 2025-04-03 by github-actions[bot]