Koenkk/zigbee2mqtt #25012: Update issue with Nodon SIN-4-FP-21 module in Home Assistant and Z2M

lennvilardi, 2024-12-02:
### What happened?

I have a Nodon SIN-4-FP-21 module that constantly asks for an update in Home Assistant, but when I click on it, nothing happens.
[image]

### What did you expect to happen?

Update correctly

### How to reproduce it (minimal and precise)

Version information:
Installed version: 66564
Latest version: 66816
Additional details:
When I try to update it via Zigbee2MQTT, I get the following error:
```
info 2024-12-02 09:37:16z2m: Updating 'interrupteur_filpilote.salledebain' to latest firmware
info 2024-12-02 09:37:18z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/interrupteur_filpilote.salledebain', payload '{"energy":138.58,"linkquality":43,"pilot_wire_mode":"eco","power":0,"state":"OFF","update":{"installed_version":66564,"latest_version":66816,"state":"available"},"update_available":null}'
info 2024-12-02 09:37:18z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{"id":"0xbc026efffeae2bb5"},"error":"Update of 'interrupteur_filpilote.salledebain' failed (No new image available)","status":"error"}'
```
[image]

I also downloaded the Nodon Connect app to perform the update, but it does not detect any update for this module. Additionally, it identifies another multifunction module SIN2BB5 as already up to date with version 3.5.0.

[image]

### Zigbee2MQTT version

1.42.0

### Adapter firmware version

20230507

### Adapter

zStack3x0

### Setup

Addon on HA on a VM

Jeepmg, 2024-12-03:
Have you looked on the French HA forum? Several topics talk about this module. I've been hesitating to buy one for months but I keep putting it off.

[https://forum.hacf.fr/latest](url)

lennvilardi, 2024-12-04:
Thank you for the link and your advice! I spent my evening browsing the HACF forum looking for discussions about this module, but unfortunately, I couldn’t find a solution to my problem. I’ve already tried pairing via Bluetooth, but without success. If you have any other suggestions or come across a solution, I’d greatly appreciate it! 😊

keisersuze, 2024-12-10:
Je rencontre le même problème...

floow77, 2025-01-03:
I have the same issue

lennvilardi, 2025-01-03:
I successfully flashed the SONOFF Zigbee 3.0 USB Dongle Plus (ZBDongle-P) with the latest Coordinator firmware: CC1352P2_CC2652P_launchpad_coordinator_20240710.

This update seems to have resolved the issue. The dongle is functioning as expected, and for the sin-4-fp-21, it appears to already be on the latest firmware with no further updates available.

I’ll continue monitoring the setup over the next few days to confirm stability.

lennvilardi, 2025-01-06:
error 2025-01-06 15:41:00z2m: Update of 'interrupteur_filpilote.salledebain' failed (No new image available)
seems the problem is back

ebozonne, 2025-01-10:
same issu

gael1980, 2025-01-14:
same problem

pmpbpl, 2025-02-11:
Have anyone find a solution?

96Ems, 2025-02-12:
Same issue there. Moreover, once it happens, changing pilote_mode not working and have to restart z2m addon (via service) it works again Tillaut next update notification

chrisvec, 2025-02-13:
same problem

ggarnache, 2025-04-11:
Same problem here (exact same versions)

RastaChaum, 2025-04-17:
#metoo :(

benisr007, 2025-04-28:
Same problem here

kerdijo, 2025-05-22:
(translated by Google)
Same problem with four modules under Home Assistant and Z2M.
Resolved with the NODON CONNECT Android app.
Update successful, but it took several attempts.
It's been ten days, and since then, there have been no new updates.

closed 2025-07-29 by github-actions[bot]