Koenkk/zigbee2mqtt #25013: Unable to load the network map

henrykuijpers, 2024-12-02:
### What happened?

When loading the network map, I see:
[image]

But then I don't see any network requests happening (browser network tab). I never see a map.

### What did you expect to happen?

A map to display.

### How to reproduce it (minimal and precise)

Go to the "Map"-tab and click "Load map".

### Zigbee2MQTT version

1.32.2 commit: 1ec1e57

### Adapter firmware version

0x26720700

### Adapter

0x00212effff05223d

### Setup

Docker container

Nerivec, 2024-12-02:
Your version of Z2M is very old (August 2023). Can you try updating to a more recent version?

vlavrynovych, 2024-12-11:
Hi @Nerivec, 
I am experiencing same issue with the latest version 1.42. I recently migrated from 1.35.1-1 to 1.42.0-2. 

Here is my comment in another thread: https://github.com/Koenkk/zigbee2mqtt/issues/25132#issuecomment-2537343754
I am not 100% sure that the version I have today is overall work correctly, because I cannot start joining mode (which is more critical for me). Trying to figure it out if it's was bad migration or a problem of newer version.

JimGermany, 2024-12-19:
I currently have the same problem: The Z2M card is no longer loading. I updated from HA Core 2024.11.3 to Core 2024.12.x just last week (currently running Core 2024.12.4) and before that there was no problem with the Z2M card. So I suspect it has something to do with HA Core 2024.12.x. I tested it with Firefox and Chromium.

HA Version (Proxmox VM)
Core 2024.12.4
Supervisor 2024.12.0
Operating System 14.0
Frontend 20241127.8

Z2M: 1.42.0-2 as HA Addon

Firefox: 133.0.3
Chromium: 131.0.6778.85

Restarting Z2M or rebooting the HA host does not change anything.

Nerivec, 2024-12-19:
Did you try this out (if applicable)? https://github.com/Koenkk/zigbee2mqtt/issues/25132#issuecomment-2539182974

What exactly do you mean by Z2M card, do you mean a HA UI custom card for Z2M?

JimGermany, 2024-12-20:
> What exactly do you mean by Z2M card, do you mean a HA UI custom card for Z2M?

Thanks for your answer and sorry, that was a translation error. :-) In the German translation, the menu item is called "Karte" 
[image]
and then, without thinking too much, I wrote card. But of course I meant the Z2M map.

BTW: As a German translation, the term "Netzwerkkarte" would be (a little) better/clearer than "Karte".

Normally the network map would load and be displayed after about 10 - 20 seconds (I use about 40 Zigbee devices). Now it is no longer displayed, no matter how many minutes I wait.
[image]

> Did you try this out (if applicable)? [#25132 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/25132#issuecomment-2539182974)
> 
The entry permit_join: is in my Z2M configuration.yaml, but it is set to false: permit_join: false
I deleted the permit_join: entry as a test, but unfortunately that didn't help. The network map is still not loading. Both the Z2M addon and the HA host were restarted after the change.

Nerivec, 2024-12-20:
Can you set log level to `debug` and re-trigger a network map? We'd need to see why these are failing. It would appear it is timing out (that would explain why it is taking so long), but if the devices are still available in frontend, something strange is going on.

JimGermany, 2024-12-20:
First of all: My Zigbee Mesh map (Coordinator + Routers) currently looks like this. There is the SLZB-06 and 16 different Zigbee plugs as routers. All Zigbee connections are stable and all LQI values ​​are actually more than sufficient (between about 80 and about 150).
[image]

FYI: Availability is set to simple, but even setting it to advanced with the default settings (timeout 10 for active devices) does not change anything.
[image]
The SLZB-06 dongle has also been restarted.

The debug log is a bit difficult because loading the map takes several minutes and the log quickly has more than 1,000 lines. Also, when I reload the map, the debug log only shows **error** messages about loading the LQI values like

`2024-12-20 15:41:31z2m: Failed to execute LQI for 'L_Plug2_Nest_Hub_Kueche'
2024-12-20 15:41:57z2m: Failed to execute LQI for 'L_Plug3_Waschmaschine'
2024-12-20 15:44:01z2m: Failed to execute LQI for 'L_Plug3_Waschmaschine'
2024-12-20 15:44:40z2m: Failed to execute LQI for 'L_Plug4_Nest_Hub_Arbeitszimmer'
2024-12-20 15:45:08z2m: Failed to execute LQI for 'T_Plug2_SonyTV'`

All other entries are just **info** entries about MQTT. So here is a short excerpt from the debug log (log level to debug).

`info 2024-12-20 15:50:29z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/T_Plug3_Stehlampe', payload '{"child_lock":"UNLOCK","current":0,"device":{"applicationVersion":112,"friendlyName":"T_Plug3_Stehlampe","hardwareVersion":0,"ieeeAddr":"0xa4c138b44650f0dc","manufacturerID":4107,"manufacturerName":"_TZ3000_0zfrhq4i","model":"TS011F_plug_3","networkAddress":26145,"powerSource":"Mains (single phase)","softwareBuildID":"1.0.5\u0000","stackVersion":2,"type":"Router","zclVersion":3},"energy":1.38,"indicator_mode":"off/on","last_seen":"2024-09-10T18:32:23+02:00","linkquality":87,"power":0,"power_outage_memory":"off","state":"ON","update":{"installed_version":268514823,"latest_version":268514823,"state":"idle"},"update_available":false,"voltage":220}'
error 2024-12-20 15:50:31z2m: Failed to execute LQI for 'L_Plug2_Nest_Hub_Kueche'
info 2024-12-20 15:50:37z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/T_Plug2_SonyTV', payload '{"child_lock":"UNLOCK","current":0,"device":{"applicationVersion":112,"friendlyName":"T_Plug2_SonyTV","hardwareVersion":0,"ieeeAddr":"0xa4c13844cddc7d10","manufacturerID":4107,"manufacturerName":"_TZ3000_0zfrhq4i","model":"TS011F_plug_3","networkAddress":56700,"powerSource":"Mains (single phase)","softwareBuildID":"1.0.5\u0000","stackVersion":2,"type":"Router","zclVersion":3},"energy":54.33,"indicator_mode":"off/on","last_seen":"2024-09-10T18:31:44+02:00","linkquality":98,"power":0,"power_outage_memory":"restore","state":"ON","update":{"installed_version":268514823,"latest_version":268514823,"state":"idle"},"update_available":false,"voltage":221}'
info 2024-12-20 15:50:40z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/BWM_01', payload '{"battery":100,"battery_low":false,"device":{"dateCode":"20200923 14:59","friendlyName":"BWM_01","hardwareVersion":"2.1.1","ieeeAddr":"0x0015bc001a01a9b0","manufacturerID":4117,"manufacturerName":"frient A/S","model":"MOSZB-140","networkAddress":55410,"powerSource":"Battery","softwareBuildID":"4.0.6","type":"EndDevice","zclVersion":1},"illuminance":13931,"illuminance_lux":25,"last_seen":"2024-09-10T18:32:18+02:00","led_control":"both","linkquality":36,"occupancy":false,"occupancy_timeout":20,"tamper":false,"temperature":16.81,"update":{"installed_version":262150,"latest_version":262150,"state":"idle"},"update_available":false,"voltage":3000}'
info 2024-12-20 15:50:40z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/L_Plug3_Waschmaschine', payload '{"child_lock":"UNLOCK","current":0,"device":{"applicationVersion":192,"dateCode":"","friendlyName":"L_Plug3_Waschmaschine","hardwareVersion":1,"ieeeAddr":"0xa4c1381e8a65616e","manufacturerID":4417,"manufacturerName":"_TZ3000_j1v25l17","model":"HG08673","networkAddress":27865,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"energy":13.42,"indicator_mode":"off/on","last_seen":"2024-09-10T18:31:58+02:00","linkquality":98,"power":0,"power_outage_memory":"off","state":"OFF","update":{"installed_version":192,"latest_version":192,"state":"idle"},"update_available":false,"voltage":223}'
info 2024-12-20 15:50:52z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/BWM_01', payload '{"battery":100,"battery_low":false,"device":{"dateCode":"20200923 14:59","friendlyName":"BWM_01","hardwareVersion":"2.1.1","ieeeAddr":"0x0015bc001a01a9b0","manufacturerID":4117,"manufacturerName":"frient A/S","model":"MOSZB-140","networkAddress":55410,"powerSource":"Battery","softwareBuildID":"4.0.6","type":"EndDevice","zclVersion":1},"illuminance":13971,"illuminance_lux":25,"last_seen":"2024-09-10T18:32:18+02:00","led_control":"both","linkquality":36,"occupancy":false,"occupancy_timeout":20,"tamper":false,"temperature":16.81,"update":{"installed_version":262150,"latest_version":262150,"state":"idle"},"update_available":false,"voltage":3000}'
info 2024-12-20 15:50:53z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/T_Plug3_Stehlampe', payload '{"child_lock":"UNLOCK","current":0,"device":{"applicationVersion":112,"friendlyName":"T_Plug3_Stehlampe","hardwareVersion":0,"ieeeAddr":"0xa4c138b44650f0dc","manufacturerID":4107,"manufacturerName":"_TZ3000_0zfrhq4i","model":"TS011F_plug_3","networkAddress":26145,"powerSource":"Mains (single phase)","softwareBuildID":"1.0.5\u0000","stackVersion":2,"type":"Router","zclVersion":3},"energy":1.38,"indicator [... cut ...] ,"voltage":3000}'`

As already mentioned, the Zigbee map does not always show the same plugs with an error regarding the LQI value. Sometimes the LQI values ​​cannot be read for just three plugs and then sometimes for four or five plugs.

I have also looked at various similar issue reports from the last few years, but unfortunately I couldn't find any information about what else I could try or perhaps configure somewhere else. And I haven't changed anything in Z2M, apart from the HA Core Update.

closed 2025-02-26 by github-actions[bot]