Koenkk/zigbee2mqtt #25014: User extensions can't import settings and logger

XenorPLxx, 2024-12-02:
### What happened?

I've had a user extension that utilizes settings and logger, code:
```
class ExternalSensorThermostatExtension {
    mapping = {
        // Temperature Sensor Id : Thermostat Names
        "0x00158d00044f0829" : [
            "Thermostat_Bathroom",
        ],
        "0x00158d00049fb9c6" : [
            "Thermostat_Office",
            "Thermostat_Living-Room",
        ],
        "0x00158d0004ab6053" : [
            "Thermostat_Bedroom",
        ],
    }

    constructor(zigbee, mqtt, state, publishEntityState, eventBus, settings, logger) {
        this.mqtt = mqtt;
        this.logger = logger;
        this.baseTopic = settings.get().mqtt.base_topic;
        this.eventBus = eventBus;

        // recalibrate the thermostats
        Object.keys(this.mapping).forEach(
            sensorId => this.mapping[sensorId].forEach(
                thermostatName => {
                    mqtt.onMessage(`${this.baseTopic}/${thermostatName}/set`, JSON.stringify({calibrate: "calibrate"}));
                    logger.info(`Calibrated ${thermostatName}`);
                }
            )
        )
        eventBus.on('stateChange', this.onStateChange.bind(this), this.constructor.name);

        logger.info(`Loaded  ${this.constructor.name}`);
    }

    async onStateChange(data) {
        const { entity, update } = data;
        if (!Object.keys(this.mapping).includes(entity.ID)) { // check if Temperature Sensor
            return
        }

        const { temperature } = update;
        if (temperature === undefined) { // no temperature update
            return
        }

        // set new temperature value to thermostats
        this.mapping[entity.ID].forEach(
            thermostatName => {
                this.mqtt.onMessage(`${this.baseTopic}/${thermostatName}/set`, JSON.stringify({ external_temperature_input: temperature }));
                this.logger.info(`The external_temperature_input of ${thermostatName} was set to ${temperature}`);
            }
        )
    }

    async onMQTTMessage(topic, message) {
    }

    async stop() {
        this.eventBus.removeListeners(this.constructor.name);
    }
}

module.exports = ExternalSensorThermostatExtension;
```

After a recent update, I see that constructor has changed (it's not updated in [the docs](https://www.zigbee2mqtt.io/advanced/more/user_extensions.html) yet tho). It doesn't include settings and logger anymore, and instead [internal extensions](https://github.com/Koenkk/zigbee2mqtt/tree/master/lib/extension) like [availability](https://github.com/Koenkk/zigbee2mqtt/blob/master/lib/extension/availability.ts) import directly. I've tried to replicate that in my user extension, but I'm getting `ExternalSensorThermostatExtension.js contains invalid code: Cannot use import statement outside a module`, code below.

```
import logger from '../util/logger';
import * as settings from '../util/settings';

class ExternalSensorThermostatExtension {
    mapping = {
        // Temperature Sensor Id : Thermostat Names
        "0x00158d00044f0829" : [
            "Thermostat_Bathroom",
        ],
        "0x00158d00049fb9c6" : [
            "Thermostat_Office",
            "Thermostat_Living-Room",
        ],
        "0x00158d0004ab6053" : [
            "Thermostat_Bedroom",
        ],
    }

    constructor(zigbee, mqtt, state, publishEntityState, eventBus) {
        this.mqtt = mqtt;
        this.logger = logger;
        this.baseTopic = settings.get().mqtt.base_topic;
        this.eventBus = eventBus;

        // recalibrate the thermostats
        Object.keys(this.mapping).forEach(
            sensorId => this.mapping[sensorId].forEach(
                thermostatName => {
                    mqtt.onMessage(`${this.baseTopic}/${thermostatName}/set`, JSON.stringify({calibrate: "calibrate"}));
                    logger.info(`Calibrated ${thermostatName}`);
                }
            )
        )
        eventBus.on('stateChange', this.onStateChange.bind(this), this.constructor.name);

        logger.info(`Loaded  ${this.constructor.name}`);
    }

    async onStateChange(data) {
        const { entity, update } = data;
        if (!Object.keys(this.mapping).includes(entity.ID)) { // check if Temperature Sensor
            return
        }

        const { temperature } = update;
        if (temperature === undefined) { // no temperature update
            return
        }

        // set new temperature value to thermostats
        this.mapping[entity.ID].forEach(
            thermostatName => {
                this.mqtt.onMessage(`${this.baseTopic}/${thermostatName}/set`, JSON.stringify({ external_temperature_input: temperature }));
                this.logger.info(`The external_temperature_input of ${thermostatName} was set to ${temperature}`);
            }
        )
    }

    async onMQTTMessage(topic, message) {
    }

    async stop() {
        this.eventBus.removeListeners(this.constructor.name);
    }
}

module.exports = ExternalSensorThermostatExtension;
```

I've tried using requires as below as well, but I'm getting `ExternalSensorThermostatExtension.js contains invalid code: Cannot read properties of undefined (reading 'info')`.

```
const { logger } = require('../util/logger');
const settings = require('../util/settings');
```

I'm unsure how to proceed to convert my user extension to be compatible with the new code.

### Zigbee2MQTT version

1.42.0-dev commit: acba533397dbfaf54f26a9bc5c5b199e46f2a0f5

### Adapter firmware version

20211217

### Adapter

Sonoff Zigbee USB CC2652P

### Setup

plain docker on unraid

Nerivec, 2024-12-02:
Docs are not yet updated to 2.0.0 since it's not released yet.
The constructor has been updated to have the same signature as internal extensions, so settings and logger have been pushed a few parameters, it can be seen [here](https://github.com/Koenkk/zigbee2mqtt/blob/dev/lib/extension/externalExtensions.ts#L43-L54) (the types for all parameters can be seen at the top in the extension's constructor itself)

If your case, this should do:

```diff
-    constructor(zigbee, mqtt, state, publishEntityState, eventBus, settings, logger) {
+    constructor(zigbee, mqtt, state, publishEntityState, eventBus, enableDisableExtension, restartCallback, addExtension, settings, logger) {
```

(make sure to remove all the import/require you tried with)

XenorPLxx, 2024-12-02:
Amazing! Didn't notice that before.
I've changed the constructor parameters and it works as expected, thanks!

closed 2024-12-02 by XenorPLxx