Koenkk/zigbee2mqtt #25019: [New device support]: TRV701Z ( TV02-ZG) - SILUMEN - Thermostatic radiator valve

cedbertrand, 2024-12-02:
### Link

https://silumen.com/collections/tetes-thermostatiques-connectees/products/tete-thermostatique-connectee-universelle

### Database entry

{"id":63,"type":"EndDevice","ieeeAddr":"0xa4c138dba4b3267f","nwkAddr":34251,"manufId":4417,"manufName":"_TZE204_zljqtner","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"9��.i:��.i;��.f;��.i<��.\u0012","65506":56,"65508":0,"modelId":"TS0601","manufacturerName":"_TZE204_zljqtner","powerSource":3,"zclVersion":3,"appVersion":73,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[{"cluster":0,"type":"endpoint","deviceIeeeAddress":"0x00212effff096451","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":73,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1733158177567}

### Zigbee2MQTT version

1.42.0

### Comments

The link I gave you does not correspond exactly to the thermostatic radiator valve I have, but it is from the same vendor and it is really close.English is not my natural language and I didn't understand correctly the difference steps to do to add manually support for my device...

I have created a file sillumen.js in my zigbee2mqtt folder in home assistant, next to the configuration.yaml, and I have edit the configuration.yaml file of zigbee2mqtt with theses lines:

external_converters:
  - silumen.js

for the silumen.js file, I used the "external definition" and used what I have found here : 
https://github.com/Koenkk/zigbee2mqtt/issues/24958

I have restarted zigbee2mqtt...
But nothing is working.

Could you please help me?

### External definition

```shell
// The generated one with out modifications
const {} = require('zigbee-herdsman-converters/lib/modernExtend');
// Add the lines below
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const ota = require('zigbee-herdsman-converters/lib/ota');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE204_zljqtner',
    description: 'SILUMEN Vanne thermostatique',
    extend: [],
    fromZigbee: [], // We will add this later
    toZigbee: [], // Should be empty, unless device can be controlled (e.g. lights, switches).
    exposes: [e.battery(), e.temperature(), e.humidity()], 
};

module.exports = definition;

// The one I have updated...
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE204_zljqtner',
    description: 'Tete thermostatique connectee',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime,
    configure: tuya.configureMagicPacket,
    exposes: [
        e.child_lock(),
        e.battery_low(),
        e
            .climate()
            .withSetpoint('current_heating_setpoint', 5, 35, 0.5, ea.STATE_SET)
            .withLocalTemperature(ea.STATE)
            .withSystemMode(['heat', 'off'], ea.STATE)
            .withLocalTemperatureCalibration(-9, 9, 0.1, ea.STATE_SET),
        ...tuya.exposes.scheduleAllDays(ea.STATE_SET, 'HH:MM/C HH:MM/C HH:MM/C HH:MM/C HH:MM/C HH:MM/C'),
        e
            .binary('scale_protection', ea.STATE_SET, 'ON', 'OFF')
            .withDescription(
                'If the heat sink is not fully opened within ' +
                    'two weeks or is not used for a long time, the valve will be blocked due to silting up and the heat sink will not be ' +
                    'able to be used. To ensure normal use of the heat sink, the controller will automatically open the valve fully every ' +
                    'two weeks. It will run for 30 seconds per time with the screen displaying "Ad", then return to its normal working state ' +
                    'again.',
            ),
        e
            .binary('frost_protection', ea.STATE_SET, 'ON', 'OFF')
            .withDescription(
                'When the room temperature is lower than 5 °C, the valve opens; when the temperature rises to 8 °C, the valve closes.',
            ),
        e.numeric('error', ea.STATE).withDescription('If NTC is damaged, "Er" will be on the TRV display.'),
    ],
    meta: {
        tuyaDatapoints: [
            [3, 'system_mode', tuya.valueConverterBasic.lookup({heat: tuya.enum(1), off: tuya.enum(0)})],
            [4, 'current_heating_setpoint', tuya.valueConverter.divideBy10],
            [5, 'local_temperature', tuya.valueConverter.divideBy10],
            [7, 'child_lock', tuya.valueConverter.lockUnlock],
            [28, 'schedule_monday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(1)],
            [29, 'schedule_tuesday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(2)],
            [30, 'schedule_wednesday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(3)],
            [31, 'schedule_thursday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(4)],
            [32, 'schedule_friday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(5)],
            [33, 'schedule_saturday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(6)],
            [34, 'schedule_sunday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(7)],
            [35, null, tuya.valueConverter.errorOrBatteryLow],
            [36, 'frost_protection', tuya.valueConverter.onOff],
            [39, 'scale_protection', tuya.valueConverter.onOff],
            [47, 'local_temperature_calibration', tuya.valueConverter.localTempCalibration1],
        ],
    },
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

Both config shows up on the device, but nothing seems to link with the device...
I hae changed some value in the web interface of the device in the zigbee2mqtt device, but it doesn't do anything to the device itself :
{
    "current_heating_setpoint": 22,
    "linkquality": 255,
    "battery_low": null,
    "child_lock": null,
    "error": null,
    "frost_protection": "ON",
    "local_temperature_calibration": 0,
    "scale_protection": "ON",
    "schedule_friday": null,
    "schedule_monday": null,
    "schedule_saturday": null,
    "schedule_sunday": null,
    "schedule_thursday": null,
    "schedule_tuesday": null,
    "schedule_wednesday": null
}

I have tryed to change the temperature required on the device, nothing changes on the zigbee2mqtt device's interface.

vjaunet, 2025-01-08:
Have you managed to get it working ? I have the exact same devices and I was planning to program them through z2m but realized too late their lack of support...
Bon courage ;),
Vincent

cedbertrand, 2025-01-08:
Hello/Bonjour :),
unfortunately I did not...  if you find a way, please let me know! :)

vjaunet, 2025-01-08:
This might help : https://github.com/Koenkk/zigbee2mqtt/issues/23754

vjaunet, 2025-01-16:
Hi there,  
here are the codes I got for the devices tuyadatapoints.
I am working on their implementation. Note that I could get the battery to report to z2m after I connected the device to the Tuya app instead of the Silumen one.
good luck.

{
"1":"Mode",
"2":"Target temperature",
"3":"Current temperature",
"6":"Working status",
"7":"Window status",
"8":"Open window",
"9":"Max. limit temperature",
"10":"Min. limit temperature",
"12":"Child lock",
"13":"Battery",
"14":"Fault alarm",
"101":"Room sensor calibration",
"102":"周程序1", // Weekly program 1
"103":"周程序2", // Weekly program 2
"104":"周程序3",	// Weekly program 3
"105":"周程序4", // Weekly program 4
"106":"周程序5", // Weekly program 5
"107":"周程序6", // Weekly program 6
"108":"周程序7", // Weekly program 7
"109":"机型", //model
"110":"Motor thrust",
"111":"Display brightness",
"112":"Software version",
"113":"Screen orientation",
"114":"Valve",
"115":"Switch deviation (energy-saving mode only)",
"116":"电机数据", //motor data
"117":"假期时长", //Holiday duration
"118":"Boost 模式时长", //Boost mode duration
"119":"舒适模式温度", //Comfort mode temperature
"120":"节能模式温度", //Energy saving mode temperature
"121":"防冻模式温度", //Antifreeze mode temperature
"122":"防冻使能", //Antifreeze enable
"123":"气压指数", //pressure index
"124":"临时模式开始时间", //Temporary mode start time
"125":"功能点",	//Function point
"126":"app支持特性", //app support features
"127":"System mode"
}

vjaunet, 2025-01-16:
Hello, here is what I could come up with so far. Feedback from the device seems consistent with the inputs. Still need to thoroughly test though.
++

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
fingerprint: [
{modelID: 'TS0601', manufacturerName: '_TZE204_zljqtner'},
],
model: 'TS0601_thermostat_1',
vendor: 'TuYa',
description: 'Thermostatic radiator valve',
whiteLabel: [
{vendor: 'Unknown/id3.pl', model: 'GTZ06'},
],
onEvent: tuya.onEventSetLocalTime,
fromZigbee: [tuya.fz.datapoints],
toZigbee: [tuya.tz.datapoints],
configure: tuya.configureMagicPacket,
exposes: [
    e.battery(),

    e.climate()
	.withLocalTemperature(ea.STATE)
	.withSetpoint('current_heating_setpoint', 5, 25, 0.5, ea.STATE_SET)
	.withLocalTemperatureCalibration(-3, 3, 0.1, ea.STATE_SET)
	.withPreset(['schedule', 'manual', 'off', 'on'],
		    'MANUAL MODE ☝ - In this mode, the device executes manual temperature setting. ' +
		    'When the set temperature is lower than the "minimum temperature", the valve is closed (forced closed). ' +
		    'SCHEDULE MODE ⏱ - In this mode, the device executes a preset week programming temperature time and temperature. ' +
		    'ON - In this mode, the thermostat stays open ' +
		    'OFF - In this mode, the thermostat stays closed')
	.withSystemMode(['auto', 'heat', 'off'], ea.STATE)
	.withRunningState(['idle', 'heat'], ea.STATE),

    e.enum('mode', ea.STATE_SET, ['comfort', 'eco'])
	.withDescription('Hysteresis - comfort > switches off/on exactly at reached temperature with valve smooth from 0 to 100%, eco > 0.5 degrees above or below, valve either 0 or 100%'),

    e.comfort_temperature().withValueMin(19).withValueMax(35).withValueStep(0.5).withDescription('Comfort mode temperature'),

    e.eco_temperature()
	.withValueMin(5.5)
	.withValueMax(19.5)
	.withValueStep(0.5)
	.withDescription('Eco mode temperature'),

    e.binary('frost_protection', ea.STATE_SET, 'ON', 'OFF')
	.withDescription('When the room temperature is lower than 5 °C (or the choosen one), the valve opens; when the temperature rises to 8 °C, the valve closes'),
    e.numeric('frost_protection_temperature', ea.STATE_SET)
        .withUnit('°C')
        .withValueMin(5).withValueMax(10).withValueStep(0.5)
        .withDescription(''),

    e.window_detection(),
    e.binary('window', ea.STATE, 'CLOSED', 'OPEN')
	.withDescription('Window status closed or open '),

    // not working : dp 114 "Valve"---------
   // e.binary('valve_state', ea.STATE, 'CLOSED', 'OPEN')
   //	.withDescription('Valve state closed or open '),

[... 84 lines cut ...]
"106":"周程序5", // Weekly program 1
"107":"周程序6", // Weekly program 1
"108":"周程序7", // Weekly program 7
"109":"机型", //model
"110":"Motor thrust",
"111":"Display brightness",
"112":"Software version",
"113":"Screen orientation",
"114":"Valve",
"115":"Switch deviation (energy-saving mode only)",
"116":"电机数据", //motor data
"117":"假期时长", //Holiday duration
"118":"Boost 模式时长", //Boost mode duration
"119":"舒适模式温度", //Comfort mode temperature
"120":"节能模式温度", //Energy saving mode temperature
"121":"防冻模式温度", //Antifreeze mode temperature
"122":"防冻使能", //Antifreeze enable
"123":"气压指数", //pressure index
"124":"临时模式开始时间", //Temporary mode start time
"125":"功能点", //Function point
"126":"app支持特性", //app support features
"127":"System mode"
}
*/
'''

vjaunet, 2025-01-22:
Yet another version : manual, eco, comfort, on, off, anti-frost control seem to work. Scheduled and Holiday are yet to be retro-engineered. From the Tuya App it seems that we can set as many 30min periods as possible and temperature should be set to comfort/eco/frost settings.

...
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
fingerprint: [
{modelID: 'TS0601', manufacturerName: '_TZE204_zljqtner'},
],
model: 'TS0601_thermostat_1',
vendor: 'TuYa',
description: 'Thermostatic radiator valve',
whiteLabel: [
{vendor: 'Unknown/id3.pl', model: 'GTZ06'},
],
onEvent: tuya.onEventSetLocalTime,
fromZigbee: [tuya.fz.datapoints],
toZigbee: [tuya.tz.datapoints],
configure: tuya.configureMagicPacket,
exposes: [
    e.battery(),

    e.climate()
	.withLocalTemperature(ea.STATE)
	.withSetpoint('current_heating_setpoint', 5, 22, 0.5, ea.STATE_SET)
	.withLocalTemperatureCalibration(-3, 3, 0.1, ea.STATE_SET)
	.withPreset(['off', 'comfort', 'manual', 'on','eco','anti-frost','schedule','holiday'],
		    'Comfort -- the device setpoint is set to comfort temperature.  ' +
		    'Eco -- the device setpoint is set to eco temperature.  ' +
		    'Manual --  the device setpoint can be adjusted manually. ' +
		    'When the set temperature is lower than the "minimum temperature", the valve is closed (forced closed). ' +
		    'on - In this mode, the thermostat stays open ' +
		    'off - In this mode, the thermostat stays closed' +
		    'anti-frost -- device setpoint is set to anti frost temperature' +
		    'Schedule -- device setpoint is cahnged according to the scheduled parameters (NOT WORKING)' +
    		    'Holiday --  (NOT WORKING)')
	.withSystemMode(['auto', 'heat', 'off'], ea.STATE)
	.withRunningState(['idle', 'heat'], ea.STATE),

    e.enum('mode', ea.STATE_SET, ['comfort', 'eco'])
	.withDescription('Hysteresis - comfort > switches off/on exactly at reached temperature with valve smooth from 0 to 100%, eco > 0.5 degrees above or below, valve either 0 or 100%'),
        
    e.comfort_temperature()
	.withValueMin(18)
	.withValueMax(25)
	.withValueStep(0.5)
	.withDescription('Comfort mode temperature'),
    
    e.eco_temperature()
	.withValueMin(5)
	.withValueMax(16)
	.withValueStep(0.5)
	.withDescription('Eco mode temperature'),

    e.binary('frost_protection', ea.STATE_SET, 'ON', 'OFF')
	.withDescription('When the room temperature is lower than 5 °C (or the choosen one), the valve opens; when the temperature rises to 8 °C, the valve closes'),
    e.numeric('frost_protection_temperature', ea.STATE_SET)
        .withUnit('°C')
        .withValueMin(5).withValueMax(10).withValueStep(0.5)
        .withDescription(''),
    
    e.window_detection(),
[... 87 lines cut ...]
"106":"周程序5", // Weekly program 1
"107":"周程序6", // Weekly program 1
"108":"周程序7", // Weekly program 7
"109":"机型", //model
"110":"Motor thrust",
"111":"Display brightness",
"112":"Software version",
"113":"Screen orientation",
"114":"Valve",
"115":"Switch deviation (energy-saving mode only)",
"116":"电机数据", //motor data
"117":"假期时长", //Holiday duration
"118":"Boost 模式时长", //Boost mode duration
"119":"舒适模式温度", //Comfort mode temperature
"120":"节能模式温度", //Energy saving mode temperature
"121":"防冻模式温度", //Antifreeze mode temperature
"122":"防冻使能", //Antifreeze enable
"123":"气压指数", //pressure index
"124":"临时模式开始时间", //Temporary mode start time
"125":"功能点", //Function point
"126":"app支持特性", //app support features
"127":"System mode"
}
*/
...

closed 2025-03-31 by github-actions[bot]

vjaunet, 2025-11-12:
Hello, No success on my side. nor the battery, nor the schedule seem to work.
Best