Koenkk/zigbee2mqtt #25020: Aqara Thermostat E1: Parsing Zigbee Packet fails

andrekupka, 2024-12-02:
### What happened?

After waiting for a time with an Aqara Thermostat E1 (https://www.zigbee2mqtt.io/devices/SRTS-A01.html) the following error occurs in the logs:
```
[2024-12-02 19:49:59] debug: 	z2m: Received Zigbee message from 'Living Room Right Thermostat', type 'attributeReport', cluster 'manuSpecificLumi', data '{"625":1,"629":4,"636":0}' from endpoint 1 with groupID 0
[2024-12-02 19:49:59] error: 	z2m: Exception while calling fromZigbee converter: Value: '4' not found in: [0, 1]}
[2024-12-02 19:49:59] debug: 	z2m: Error: Value: '4' not found in: [0, 1]
    at getFromLookup (/app/node_modules/zigbee-herdsman-converters/src/lib/utils.ts:679:15)
    at /app/node_modules/zigbee-herdsman-converters/src/lib/lumi.ts:2762:62
    at Array.forEach (<anonymous>)
    at Object.convert (/app/node_modules/zigbee-herdsman-converters/src/lib/lumi.ts:2727:38)
    at Receive.onDeviceMessage (/app/lib/extension/receive.ts:170:51)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:211:23)
    at EventEmitter.emit (node:events:531:35)
    at EventBus.emitDeviceMessage (/app/lib/eventBus.ts:137:22)
    at Controller.<anonymous> (/app/lib/zigbee.ts:127:27)
```
The value "4" for key "629" cannot be parsed by the zigbee-herdsman-converter (629 = 0x275, matches the valve alarm state, see https://github.com/Koenkk/zigbee-herdsman-converters/blob/v20.58.0/src/lib/lumi.ts#L2761-L2763).
The code currently expects the value to be either 0 or 1.

### What did you expect to happen?

The value sent by the thermostat should be parsed correctly without an error.

### How to reproduce it (minimal and precise)

Pair an Aqara Thermostat E1 with zigbee2mqtt in Home Assistant and let it heat until the error occurs randomly.

### Zigbee2MQTT version

1.42.0

### Adapter firmware version

20230507

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus

### Setup

Add-on in Home-Assistent on Raspberry Pi 4

### Debug log

[log1.log](https://github.com/user-attachments/files/17983527/log1.log)

Ra72xx, 2024-12-02:
Maybe the same issue: I cannot re-pair those thermostats any more after the update to 1.42.0

phil-schneider, 2024-12-11:
> Maybe the same issue: I cannot re-pair those thermostats any more after the update to 1.42.0

I think this is a general issue with 1.42.0
I have the same problem

Koenkk, 2024-12-13:
When it sends `4`, is the valve_alarm on or off?

andrekupka, 2025-01-07:
The valve alarm was activated when it sent 4.
(Un-)luckily I had to replace the batteries of the affected thermostat.
Since then the issue is gone.

Koenkk, 2025-01-07:
Thanks, fixed the issue.

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2025-01-07 by Koenkk