Koenkk/zigbee2mqtt #25023: [New device support]: GreenPower Schneider Electric A9MEM1563

pawpro, 2024-12-02:
### Link

https://www.se.com/uk/en/product/A9MEM1563/energy-sensor-powertag-phaseneutral-63a-1p+n-bottom-position-for-rcbo-18mm-slim/

### Database entry

{"id":16,"type":"GreenPower","ieeeAddr":"0x00000000e204de39","nwkAddr":56889,"modelId":"GreenPower_254","epList":[242],"endpoints":{"242":{"epId":242,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"interviewCompleted":true,"meta":{},"lastSeen":1733167153824}

### Zigbee2MQTT version

1.40.0

### Comments

The device gets recognized as [A9MEM1570](https://www.zigbee2mqtt.io/devices/A9MEM1570.html#schneider%2520electric-a9mem1570) which is not entirely correct. The problem is there are no values. All the current, power, voltage etc. vales are empty.

I am using Tradfri [LED2103G5](https://www.zigbee2mqtt.io/devices/LED2103G5.html#ikea-led2103g5) as GP proxy.

### External definition

```shell
const definition = {
    zigbeeModel: ['GreenPower_254'],
    model: 'GreenPower_254',
    vendor: '',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

Only Linkquality works.

rbswift, 2025-01-05:
I have two powertags - one is a bit older and it works fine with all the data updating. The newer one has the same problem you describe: Link quality but nothing else. This is the error I get:

```
z2m: Request 'zigbee2mqtt/bridge/request/device/interview' failed with error: 'interview of 'powertag3ph' (0x00000000e2026622) failed: Error: Interview failed because can not get node descriptor ('0x00000000e2026622')'
z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/interview', payload '{"data":{},"error":"inz2m: Request 'zigbee2mqtt/bridge/request/device/interview' failed with error: 'interview of 'powertag3ph' (0x00000000e2026622) failed: Error: Interview failed because can not get node descriptor ('0x00000000e2026622')'
z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/interview', payload '{"data":{},"error":"interview of 'powertag3ph' (0x00000000e2026622) failed: Error: Interview failed because can not get node descriptor ('0x00000000e2026622')","status":"error","transaction":"io3sc-2"}'terview of 'powertag3ph' (0x00000000e2026622) failed: Error: Interview failed because can not get node descriptor ('0x00000000e2026622')","status":"error","transaction":"io3sc-2"}'
```

rbswift, 2025-03-07:
not stale - still ongoing. Can i provide any more useful information / logs?

KCarlsen, 2025-03-30:
i also have a one phase powertag. works fine with an aeotec repeater. however it is discovered as A9MEM1570 as well.

sShrek1, 2025-08-04:
> still an issue, would like to know how to identify the different devices

I think you need to enable "debug" in configuration file and then re-interview the device. Information will be in the logs

Compare the 1 phase and 3-phase and you will see a difference, you should be able to create custom converter and make it work. 

Mine 1-phase pairs but exposes extra values (power_b, power_c... which is 0 always).