Koenkk/zigbee2mqtt #25027: Devices paired between 1.42.0-1 and -2 missing

ki4hrg, 2024-12-03:
### What happened?

I had a device that I paired after upgrading to 1.42.0-1 and before upgrading to the -2 release earlier today that disappeared following the upgrade. Admittedly a bit of an edge case but could see the root issue causing issues down the line. 

### Zigbee2MQTT version

1.42.0-2

### Adapter firmware version

7.1.1.0 build 273

### Adapter

EZSP v9

### Setup

Add-on on HAOS on HA Green

jamiegt10, 2024-12-03:
I have the same issue. i re-added all my devices after the 1.42.0-1 issue and now after the upgrade to -2 all are no longer there.

Koenkk, 2024-12-03:
See https://github.com/zigbee2mqtt/hassio-zigbee2mqtt/issues/664#issuecomment-2514982450, lets' continue there.

closed 2024-12-03 by Koenkk