Koenkk/zigbee2mqtt #25028: [Wrong device]: Waterleak Sensor SQ510A now a sonoff SNZB-03 again

LinuxUnix80, 2024-12-03:
### Link

https://de.aliexpress.com/item/1005006966953116.html?gatewayAdapt=glo2deu

### Model

SQ510A

### Description

Water Leak  Sensor

### Vendor

eWelink DYGSM

### Picture (link)

https://www.zigbee2mqtt.io/devices/SQ510A.html

### Database entry

{"id":19,"type":"EndDevice","ieeeAddr":"0xa4c138b929a3f0f2","nwkAddr":64353,"manufId":4742,"manufName":"eWeLink","powerSource":"Battery","modelId":"SNZB-03","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,3,1,1280,32],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"SNZB-03","manufacturerName":"eWeLink","powerSource":3,"zclVersion":3,"appVersion":113,"stackVersion":2,"hwVersion":0,"swBuildId":"0122052017"}},"ssIasZone":{"attributes":{"iasCieAddr":"0x7cc6b6fffe464990","zoneState":1}},"genPollCtrl":{"attributes":{"checkinInterval":14400}},"genPowerCfg":{"attributes":{"batteryVoltage":30,"batteryPercentageRemaining":200}}},"binds":[{"cluster":32,"type":"endpoint","deviceIeeeAddress":"0x7cc6b6fffe464990","endpointID":1},{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x7cc6b6fffe464990","endpointID":1}],"configuredReportings":[{"cluster":1,"attrId":32,"minRepIntval":3600,"maxRepIntval":7200,"repChange":0},{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":7200,"repChange":0}],"meta":{}}},"appVersion":113,"stackVersion":2,"hwVersion":0,"swBuildId":"0122052017","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-268385843},"lastSeen":1733198689082,"checkinInterval":3600}

### Notes

Hi,

i dont know, if its come with the Update to 1.42.0, but now the Waterleak Sensor is now a SNZB-03 again. See #24652

https://github.com/Koenkk/zigbee2mqtt/issues/24652

Even the Site is missing: https://www.zigbee2mqtt.io/devices/SQ510A.html

jojoferari, 2024-12-03:
I am having the same issue.  New here.  How can I help?

closed 2025-02-09 by github-actions[bot]

jojoferari, 2025-03-28:
I am still able to use it in HA via z2m.  It is quirky, but it works.