Koenkk/zigbee2mqtt #25029: Zigbee2MQTT container starts and stops after update

wachtell, 2024-12-03:
### What happened?

My Zigbee2MQTT was working but has now stopped. It continues to restart:
I am getting the following error in the logs (see below)

I did add "adapter: zstack" to the config file, but that did not help. I am using a SONOFF ZB Dongle-P Zigbee 3.0 USB Dongle Plus adapter.
Thank you!

### What did you expect to happen?

Zigbee2MQTT container starrted

### How to reproduce it (minimal and precise)

It starts and stops again

### Zigbee2MQTT version

1.19.0.

### Adapter firmware version

20221102

### Adapter

SONOFF ZB Dongle-P Zigbee 3.0 USB Dongle Plus

### Setup

Docker on Synology

### Debug log

    at Zigbee.start (/app/lib/zigbee.ts:69:27)

    at Controller.start (/app/lib/controller.ts:139:13)

    at start (/app/index.js:161:5)

[2024-12-03 08:44:30] error: 	z2m: If this happens after updating to Zigbee2MQTT 2.0.0, see https://github.com/Koenkk/zigbee2mqtt/discussions/24364

Using '/app/data' as data directory

Starting Zigbee2MQTT without watchdog.

[2024-12-03 08:45:06] info: 	z2m: Logging to console, file (filename: log.log)

[2024-12-03 08:45:06] debug: 	z2m: Loaded state from file /app/data/state.json

[2024-12-03 08:45:06] info: 	z2m: Starting Zigbee2MQTT version 1.42.0-dev (commit #acba533397dbfaf54f26a9bc5c5b199e46f2a0f5)

[2024-12-03 08:45:06] debug: 	z2m: sd-notify loaded

[2024-12-03 08:45:06] info: 	z2m: Starting zigbee-herdsman (3.0.0)

[2024-12-03 08:45:06] debug: 	z2m: Using zigbee-herdsman with settings: '"{\"network\":{\"panID\":6754,\"extendedPanID\":[221,221,221,221,221,221,221,221],\"channelList\":[11],\"networkKey\":\"HIDDEN\"},\"databasePath\":\"/app/data/database.db\",\"databaseBackupPath\":\"/app/data/database.db.backup\",\"backupPath\":\"/app/data/coordinator_backup.json\",\"serialPort\":{\"path\":\"/dev/ttyACM0\"},\"adapter\":{\"disableLED\":false}}"'

[2024-12-03 08:45:08] debug: 	zh:adapter:discovery: Connected devices: [{"path":"/dev/ttyS0"},{"path":"/dev/ttyS1"},{"path":"/dev/ttyS2"},{"path":"/dev/ttyS3"},{"path":"/dev/ttyUSB0"},{"path":"/dev/ttyUSB0"}]

[2024-12-03 08:45:08] error: 	z2m: Error while starting zigbee-herdsman

[2024-12-03 08:45:08] error: 	z2m: Failed to start zigbee-herdsman

[2024-12-03 08:45:08] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions

[2024-12-03 08:45:08] error: 	z2m: Exiting...

[2024-12-03 08:45:09] error: 	z2m: Error: USB adapter discovery error (No valid USB adapter found). Specify valid 'adapter' and 'port' in your configuration.

    at discoverAdapter (/app/node_modules/.pnpm/zigbee-herdsman@3.0.0/node_modules/zigbee-herdsman/src/adapter/adapterDiscovery.ts:524:15)

    at processTicksAndRejections (node:internal/process/task_queues:95:5)

    at Function.create (/app/node_modules/.pnpm/zigbee-herdsman@3.0.0/node_modules/zigbee-herdsman/src/adapter/adapter.ts:68:33)

    at Controller.start (/app/node_modules/.pnpm/zigbee-herdsman@3.0.0/node_modules/zigbee-herdsman/src/controller/controller.ts:133:24)

    at Zigbee.start (/app/lib/zigbee.ts:69:27)

    at Controller.start (/app/lib/controller.ts:139:13)

    at start (/app/index.js:161:5)

[2024-12-03 08:45:09] error: 	z2m: If this happens after updating to Zigbee2MQTT 2.0.0, see https://github.com/Koenkk/zigbee2mqtt/discussions/24364
Using '/app/data' as data directory

Koenkk, 2024-12-03:
> adapter: zstack

Did you add this under the `serial` section?

wachtell, 2024-12-03:
Yes, I did

Koenkk, 2024-12-03:
could you provide your comolete configuration.yaml?

wachtell, 2024-12-04:
homeassistant: true
permit_join: true
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://192.168.0.21
serial:
  port: /dev/ttyACM0
  adapter: zstack
frontend:
  port: 8020
  ssl_cert: ./data/fullchain.pem
  ssl_key: ./data/privkey.pem
advanced:
  network_key:
    - 193
    - 114
    - 1
    - 219
    - 33
    - 104
    - 108
    - 16
    - 46
    - 253
    - 40
    - 90
    - 70
    - 239
    - 155
    - 142
  homeassistant_legacy_entity_attributes: false
  homeassistant_legacy_triggers: false
  legacy_api: false
  legacy_availability_payload: false
  log_level: debug
  last_seen: ISO_8601  
  
device_options:
  legacy: false
  homeassistant:
    last_seen:
      enabled_by_default: true
devices:
...

wachtell, 2024-12-04:
homeassistant: true
permit_join: true
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://192.168.0.21
serial:
  port: /dev/ttyACM0
  adapter: zstack
frontend:
  port: 8020
  ssl_cert: ./data/fullchain.pem
  ssl_key: ./data/privkey.pem
advanced:
  network_key:
    - 193
    - 114
    - 1
    - 219
    - 33
    - 104
    - 108
    - 16
    - 46
    - 253
    - 40
    - 90
    - 70
    - 239
    - 155
    - 142
  homeassistant_legacy_entity_attributes: false
  homeassistant_legacy_triggers: false
  legacy_api: false
  legacy_availability_payload: false
  log_level: debug
  last_seen: ISO_8601  
  
device_options:
  legacy: false
  homeassistant:
    last_seen:
      enabled_by_default: true
devices:

Koenkk, 2024-12-04:
I think `/dev/ttyACM0` is not correct? Try with `/dev/ttyUSB0`

wachtell, 2024-12-06:
No, unfortunately the port name is correct and has worked for years until the recent update. Changing it does not help. Do you have other suggestions?
Thank you in advance for all you do!
Kind regards

Koenkk, 2024-12-09:
Can you try to not put anything under `serial` does that work?

wachtell, 2024-12-10:
No, unfortunately. It starts and stops again continuously. Here is the log:
[image]

wachtell, 2024-12-11:
Could it be that the SONOFF ZB Dongle-P Zigbee 3.0 USB Dongle Plus is not recognized with the key zstack?

Koenkk, 2024-12-13:
Is `adapter: zstack` correctly indented? I would expect to see it here: `,"serialPort":{"path":"/dev/ttyACM0"}`. Your config should be:

````
serial:
  port: /dev/ttyUSB0
  adapter: zstack
``

(the spaces matter!)

ZenoBell, 2025-01-03:
I have the same issue , restoring the backup of  1.42.0-2  fixes the issue for me. I'm using the SONOFF Zigbee 3.0 USB Dongle Plus, Zigbee Gateway TI CC2652P. Maybe a firmware update is needed but didnt have time to try out cause the entire family was lamenting that the entire house was without light :-) Lesson learned , no updates on critical times. hehe

slado, 2025-01-04:
I have similar issue. After upgrading zigbee2mqtt to 2.0.0-1 it fails to start. I put everything in the configuration file. I'm using Sonoff Zigbee 3.0 USB Dongle Plus as well.

`serial:
  port: /dev/serial/by-id/usb-1a86_USB_Serial-if00-port0
  adapter: zstack
`

and after starting it stops at this point and does not continue
`
[2025-01-04 13:25:20] debug: 	z2m: Removing old log directory '/share/zigbee2mqtt/log/2025-01-04.12-55-07'
[2025-01-04 13:25:20] info: 	z2m: Logging to console, file (filename: log.log)
[2025-01-04 13:25:20] debug: 	z2m: Loaded state from file /share/zigbee2mqtt/state.json
[2025-01-04 13:25:20] info: 	z2m: Starting Zigbee2MQTT version 2.0.0 (commit #unknown)
[2025-01-04 13:25:20] debug: 	z2m: sd-notify loaded
[2025-01-04 13:25:20] info: 	z2m: Starting zigbee-herdsman (3.2.1)
[2025-01-04 13:25:20] debug: 	z2m: Using zigbee-herdsman with settings: '"{\"network\":{\"panID\":6754,\"extendedPanID\":[221,221,221,221,221,221,221,221],\"channelList\":[11],\"networkKey\":\"HIDDEN\"},\"databasePath\":\"/share/zigbee2mqtt/database.db\",\"databaseBackupPath\":\"/share/zigbee2mqtt/database.db.backup\",\"backupPath\":\"/share/zigbee2mqtt/coordinator_backup.json\",\"serialPort\":{\"path\":\"/dev/ttyUSB0\",\"adapter\":\"zstack\"},\"adapter\":{\"disableLED\":false,\"transmitPower\":15}}"'
[2025-01-04 13:25:21] debug: 	zh:adapter:discovery: Connected devices: [{"path":"/dev/ttyUSB0","manufacturer":"1a86","pnpId":"usb-1a86_USB_Serial-if00-port0","vendorId":"1a86","productId":"7523"}]
[2025-01-04 13:25:21] debug: 	zh:adapter:discovery: Unable to match USB adapter: zstack | /dev/ttyUSB0
[2025-01-04 13:25:21] debug: 	zh:controller: Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":"HIDDEN","panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"path":"/dev/ttyUSB0","adapter":"zstack"},"adapter":{"disableLED":false,"transmitPower":15},"databasePath":"/share/zigbee2mqtt/database.db","databaseBackupPath":"/share/zigbee2mqtt/database.db.backup","backupPath":"/share/zigbee2mqtt/coordinator_backup.json"}'
[2025-01-04 13:25:21] info: 	zh:zstack:znp: Opening SerialPort with {"path":"/dev/ttyUSB0","baudRate":115200,"rtscts":false,"autoOpen":false}

`

wachtell, 2025-01-05:
I have simply not been able to solve this yet with the  SONOFF Zigbee 3.0 USB Dongle Plus, Zigbee Gateway TI CC2652P - maybe we need an update. Can you help?

wachtell, 2025-01-05:
> Is `adapter: zstack` correctly indented? I would expect to see it here: `,"serialPort":{"path":"/dev/ttyACM0"}`. Your config should be:
> 
> ```
> serial:
>   port: /dev/ttyUSB0
>   adapter: zstack
> ```
> 
> (the spaces matter!)

and my spaces are OK.

sceiler, 2025-01-05:
I have the same issue after an auto-update to 2.0.0-1.

My adapter is SONOFF ZigBee 3.0 Dongle Plus (CC2652P, https://sonoff.tech/product/gateway-and-sensors/sonoff-zigbee-3-0-usb-dongle-plus-p/)

Log

```
[14:13:43] INFO: Preparing to start...
[14:13:43] INFO: Socat not enabled
[14:13:45] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2025-01-05 14:13:51] debug: 	z2m: Removing old log directory '/config/zigbee2mqtt/log/2025-01-05.14-07-02'
[2025-01-05 14:13:51] info: 	z2m: Logging to console, file (filename: log.log)
[2025-01-05 14:13:51] debug: 	z2m: Loaded state from file /config/zigbee2mqtt/state.json
[2025-01-05 14:13:51] info: 	z2m: Starting Zigbee2MQTT version 2.0.0 (commit #unknown)
[2025-01-05 14:13:51] debug: 	z2m: sd-notify loaded
[2025-01-05 14:13:51] info: 	z2m: Starting zigbee-herdsman (3.2.1)
[2025-01-05 14:13:51] debug: 	z2m: Using zigbee-herdsman with settings: '"{\"network\":{\"panID\":6754,\"extendedPanID\":[221,221,221,221,221,221,221,221],\"channelList\":[11],\"networkKey\":\"HIDDEN\"},\"databasePath\":\"/config/zigbee2mqtt/database.db\",\"databaseBackupPath\":\"/config/zigbee2mqtt/database.db.backup\",\"backupPath\":\"/config/zigbee2mqtt/coordinator_backup.json\",\"serialPort\":{\"path\":\"/dev/serial/by-id/usb-Silicon_Labs_Sonoff_Zigbee_3.0_USB_Dongle_Plus_0001-if00-port0\",\"adapter\":\"zstack\"},\"adapter\":{\"disableLED\":false}}"'
[2025-01-05 14:13:51] debug: 	zh:adapter:discovery: Connected devices: [{"path":"/dev/ttyACM0","manufacturer":"dresden elektronik ingenieurtechnik GmbH","serialNumber":"DE2147142","pnpId":"usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2147142-if00","vendorId":"1cf1","productId":"0030"},{"path":"/dev/ttyUSB0","manufacturer":"Silicon Labs","serialNumber":"0001","pnpId":"usb-Silicon_Labs_Sonoff_Zigbee_3.0_USB_Dongle_Plus_0001-if00-port0","vendorId":"10c4","productId":"ea60"}]
[2025-01-05 14:13:51] debug: 	zh:adapter:discovery: Unable to match USB adapter: zstack | /dev/serial/by-id/usb-Silicon_Labs_Sonoff_Zigbee_3.0_USB_Dongle_Plus_0001-if00-port0
[2025-01-05 14:13:51] debug: 	zh:controller: Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":"HIDDEN","panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"path":"/dev/serial/by-id/usb-Silicon_Labs_Sonoff_Zigbee_3.0_USB_Dongle_Plus_0001-if00-port0","adapter":"zstack"},"adapter":{"disableLED":false},"databasePath":"/config/zigbee2mqtt/database.db","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","backupPath":"/config/zigbee2mqtt/coordinator_backup.json"}'
[2025-01-05 14:13:51] info: 	zh:zstack:znp: Opening SerialPort with {"path":"/dev/serial/by-id/usb-Silicon_Labs_Sonoff_Zigbee_3.0_USB_Dongle_Plus_0001-if00-port0","baudRate":115200,"rtscts":false,"autoOpen":false}
```

Config YAML

```
data_path: /config/zigbee2mqtt
socat:
  enabled: false
  master: pty,raw,echo=0,link=/tmp/ttyZ2M,mode=777
  slave: tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5
  options: "-d -d"
  log: false
mqtt: {}
serial:
  port: >-
    /dev/serial/by-id/usb-Silicon_Labs_Sonoff_Zigbee_3.0_USB_Dongle_Plus_0001-if00-port0
  adapter: zstack
```

Troubleshooting steps I tried:

- Unplug and plug dongle
- Restart Raspberry Pi 4 where HA is running
- Tried `Device path: /dev/ttyUSB0` and `Device ID: /dev/serial/by-id/usb-Silicon_Labs_Sonoff_Zigbee_3.0_USB_Dongle_Plus_0001-if00-port0`

sceiler, 2025-01-05:
https://github.com/Koenkk/zigbee2mqtt/issues/25464#issuecomment-2571624389 seems to have fixed it for me. Trying more...

wachtell, 2025-01-15:
I am still trying to get this to work.
1) I updated the SONOFF ZB Dongle-P Zigbee 3.0 USB Dongle Plus to the latest[ firmware] CC1352P2_CC2652P_launchpad_coordinator_20240710 (https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/Z-Stack_3.x.0/bin/CC1352P2_CC2652P_launchpad_coordinator_20240710.zip).
2) I upgraded the Docker Image to Zigbee2MQTT version 2.0.0-dev.
3) I secured the config file has indentations:
```
 serial:
  adaptor: zstack
  port: /dev/ttyUSB0/ 

```4) I did an `lsusb -cui` and found the dongle on: 
```
|__7-2         10c4:ea60:0100 00  1.10   12MBit/s 100mA 1IF  (Silicon Labs Sonoff Zigbee 3.0 USB Dongle Plus 0001)
  7-2:1.0         (IF) ff:00:00 2EPs () cp210x ttyUSB0

```5) I ensured that the permissions with `sudo chmod 666 /dev/ttyUSB0`

Still, I don't have any connection:

[image]

I am not sure what to more. Do you have any suggestions?

wachtell, 2025-01-17:
> [#25464 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/25464#issuecomment-2571624389) seems to have fixed it for me. Trying more...

Exactly what from the comment fixed it for you?

closed 2025-03-28 by github-actions[bot]