Koenkk/zigbee2mqtt #25030: "throttle" not throttling (some power meter are too talkative)

happyjaxx, 2024-12-03:
### What happened?

Hello !

Updated to 1.40.1 (and 1.40.2 since yesterday with a bit of hope) with the intent of using the 'thottle' (from pull #24122 ) on two _TZE204_ac0fhfiq TS0601_bidirectional_energy meters I set up a month ago to stat some energy in and outs

```
./update.sh
[....blabla...]
Building...

> zigbee2mqtt@1.42.0 build
> tsc && node index.js writehash

Restore configuration...
Starting Zigbee2MQTT...
Done!
```

(for context, one is set at the edge of the 'household' after the main grid meter, the other at the leads to the solar panels on the roof, these clamps have a dead band of +/- 300VA of realtime power but cumulative kWh energy metrics are good enough)

The clamps are very talkative and average a message on the zigbee network every 0.2 to 3 seconds on average
[image]

Issue being: I have HASS with the homekit addon and I strongly suspect that these overly talkative meters are generating enough homekit traffic to drain ios/watchos devices (I used homebridge before switching to HASS around the same time, but it was struggling, maybe for the same reasons, dunno, that wasn't very log verbose)... I'm kind of fed up to have to charge my phone and watch more than twice a day... So when I saw the throttle option, I went for it :) (I'd actually end up filtering the meters in HASS if I knew how I could)

Thanks in advance :)

### What did you expect to happen?

Having the 'throttle' (without 'debounce') set in the config:

```
  '0xa4c138c1700ba5ee':
    friendly_name: power_meter_main
    throttle: 10
    homeassistant: {}
    optimistic: true
  '0xa4c13873b945ccd4':
    friendly_name: power_meter_solar
    homeassistant: {}
    optimistic: true
    throttle: 10
```

I expected to see messages no more often than every ten seconds... but it's still coming it faster than wanted (screenshot above)

### Zigbee2MQTT version

1.40.2

### Adapter firmware version

20240710

### Adapter

Sonoff Zigbee Plus

### Setup

Debian container on Proxmox (with usb binding), mqtt on an Alpine Container

### Debug log

[log.log](https://github.com/user-attachments/files/17989604/log.log)

RobLeighton67, 2024-12-04:
Throttle is in 1.41.0 - the versions you mention precede the change ?

RobLeighton67, 2024-12-04:
If the problem persists in 1.41.0 make sure that you don;t have a global 'debounce' setting as this takes preference.

happyjaxx, 2024-12-04:
Well thought, but unfortunatly no global debounce to be found
```
zigbee@z2m:/opt/zigbee2mqtt/data$ grep debounce configuration.yaml
zigbee@z2m:/opt/zigbee2mqtt/data$
```

mociulski, 2024-12-11:
I do experience the same issue, tried with 1.41.0 and 1.42.0 as well:

[image]
`          "0x90395efffe896fc9": {
                "energy_precision": 0,
                "friendly_name": "Climate Energy",
                "homeassistant": {},
                "optimistic": true,
                "throttle": 10
            },`

I do not have debounce set globally.

voidzero, 2025-02-13:
Same, I set throttle to 60, but it still spams.

closed 2025-04-23 by github-actions[bot]