Koenkk/zigbee2mqtt #25031: Linux fresh-installation instructions using pnpm are incorrect

habitats-tech, 2024-12-03:
### What happened?

instructions under https://github.com/Koenkk/zigbee2mqtt.io/blob/dev/docs/guide/installation/01_linux.md are problematic as they fail to create the installation

### What did you expect to happen?

some commands require finetuning

### How to reproduce it (minimal and precise)

`pnpm i --frozen-lockfile` command fails to execute; run this instead `pnpm install --no-frozen-lockfile`
`pnpm run build` command fails with mqtt related issues; 

lib/mqtt.ts:1:24 - error TS2307: Cannot find module 'mqtt-packet' or its corresponding type declarations.

1 import type {QoS} from 'mqtt-packet';
                         ~~~~~~~~~~~~~

lib/types/types.d.ts:8:24 - error TS2307: Cannot find module 'mqtt-packet' or its corresponding type declarations.

8 import type {QoS} from 'mqtt-packet';
                         ~~~~~~~~~~~~~

Found 2 errors in 2 files.

Errors  Files
     1  lib/mqtt.ts:1
     1  lib/types/types.d.ts:8
 ELIFECYCLE  Command failed with exit code 2.

### Zigbee2MQTT version

1.42

### Adapter firmware version

N/A

### Adapter

N/A

### Setup

Debian 12 LXC under Proxmox 8

### Debug log

lib/mqtt.ts:1:24 - error TS2307: Cannot find module 'mqtt-packet' or its corresponding type declarations.

1 import type {QoS} from 'mqtt-packet';
                         ~~~~~~~~~~~~~

lib/types/types.d.ts:8:24 - error TS2307: Cannot find module 'mqtt-packet' or its corresponding type declarations.

8 import type {QoS} from 'mqtt-packet';
                         ~~~~~~~~~~~~~

Found 2 errors in 2 files.

Errors  Files
     1  lib/mqtt.ts:1
     1  lib/types/types.d.ts:8
 ELIFECYCLE  Command failed with exit code 2.

habitats-tech, 2024-12-03:
installed `npm install mqtt-packet --save` and problem solved

Nerivec, 2024-12-03:
Looks like you may be mixing `dev` branch with `master` branch instructions.
Which one are you using? If using `master`, it has not yet migrated to `pnpm` and you shouldn't be referring to `dev` docs.

`pnpm i --frozen-lockfile` is the correct way, as you otherwise will install updated dependencies that Z2M may not have support for.
`mqtt-packet` dependency was removed in `dev` branch.

closed 2024-12-05 by Nerivec

habitats-tech, 2024-12-06:
I ended up creating a new fresh instance dedicated to dev. The guide on the Z2M website for Linux installation is incomplete. I have created a complete version which will publish once v2 ships.