Koenkk/zigbee2mqtt #25033: PTM 215Z action not going to HA

RuneNyhuus, 2024-12-03:
### What happened?

I've updated to 1.42.0-dev commit: [acba533](https://github.com/Koenkk/zigbee2mqtt/commit/acba533)
And wanted to add PTM 215Z, and everything looks fine in z2m, but action is not exposed to HA

### What did you expect to happen?

To see action in HA

### How to reproduce it (minimal and precise)

Just delete it and readd, then its gone from HA

### Zigbee2MQTT version

1.42.0

### Adapter firmware version

7.4.5 [GA]

### Adapter

SLZB-06M

### Setup

Home Assistant Add-on: Zigbee2MQTT Edge

### Debug log

[log.log](https://github.com/user-attachments/files/17993034/log.log)

ncrmnt, 2024-12-09:
I faced with the same problem on 1.42.0
I've rolled back to 1.40.2 and now all my buttons ([SNZB-01](https://www.zigbee2mqtt.io/devices/SNZB-01.html#sonoff-snzb-01), [E1810](https://www.zigbee2mqtt.io/devices/E1524_E1810.html#ikea-e1524%252Fe1810)) are working again.

Toniob, 2024-12-22:
Same issue on 1.42.0 with some switches (QBKG40LM). The action value is not present in HA anymore.

kds69, 2025-01-05:
please check on this, and look for « action »: [https://github.com/Koenkk/zigbee2mqtt/discussions/24198](url)

closed 2025-01-08 by RuneNyhuus