Koenkk/zigbee2mqtt #25038: [New device support]: Moes Thermostat - _TZE204_lpedvtvr

TesterTurner, 2024-12-03:
### Link

https://www.aliexpress.com/item/1005008017004760.html?src=google&pdp_npi=4%40dis%21GBP%2170.30%2130.29%21%21%21%21%21%40%2112000043377898982%21ppc%21%21%21&src=google&albch=shopping&acnt=494-037-6276&isdl=y&slnk=&plac=&mtctp=&albbt=Google_7_shopping&aff_platform=google&aff_short_key=UneMJZVf&gclsrc=aw.ds&&albagn=888888&&ds_e_adid=&ds_e_matchtype=&ds_e_device=c&ds_e_network=x&ds_e_product_group_id=&ds_e_product_id=en1005008017004760&ds_e_product_merchant_id=680478118&ds_e_product_country=GB&ds_e_product_language=en&ds_e_product_channel=online&ds_e_product_store_id=&ds_url_v=2&albcp=17859500389&albag=&isSmbAutoCall=false&needSmbHouyi=false&gad_source=1&gclid=CjwKCAiA9bq6BhAKEiwAH6bqoMyzGa27lfVQmlZ0p5iPXlytCHGEa1BQZJNKFUxZJDIftZi1K48KrBoCDSAQAvD_BwE

### Database entry

{"id":41,"type":"Router","ieeeAddr":"0xa4c1383ebb899b92","nwkAddr":57155,"manufId":4417,"manufName":"_TZE204_lpedvtvr","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\u0007\u000b�.i\u0007\u000b�.i","65506":56,"65508":0,"stackVersion":0,"dateCode":"","appVersion":74}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1733251203681}

### Zigbee2MQTT version

1.42.0-2

### Comments

Good Evening, I am struggling to get a working converter for this new thermostat released by Moes. Am unable to locate anything similar to attempt to clone or Frankenstein to get basic functionality. I apologise, I don't seem to have the required skill set to get it running, it has provided no information under the 'Expose' Tab, Could you please assist.  

### External definition

```shell
const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE204_lpedvtvr',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

the external definition only provides link quality, No further functionality.

TesterTurner, 2024-12-06:
I have still not had any luck so far trying to use an alternate configuration to get basic functionality from the thermostat.

Racermouse, 2024-12-06:
You are not alown!

Racermouse, 2024-12-07:
My coding skills are very weak, and could not get this to work either.
Here's all I could find out from the z2m debug log:

Temp set point:
'Kitchen_Floor_Therm', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,205],"type":"Buffer"},"datatype":2,"dp":50}],"seq":4352}' from endpoint 1 with groupID 0

Internal temp sensor reading:
'Kitchen_Floor_Therm', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,180],"type":"Buffer"},"datatype":2,"dp":16}],"seq":15616}' from endpoint 1 with groupID 0

External temp sensor reading:
'Kitchen_Floor_Therm', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,195],"type":"Buffer"},"datatype":2,"dp":109}],"seq":9984}' from endpoint 1 with groupID 0

Heating on:
'Kitchen_Floor_Therm', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":47}],"seq":26112}' from endpoint 1 with groupID

Heating off:
'Kitchen_Floor_Therm', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":4,"dp":47}],"seq":29184}' from endpoint 1 with groupID 0

ON/OFF

On:
'Kitchen_Floor_Therm', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":1,"dp":1}],"seq":23040}' from endpoint 1 with groupID 0

Off:
'Kitchen_Floor_Therm', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":1,"dp":1}],"seq":34048}' from endpoint 1 with groupID 0

WORK MODE

Manual:
'Kitchen_Floor_Therm', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":2}],"seq":60672}' from endpoint 1 with groupID 0

Temporary manual:
'Kitchen_Floor_Therm', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":4,"dp":2}],"seq":51968}' from endpoint 1 with groupID 0

Programming
'Kitchen_Floor_Therm', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[2],"type":"Buffer"},"datatype":4,"dp":2}],"seq":46336}' from endpoint 1 with groupID 0

Eco:
'Kitchen_Floor_Therm', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[3],"type":"Buffer"},"datatype":4,"dp":2}],"seq":42240}' from endpoint 1 with groupID 0

SCREEN BRIGHTNESS

Standby brightness adjustment:
'Kitchen_Floor_Therm', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,66],"type":"Buffer"},"datatype":2,"dp":48}],"seq":4608}' from endpoint 1 with groupID 0 

CHILD LOCK

Child lock on:
'Kitchen_Floor_Therm', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":1,"dp":39}],"seq":13312}' from endpoint 1 with groupID 0

Child lock off:
'Kitchen_Floor_Therm', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":1,"dp":39}],"seq":16128}' from endpoint 1 with groupID 0

SENSOR TYPE

In:
'Kitchen_Floor_Therm', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":32}],"seq":30208}' from endpoint 1 with groupID 0

AL:
'Kitchen_Floor_Therm', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":4,"dp":32}],"seq":35328}' from endpoint 1 with groupID 0

Ou:
'Kitchen_Floor_Therm', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[2],"type":"Buffer"},"datatype":4,"dp":32}],"seq":37376}' from endpoint 1 with groupID 0

What the hell is this:
'Kitchen_Floor_Therm', type 'attributeReport', cluster 'genBasic', data '{"65503":"8\u000f .iF\u000f .iG\u000f .iL\u000f .i[\u000f .if\u000f .ig\u000f .ip\u000f .iv\u000f .iw\u000f .i"}' from endpoint 1 with groupID 0

This Is from startup:

'Kitchen_Floor_Therm', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,1,194],"type":"Buffer"},"datatype":2,"dp":34}],"seq":256}' from endpoint 1 with groupID 0
[... 6 lines cut ...]
'Kitchen_Floor_Therm', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,185],"type":"Buffer"},"datatype":2,"dp":16}],"seq":1792}' from endpoint 1 with groupID 0
'Kitchen_Floor_Therm', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":1,"dp":1}],"seq":2048}' from endpoint 1 with groupID 0
'Kitchen_Floor_Therm', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":4,"dp":2}],"seq":2304}' from endpoint 1 with groupID 0
'Kitchen_Floor_Therm', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,250],"type":"Buffer"},"datatype":2,"dp":50}],"seq":2560}' from endpoint 1 with groupID 0
'Kitchen_Floor_Therm', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":47}],"seq":2816}' from endpoint 1 with groupID 0
[... cut ...] upID 0
'Kitchen_Floor_Therm', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[6,6,0,0,200,11,30,0,200,13,30,0,200,17,30,0,200],"type":"Buffer"},"datatype":0,"dp":107}],"seq":7424}' from endpoint 1 with groupID 0
'Kitchen_Floor_Therm', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[7,6,0,0,200,11,30,0,200,13,30,0,200,17,30,0,200],"type":"Buffer"},"datatype":0,"dp":108}],"seq":7680}' from endpoint 1 with groupID 0

magic1447, 2024-12-07:
Hello! Im also very interested in support for this device!

Racermouse, 2024-12-07:
I got it working enough to do the stuff I need to do in Node-Red.
I'll post the converter file later tonight.
Be warned that It is very untidy and contains stuff that does not work.

TesterTurner, 2024-12-08:
That zip file you have dropped in flags as containing a virus/malware, so I shall play it safe by not opening it, I'm sure its probably just my system playing it safe however to play on the safe side, could you paste your converter directly in?

Racermouse, 2024-12-08:
Sure!

`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
        fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE204_lpedvtvr'}],
        model: 'ZHT_SR',
        vendor: 'Moes',
        description: 'Thermostat for electric floor',
        fromZigbee: [tuya.fz.datapoints],
        toZigbee: [tuya.tz.datapoints],
        onEvent: tuya.onEventSetTime,
        configure: tuya.configureMagicPacket,
        exposes: [
            e.binary('child_lock', ea.STATE_SET, 'ON', 'OFF').withLabel('Child lock'),
            e.binary('eco_mode', ea.STATE_SET, 'OFF', 'ON').withLabel('ECO mode NOT SET').withDescription('Default: Off'),
            e
                .numeric('eco_temperature', ea.STATE_SET)
                .withValueMin(5)
                .withValueMax(30)
                .withValueStep(1)
                .withUnit('°C')
		.withLabel('Eco temperature')
                .withDescription('Max temperature in ECO mode. Default: 20'),
            e.binary('valve_state', ea.STATE, false, true).withLabel('Heating in process'),
            e
                .climate()
                .withSystemMode(['off', 'heat'], ea.STATE_SET)
                .withPreset(['manual', 'temp_manual', 'programming', 'eco'])
                .withSetpoint('current_heating_setpoint', 5, 35, 0.5, ea.STATE_SET)
                .withLocalTemperature(ea.STATE)
                .withLocalTemperatureCalibration(-9, 9, 1, ea.STATE_SET)
                .withDescription('Default: -3'),
            e
                .numeric('deadzone_temperature', ea.STATE_SET)
                .withValueMin(1)
                .withValueMax(5)
                .withValueStep(1)
                .withUnit('°C')
		.withLabel('Deadzone temperature NOT SET')
                .withDescription('Hysteresis. Default: 1'),

            e.numeric('min_temperature', ea.STATE_SET)
		.withValueMin(5)
		.withValueMax(15)
		.withValueStep(1)
		.withUnit('°C')
		.withLabel('Minimum temperature NOT SET')
		.withDescription('Default: 5'),
            e
                .numeric('max_temperature', ea.STATE_SET)
                .withValueMin(15)
                .withValueMax(45)
                .withValueStep(1)
                .withUnit('°C')
		.withLabel('Maximum temperature NOT SET')
                .withDescription('Default: 35'),
            e
                .numeric('min_temperature_limit', ea.STATE_SET)
                .withValueMin(0)
                .withValueMax(10)
                .withValueStep(1)
                .withUnit('°C')
[... 68 lines cut ...]
                [27, 'local_temperature_calibration', tuya.valueConverter.raw],
                [28, 'factory_reset', tuya.valueConverterBasic.lookup({factory_reset: true})],
                [47, 'valve_state', tuya.valueConverter.trueFalseInvert],
                [39, 'child_lock', tuya.valueConverterBasic.lookup({ON: true, OFF: false})],
                [40, 'eco_mode', tuya.valueConverterBasic.lookup({ON: true, OFF: false})],
                [32, 'sensor', tuya.valueConverterBasic.lookup({IN: tuya.enum(0), OU: tuya.enum(2), AL: tuya.enum(1)})],
                [109, 'external_temperature_input', tuya.valueConverter.divideBy10],
                [110, 'deadzone_temperature', tuya.valueConverter.raw],
                [104, 'max_temperature_limit', tuya.valueConverter.divideBy10],
                [102, 'schedule_monday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(1)],
                [103, 'schedule_tuesday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(2)],
                [104, 'schedule_wednesday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(3)],
                [105, 'schedule_thursday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(4)],
                [106, 'schedule_friday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(5)],
                [107, 'schedule_saturday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(6)],
                [108, 'schedule_sunday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(7)],
                [111, 'min_temperature_limit', tuya.valueConverter.divideBy10],
                [114, 'eco_temperature', tuya.valueConverter.divideBy10],
                [115, 'brightness', tuya.valueConverter.raw],
                [48, 'display_brightness', tuya.valueConverter.raw],
            ],
        },
};

module.exports = definition;`

szheliab, 2024-12-08:
Hi there!

Here are datapoints that I've got from Tuya hub:

```json
{
  "1": "Switch",
  "2": "Work Mode",
  "16": "Current temperature",
  "18": "The lower limit of temperature",
  "28": "Factory data reset",
  "32": "Sensor selection",
  "34": "Set temperature ceiling",
  "39": "Child lock",
  "47": "State of the valve",
  "48": "Backlight brightness",
  "50": "Set temperature",
  "101": "Temp Calibration",
  "102": "Week Program 13 1",
  "103": "Week Program 13 2",
  "104": "Week Program 13 3",
  "105": "Week Program 13 4",
  "106": "Week Program 13 5",
  "107": "Week Program 13 6",
  "108": "Week Program 13 7",
  "109": "Floor temp.",
  "110": "Dead zone temp.",
  "111": "High protect temp.",
  "112": "Low protection temp.",
  "113": "Eco cool temp.",
  "114": "Screen Time Set",
  "115": "Rgblight"
}
```

szheliab, 2024-12-08:
...And here's mine version of external converter for this thermostat 😃 
```javascript
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['TS0601'],
    model: 'ZHT-SR-GB',
    vendor: 'Moes',
    description: 'Smart knob thermostat',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    exposes: [
        e.binary('child_lock', ea.STATE_SET, 'ON', 'OFF').withLabel('Child lock'),
        e.binary('rgb_backlight', ea.STATE_SET, 'OFF', 'ON').withLabel('RGB backlight').withDescription('Default: Off'),
        e.numeric('eco_temperature', ea.STATE_SET)
            .withValueMin(5)
            .withValueMax(30)
            .withValueStep(1)
            .withUnit('°C')
            .withDescription('Max temperature in ECO mode. Default: 20'),
        e.binary('valve_state', ea.STATE, false, true).withLabel('Heating in process'),
        e.climate()
            .withSystemMode(['off', 'heat'], ea.STATE_SET)
            .withPreset(['manual', 'temporary manual', 'schedule', 'eco'])
            .withSetpoint('current_heating_setpoint', 5, 35, 0.5, ea.STATE_SET)
            .withLocalTemperature(ea.STATE)
            .withLocalTemperatureCalibration(-9, 9, 0.1, ea.STATE_SET)
            .withDescription('Default: -3'),
        e.numeric('deadzone_temperature', ea.STATE_SET)
            .withValueMin(1)
            .withValueMax(5)
            .withValueStep(1)
            .withUnit('°C')
            .withDescription('Hysteresis. Default: 1'),
        e.numeric('min_temperature', ea.STATE_SET)
            .withValueMin(5)
            .withValueMax(15)
            .withValueStep(1)
            .withUnit('°C')
            .withDescription('Default: 5'),
        e.numeric('max_temperature', ea.STATE_SET)
            .withValueMin(15)
            .withValueMax(45)
            .withValueStep(1)
            .withUnit('°C')
            .withDescription('Default: 35'),
        e.numeric('min_temperature_limit', ea.STATE_SET)
            .withValueMin(0)
            .withValueMax(10)
            .withValueStep(1)
            .withUnit('°C')
            .withLabel('Low temperature protection')
            .withDescription('Default: 0'),
        e.numeric('lower_limit_of_temperature', ea.STATE_SET)
            .withValueMin(5)
            .withValueMax(15)
            .withValueStep(1)
            .withUnit('°C')
            .withLabel('The lower limit of temperature')
            .withDescription('Default: 5'),
        e.enum('sensor', ea.STATE_SET, ['IN', 'OU', 'AL']).withLabel('Sensor')
            .withDescription('Choose which sensor to use. Default: AL'),
        e.numeric('external_temperature_input', ea.STATE)
            .withLabel('Floor temperature')
            .withUnit('°C')
            .withDescription('Temperature from floor sensor'),
        e.numeric('brightness', ea.STATE_SET)
[... 50 lines cut ...]
            [103, 'schedule_tuesday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(2)],
            [104, 'schedule_wednesday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(3)],
            [105, 'schedule_thursday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(4)],
            [106, 'schedule_friday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(5)],
            [107, 'schedule_saturday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(6)],
            [108, 'schedule_sunday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(7)],
            [109, 'external_temperature_input', tuya.valueConverter.divideBy10],
            [110, 'deadzone_temperature', tuya.valueConverter.raw],
            [111, 'max_temperature_limit', tuya.valueConverter.divideBy10],
            [112, 'min_temperature_limit', tuya.valueConverter.divideBy10],
            [113, 'eco_temperature', tuya.valueConverter.divideBy10],
            [114, 'screen_time_set', tuya.valueConverterBasic.lookup({
                off: tuya.enum(0), 
                '10s': tuya.enum(1), 
                '30s': tuya.enum(2), 
                '60s': tuya.enum(3)
            })],
            [115, 'rgb_backlight', tuya.valueConverterBasic.lookup({ON: true, OFF: false})],
        ],
    },
    onEvent: tuya.onEventSetLocalTime,
};

module.exports = definition;
```

Racermouse, 2024-12-08:
Nice.
I have no Tuya bridge, so thanks for the help!

Racermouse, 2024-12-08:
Screen timeouts are 20, 30, 40 s.

Racermouse, 2024-12-08:
Can't seem to get the "deadzone_temperature" and "local_temperature_calibration" to work properly. Might have something to do with factors?

magic1447, 2024-12-08:
> Can't seem to get the "deadzone_temperature" and "local_temperature_calibration" to work properly. Might have something to do with factors?

I second that. It doesn't work on my one either

szheliab, 2024-12-09:
Unfortunately, cannot find solution to make temp calibration work  😞 

Here is updated converter (fixed default values, min/max values, etc.), can you please try it? 

```javascript
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['TS0601'],
    model: 'ZHT-SR-GB',
    vendor: 'Moes',
    description: 'Smart knob thermostat',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    exposes: [
        e.binary('child_lock', ea.STATE_SET, 'ON', 'OFF').withLabel('Child lock'),
        e.binary('rgb_backlight', ea.STATE_SET, 'ON', 'OFF').withLabel('RGB backlight').withDescription('Default: On'),
        e.numeric('eco_temperature', ea.STATE_SET)
            .withValueMin(10)
            .withValueMax(30)
            .withValueStep(1)
            .withUnit('°C')
            .withDescription('Max temperature in ECO mode. Default: 30'),
        e.binary('valve_state', ea.STATE, false, true).withLabel('Heating in process'),
        e.climate()
            .withSystemMode(['off', 'heat'], ea.STATE_SET)
            .withPreset(['manual', 'temporary manual', 'schedule', 'eco'])
            .withSetpoint('current_heating_setpoint', 5, 45, 0.5, ea.STATE_SET)
            .withLocalTemperature(ea.STATE)
            .withLocalTemperatureCalibration(-10, 10, 1, ea.STATE_SET)
            .withDescription('Default: 22'),
        e.numeric('deadzone_temperature', ea.STATE_SET)
            .withValueMin(0.5)
            .withValueMax(5)
            .withValueStep(0.5)
            .withLabel('Dead zone temperature')
            .withUnit('°C')
            .withDescription('Default: 0.5'),
        e.numeric('max_temperature', ea.STATE_SET)
            .withValueMin(35)
            .withValueMax(45)
            .withValueStep(1)
            .withLabel('Set temperature ceiling')
            .withUnit('°C')
            .withDescription('Default: 45'),
        e.numeric('min_temperature_limit', ea.STATE_SET)
            .withValueMin(0)
            .withValueMax(10)
            .withValueStep(1)
            .withUnit('°C')
            .withLabel('Low temperature protection')
            .withDescription('Default: 0'),
        e.numeric('max_temperature_limit', ea.STATE_SET)
            .withValueMin(10)
            .withValueMax(70)
            .withValueStep(1)
            .withLabel('High temperature protection')
            .withUnit('°C')
            .withDescription('Default: 45'),
        e.numeric('lower_limit_of_temperature', ea.STATE_SET)
            .withValueMin(5)
            .withValueMax(15)
            .withValueStep(1)
            .withUnit('°C')
            .withLabel('The lower limit of temperature')
            .withDescription('Default: 5'),
        e.enum('sensor', ea.STATE_SET, ['IN', 'OU', 'AL']).withLabel('Sensor')
[... 57 lines cut ...]
            [105, 'schedule_thursday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(4)],
            [106, 'schedule_friday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(5)],
            [107, 'schedule_saturday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(6)],
            [108, 'schedule_sunday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(7)],
            [109, 'external_temperature_input', tuya.valueConverter.divideBy10],
            [110, 'deadzone_temperature', tuya.valueConverter.raw],
            [111, 'max_temperature_limit', tuya.valueConverter.divideBy10],
            [112, 'min_temperature_limit', tuya.valueConverter.divideBy10],
            [113, 'eco_temperature', tuya.valueConverter.divideBy10],
            [114, 'screen_time_set', tuya.valueConverterBasic.lookup({
                '10s': tuya.enum(0),
                '20s': tuya.enum(1),
                '30s': tuya.enum(2),
                '40s': tuya.enum(3),
                '50s': tuya.enum(4),
                '60s': tuya.enum(5)
            })],
            [115, 'rgb_backlight', tuya.valueConverterBasic.lookup({ON: true, OFF: false})],
        ],
    },
    onEvent: tuya.onEventSetLocalTime,
};

module.exports = definition;
```

Racermouse, 2024-12-10:
I tried the code and it worked for me anyway.

Some things I changed to my way of using a floor thermostat.
Limits have been change. Don't know what the real device limits are, but the ones in the code below works.
Dead zone (hysteresis) needs to be multiplied by 10 to work. I have no idea how to do it. Can't find a `tuya.valueConverter` to do this.

`const exposes = require('zigbee-herdsman-converters/lib/exposes');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['TS0601'],
    model: 'ZHT-SR-GB',
    vendor: 'Moes',
    description: 'Smart knob thermostat',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    exposes: [

        e.binary('child_lock', ea.STATE_SET, 'ON', 'OFF').withLabel('Child lock'),

        e.binary('rgb_backlight', ea.STATE_SET, 'ON', 'OFF').withLabel('Knob light').withDescription('Default: On'),

        e.numeric('eco_temperature', ea.STATE_SET)
            .withValueMin(10)
            .withValueMax(30)
            .withValueStep(0.5)
            .withUnit('°C')
            .withDescription('Max temperature in ECO mode. Default: 30'),

        e.binary('valve_state', ea.STATE, false, true).withLabel('Heating in process'),

        e.climate()
            .withSystemMode(['off', 'heat'], ea.STATE_SET)
            .withPreset(['manual', 'temporary manual', 'schedule', 'eco'])
            .withSetpoint('current_heating_setpoint', 5, 45, 0.5, ea.STATE_SET)
            .withLocalTemperature(ea.STATE)
            .withLocalTemperatureCalibration(-10, 10, 1, ea.STATE_SET)
            .withDescription('Default: 22'),

        e.numeric('deadzone_temperature', ea.STATE_SET)
            .withValueMin(5)
            .withValueMax(50)
            .withValueStep(5)
            .withLabel('Dead zone temperature')
            .withUnit('°C')
            .withDescription('Default: 0.5'),

        e.numeric('max_temperature', ea.STATE_SET)
            .withValueMin(20)
            .withValueMax(45)
            .withValueStep(0.5)
            .withLabel('Maximum temperature setpoint on device')
            .withUnit('°C')
            .withDescription('Default: 45'),

        e.numeric('lower_limit_of_temperature', ea.STATE_SET)
            .withValueMin(5)
            .withValueMax(15)
            .withValueStep(0.5)
            .withUnit('°C')
            .withLabel('Minimum temperature setpoint on device')
            .withDescription('Default: 5'),

        e.numeric('max_temperature_limit', ea.STATE_SET)
            .withValueMin(20)
            .withValueMax(45)
            .withValueStep(0.5)
            .withLabel('Maximum floor temperature')
[... 72 lines cut ...]
            [104, 'schedule_wednesday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(3)],
            [105, 'schedule_thursday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(4)],
            [106, 'schedule_friday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(5)],
            [107, 'schedule_saturday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(6)],
            [108, 'schedule_sunday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(7)],
            [109, 'external_temperature_input', tuya.valueConverter.divideBy10],
            [110, 'deadzone_temperature', tuya.valueConverter.raw],
            [111, 'max_temperature_limit', tuya.valueConverter.divideBy10],
            [112, 'min_temperature_limit', tuya.valueConverter.divideBy10],
            [113, 'eco_temperature', tuya.valueConverter.divideBy10],
            [114, 'screen_time_set', tuya.valueConverterBasic.lookup({
                '10s': tuya.enum(0),
                '20s': tuya.enum(1),
                '30s': tuya.enum(2),
                '40s': tuya.enum(3),
                '50s': tuya.enum(4),
                '60s': tuya.enum(5)
            })],
            [115, 'rgb_backlight', tuya.valueConverterBasic.lookup({ON: true, OFF: false})],
        ],
    },
    onEvent: tuya.onEventSetLocalTime,
};

module.exports = definition;`

Racermouse, 2024-12-10:
I tried the code and it worked for me anyway.

Some things I changed to my way of using a floor thermostat.
Limits have been change. Don't know what the real device limits are, but the ones in the code below works.
Dead zone (hysteresis) needs to be multiplied by 10 to work. I have no idea how to do it. Can't find a tuya.valueConverter to do this.

...and I don't know why the code does not show up here properly.
Sorry.

const exposes = require('zigbee-herdsman-converters/lib/exposes');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['TS0601'],
    model: 'ZHT-SR-GB',
    vendor: 'Moes',
    description: 'Smart knob thermostat',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    exposes: [
        e.binary('child_lock', ea.STATE_SET, 'ON', 'OFF').withLabel('Child lock'),

        e.binary('rgb_backlight', ea.STATE_SET, 'ON', 'OFF').withLabel('Knob light').withDescription('Default: On'),

        e.numeric('eco_temperature', ea.STATE_SET)
            .withValueMin(10)
            .withValueMax(30)
            .withValueStep(0.5)
            .withUnit('°C')
            .withDescription('Max temperature in ECO mode. Default: 30'),

        e.binary('valve_state', ea.STATE, false, true).withLabel('Heating in progress'),

        e.climate()
            .withSystemMode(['off', 'heat'], ea.STATE_SET)
            .withPreset(['manual', 'temporary manual', 'schedule', 'eco'])
            .withSetpoint('current_heating_setpoint', 5, 45, 0.5, ea.STATE_SET)
            .withLocalTemperature(ea.STATE)
            .withLocalTemperatureCalibration(-10, 10, 1, ea.STATE_SET)
            .withDescription('Default: 22'),

        e.numeric('deadzone_temperature', ea.STATE_SET)
            .withValueMin(5)
            .withValueMax(50)
            .withValueStep(5)
            .withLabel('Dead zone temperature')
            .withUnit('°C')
            .withDescription('Default: 0.5'),

        e.numeric('max_temperature', ea.STATE_SET)
            .withValueMin(20)
            .withValueMax(45)
            .withValueStep(0.5)
            .withLabel('Maximum temperature setpont on device')
            .withUnit('°C')
            .withDescription('Default: 45'),

        e.numeric('lower_limit_of_temperature', ea.STATE_SET)
            .withValueMin(5)
            .withValueMax(20)
            .withValueStep(0.5)
            .withUnit('°C')
            .withLabel('Minimum temperature setpont on device')
            .withDescription('Default: 5'),

        e.numeric('external_temperature_input', ea.STATE)
            .withLabel('Floor temperature')
            .withUnit('°C')
[... 75 lines cut ...]
            [104, 'schedule_wednesday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(3)],
            [105, 'schedule_thursday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(4)],
            [106, 'schedule_friday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(5)],
            [107, 'schedule_saturday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(6)],
            [108, 'schedule_sunday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(7)],
            [109, 'external_temperature_input', tuya.valueConverter.divideBy10],
            [110, 'deadzone_temperature', tuya.valueConverter.raw],
            [111, 'max_temperature_limit', tuya.valueConverter.divideBy10],
            [112, 'min_temperature_limit', tuya.valueConverter.divideBy10],
            [113, 'eco_temperature', tuya.valueConverter.divideBy10],
            [114, 'screen_time_set', tuya.valueConverterBasic.lookup({
                '10s': tuya.enum(0),
                '20s': tuya.enum(1),
                '30s': tuya.enum(2),
                '40s': tuya.enum(3),
                '50s': tuya.enum(4),
                '60s': tuya.enum(5)
            })],
            [115, 'rgb_backlight', tuya.valueConverterBasic.lookup({ON: true, OFF: false})],
        ],
    },
    onEvent: tuya.onEventSetLocalTime,
};

module.exports = definition;

cwtufts, 2025-01-15:
Why is it doing this? Is there a way to set it to Celsius?

[image]

ETE-Design, 2025-01-17:
@szheliab So seems like the code work right? Can this be added to Zigbee2MQTT so we don't have to use it as "External" converter?

Regards

adamkorski, 2025-01-18:
why don't you try it?

szheliab, 2025-01-20:
> [@szheliab](https://github.com/szheliab) So seems like the code work right? Can this be added to Zigbee2MQTT so we don't have to use it as "External" converter?
> 
> Regards

Hi
Unfortunately, I don't have this device installed :(

If you can confirm that my converter works correctly then I can prepare PR to add this device to supported list

cwtufts, 2025-01-20:
> [image]
> [image]
> [image]
> 
> everything works

Everything works as long as the global system is set to °C but if you use global settings as °F the temp will not change and will be a very erratic number. Anyway to fix this?

ETE-Design, 2025-01-20:
> [@ETE-Design](https://github.com/ETE-Design)
> 
> > [@szheliab](https://github.com/szheliab) So seems like the code work right? Can this be added to Zigbee2MQTT so we don't have to use it as "External" converter?
> > Regards
> 
> Hi Unfortunately, I don't have this device installed :(
> 
> If you can confirm that my converter works correctly then I can prepare PR to add this device to supported list

Same problem - Have ordered it but not received it yet... So can't test it for now :-(

raudraido, 2025-02-02:
deadzone temperature seems not to work

Johnwhutchinson, 2025-02-07:
Where do I paste this code to make it work?

ETE-Design, 2025-02-09:
Think @Koenkk  is the one who can answer you on that :-D

vpaulos, 2025-02-11:
device is not supported on version 2.1.1-1

[image]

sarangnemo, 2025-02-11:
> Where do I paste this code to make it work?

@Johnwhutchinson Assuming you're running z2m with Home Assistant, just paste the code into a new file and drop it in `/homeassistant/zigbee2mqtt/`. You may need to pair the device again.

sarangnemo, 2025-02-11:
> If you can confirm that my converter works correctly then I can prepare PR to add this device to supported list

@szheliab I can also confirm that all the obvious features are working correctly. I've only tried with the default units in Celsius though, so I can't rule out this potential issue with Fahrenheit: https://github.com/Koenkk/zigbee2mqtt/issues/25038#issuecomment-2602324026

nicmela, 2025-02-23:
Hi, ZigBee rookie here! 
I have home assistant with ZHA integration. Is this converter usable in zone way with ZHA?

Thanks and sorry if the question is silly!

sarangnemo, 2025-02-24:
> Hi, ZigBee rookie here! I have home assistant with ZHA integration. Is this converter usable in zone way with ZHA?
> 
> Thanks and sorry if the question is silly!

Sorry @nicmela, zigbee2mqtt code can't plug in to ZHA. Though someone with experience writing ZHA device 'quirks', or a lot of time to learn the ... _quirks_  ... of writing quirks could likely make something for ZHA based on the information contained within this code.

szheliab, 2025-03-03:
Hi @TesterTurner 
Support of this device is already added - https://www.zigbee2mqtt.io/devices/ZHT-SR.html :)

ETE-Design, 2025-03-05:
Seems like the On / Off Relay is also missing in the merge :-(

magic1447, 2025-03-05:
> Also cannot change preset mode from temp manual to program and cannot set schedule either. Noticed this last night. Thank you!

I have noticed the same thing. It was working fine on external converter. Another thing that causes a problem for me is that is often goes into schedule mode.
I have this in the states tab in z2m:
`"schedule_friday": "06:00/20.0 11:30/20.0 13:30/20.0 17:30/20.0",
    "schedule_monday": "06:00/20.0 11:30/20.0 13:30/20.0 17:30/20.0",
    "schedule_saturday": "06:00/20.0 11:30/20.0 13:30/20.0 17:30/20.0",
    "schedule_sunday": "06:00/20.0 11:30/20.0 13:30/20.0 17:30/20.0",
    "schedule_thursday": "06:00/20.0 11:30/20.0 13:30/20.0 17:30/20.0",
    "schedule_tuesday": "06:00/20.0 11:30/20.0 13:30/20.0 17:30/20.0",
    "schedule_wednesday": "06:00/20.0 11:30/20.0 13:30/20.0 17:30/20.0",` 
I have tried to clear it by publishing mqtt message, but I cant get to cleat those.

szheliab, 2025-03-05:
Hi there!
@Mark-Muc 
Issue with `null` values is fixed in this PR  [#8920](https://github.com/Koenkk/zigbee-herdsman-converters/pull/8920)

@magic1447 
It seems like Schedule is not supported, please check this PR [#8803](https://github.com/Koenkk/zigbee-herdsman-converters/pull/8803)

magic1447, 2025-03-05:
> Hi there! [@Mark-Muc](https://github.com/Mark-Muc) Issue with `null` values is fixed in this PR [#8920](https://github.com/Koenkk/zigbee-herdsman-converters/pull/8920) [...]

I did notice that, but somehow it keeps switching over from eco or manual to schedule mode on its own. I dont really want that as I have automations managing heating. I was hoping that clearing those fields or replacing with null would fix that proble

Johnwhutchinson, 2025-03-07:
Is there a way of using the "running state" (heat/idle) to trigger automation? I can't find it as a trigger?

Mark-Muc, 2025-03-08:
@magic1447 Thank u for the quick answer and i installed Zigbee2MQTT Edge and the issues with `NUL`L has been disappeared (as u said) . thank u again!!
But still some things didn't work as expected / designed. some mentioned in my post before. ... 
were is the best place to report this issues?
Regards 
Mark-Muc

ETE-Design, 2025-03-08:
@Mark-Muc Is the On/Off Relay there again on Zigbee2MQTT Edge or still missing also?

Mark-Muc, 2025-03-08:
@ETE-Design:
Yes it is missing also . I looked in the external converter and in this one it is mentioned and displayed:
`
        e.binary('valve_state', ea.STATE, false, true).withLabel('Heating in process'),`

z2m papped it  to:  `     - entity: binary_sensor.test_thermostat_valve_state`
also

`        e.enum('sensor', ea.STATE_SET, ['IN', 'OU', 'AL']).withLabel('Sensor')`
is just a reading of the sensor .. in the external converter it looks like above and you can change it.

Mark-Muc

SNPN, 2025-03-08:
Additional question: looks like mines do sync time (hour) but not days.
Time is correct but not the day of the week (looks like it starts from Sunday when I power them on, and then it never fixes the actual day).
Is that a known issue?
Thanks,

besbar, 2025-03-21:
Dear Zigbee2MQTT Team,

First of all, I would like to sincerely thank everyone who contributed to integrating the ZHT-SR thermostat into Zigbee2MQTT. I truly appreciate your efforts!

I'm delighted with this thermostat, as it allows temperature control with a 0.5°C sensitivity, which is a great improvement over my previous thermostat ([BHT-002](https://www.zigbee2mqtt.io/devices/BHT-002.html)), which was limited to a 1°C sensitivity.

I successfully integrated the ZHT-SR thermostat into Node-RED using the nodes from the following flow:
https://flows.nodered.org/flow/25857cba1ab7e34309133ac8354004c6/in/_yWzeA2pqaLd
The results are excellent, and I’m very satisfied!

However, I wonder if there is any hope for the following improvements to be implemented:

Synchronization between presets (Manual, Temporary Manual, Program, and Eco) and the thermostat mode:

When manually changing the thermostat mode, the preset value is correctly updated in Zigbee2MQTT.
However, when setting the preset via an MQTT set message, the thermostat remains in Manual mode.
For example, if I select the Eco preset from the Zigbee2MQTT frontend, the thermostat does not actually switch to Eco mode. But if I manually set the thermostat to Eco mode, the Eco Temperature value is correctly applied.
The ability to send schedules to the thermostat using the following endpoints:

js
Copier
Modifier
[102, 'schedule_monday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(1)],
[103, 'schedule_tuesday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(2)],
[104, 'schedule_wednesday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(3)],
[105, 'schedule_thursday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(4)],
[106, 'schedule_friday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(5)],
[107, 'schedule_saturday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(6)],
[108, 'schedule_sunday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(7)],
Minor improvements:

Correct date display: The time is accurate, but the thermostat displays the date with a two-day delay.
Enable or disable the ring light on the thermostat.
Thank you again for your hard work on this integration! I appreciate any insights or potential fixes for these issues.

besbar, 2025-03-21:
Following up on my previous message, I wanted to share additional insights that might help with the implementation of preset synchronization.

I have **captured the Zigbee data sent by the thermostat when manually switching between the different modes using a Zigbee sniffer**:

**Manual Mode**: _00c00204000100_
**Temporary Manual Mode**: _00c00204000101_
**Program Mode:** _00ba0204000102_
**Eco Mode**: _00b20204000103_

These are the raw data packets transmitted when I manually switch the thermostat between the available modes. I hope this information can be useful for improving the synchronization between the preset values and the actual thermostat mode.

Please let me know if I can provide further details or perform additional tests to assist with the development.