Koenkk/zigbee2mqtt #25050: [New device support]: _TZ3000_xa9g7rxs

leonwebcreators, 2024-12-03:
### Link

https://nl.aliexpress.com/item/1005005457929328.html

### Database entry

{"id":35,"type":"EndDevice","ieeeAddr":"0x0ceff6fffea6328a","nwkAddr":23391,"manufId":4098,"manufName":"_TZ3000_xa9g7rxs","powerSource":"Battery","modelId":"TS1002","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":261,"inClusterList":[0,1,3,4,4096,57345],"outClusterList":[25,10,3,4,5,6,8,768,4096],"clusters":{"genBasic":{"attributes":{"modelId":"TS1002","manufacturerName":"_TZ3000_xa9g7rxs","powerSource":3,"zclVersion":3,"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":""}},"manuSpecificTuya_2":{"attributes":{"53265":1}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1733255924207}

### Zigbee2MQTT version

1.42.0

### Comments

Tried this:
https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_2-1-extending-the-external-definition

But read also that the manual of the UI and configuration.yaml are not correct (anymore). Anyhow, i did the following:

- Paired the device
- Pressed all of the buttons
- Went to developer console of Zigbee2mqtt in Home Assistant Addon
- Clicked 'generate_external_definition'.
- Copied the generated content to config/TS1002.js and added 'TS1002.js' to the External Converter tab of the UI Config.
- Restarted Z2M

But nothing changes.

Pressing buttons generated this logs (even without previous steps):
`2024-12-03 21:14:51z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x0ceff6fffea6328a', payload '{"last_seen":"2024-12-03T20:14:51.650Z","linkquality":58}'
info 2024-12-03 21:14:53z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x0ceff6fffea6328a', payload '{"last_seen":"2024-12-03T20:14:53.457Z","linkquality":58}'
info 2024-12-03 21:14:54z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x0ceff6fffea6328a', payload '{"last_seen":"2024-12-03T20:14:54.492Z","linkquality":58}'`

### External definition

```shell
const {battery, commandsOnOff, commandsLevelCtrl, commandsColorCtrl} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['TS1002'],
    model: 'TS1002',
    vendor: '_TZ3000_xa9g7rxs',
    description: 'Automatically generated definition',
    extend: [battery(), commandsOnOff(), commandsLevelCtrl(), commandsColorCtrl()],
    meta: {},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

Seems like nothing works...

leonwebcreators, 2024-12-23:
any updates on this? :)

masarovofis, 2024-12-24:
Удалось кому то собрать Внешний конвертер для этого пульта?

Sergei7885, 2025-01-27:
Has anyone overcome this situation? Help add him to support

leonwebcreators, 2025-03-20:
Unfortunately, still stuck on this. It's a tuya thing, but the docs are very unclear, even after they have been updated (the docs for adding a new device).

Anyone can help?....

leonwebcreators, 2025-05-20:
Is there anyone who can help with this? Are there any related examples? I've been working for days now in the few last months, but can't get it to work....

leonwebcreators, 2025-06-19:
FIXED!

`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const e = exposes.presets;
const ea = exposes.access;

const fzLocal = {
    pk8_action: {
        cluster: 'genOnOff',
        type: ['commandTuyaAction', 'commandOn', 'commandOff'],
        convert: (model, msg, publish, options, meta) => {
            if (msg.type === 'commandTuyaAction') {
                const data = msg.data.data;
                const code = data[1];
                const actionMap = {
                    0x01: 'scene_1',
                    0x02: 'scene_2',
                    0x03: 'scene_3',
                    0x04: 'scene_4',
                    0x05: 'scene_5',
                    0x06: 'scene_6',
                    0x07: 'scene_7',
                    0x08: 'scene_8',
                };
                const action = actionMap[code];
                return action ? {action} : {};
            } else if (msg.type === 'commandOn') {
                return {action: 'hold_on'};
            } else if (msg.type === 'commandOff') {
                return {action: 'hold_off'};
            }
            return {};
        },
    },
};

const device = {
    fingerprint: tuya.fingerprint("TS1002", ["_TZ3000_xa9g7rxs"]),
    model: "TS1002_multi_8_scene",
    vendor: "Tuya",
    description: "Wall-mounted 8 scene button panel with backlight",
    fromZigbee: [fzLocal.pk8_action],
    toZigbee: [],
    exposes: [
        e.action([
            'scene_1',
            'scene_2',
            'scene_3',
            'scene_4',
            'scene_5',
            'scene_6',
            'scene_7',
            'scene_8',
            'hold_on',
            'hold_off',
        ]),
    ],
    configure: async (device, coordinatorEndpoint) => {
        await tuya.configureMagicPacket(device, coordinatorEndpoint);
    },
};

module.exports = device;`

Koenkk, 2025-06-19:
Could you make a PR by clicking [here](https://github.com/Koenkk/zigbee-herdsman-converters/edit/master/src/devices/tuya.ts)?

Sergei7885, 2025-06-20:
I spent a lot of time, but I managed to make the remote control give data from other buttons, here is my code, all buttons work
`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const e = exposes.presets;

const fzLocal = {
    pk8_action: {
        cluster: 'genOnOff',
        type: ['commandTuyaAction', 'commandOn', 'commandOff'],
        convert: (model, msg, publish, options, meta) => {
            if (msg.type === 'commandTuyaAction') {
                const data = msg.data.data;
                const code = data[1];
                const actionMap = {
                    0x01: 'scene_1',
                    0x02: 'scene_2',
                    0x03: 'scene_3',
                    0x04: 'scene_4',
                    0x05: 'scene_5',
                    0x06: 'scene_6',
                    0x07: 'scene_7',
                    0x08: 'scene_8',
                };
                const action = actionMap[code];
                return action ? {action} : {};
            } else if (msg.type === 'commandOn') {
                return {action: 'hold_on'};
            } else if (msg.type === 'commandOff') {
                return {action: 'hold_off'};
            }
            return {};
        },
    },
    pk8_brightness: {
        cluster: 'genLevelCtrl',
        type: ['commandStep'],
        convert: (model, msg, publish, options, meta) => {
            const {stepmode} = msg.data;
            if (stepmode === 1) return {action: 'brightness_up'};
            if (stepmode === 0) return {action: 'brightness_down'};
            return {};
        },
    },
    pk8_colortemp: {
        cluster: 'lightingColorCtrl',
        type: ['commandStepColorTemp'],
        convert: (model, msg, publish, options, meta) => {
            const {stepmode} = msg.data;
            if (stepmode === 1) return {action: 'color_temp_down'}; // теплее
            if (stepmode === 3) return {action: 'color_temp_up'};   // холоднее
            return {};
        },
    },
};

const device = {
    fingerprint: tuya.fingerprint("TS1002", ["_TZ3000_xa9g7rxs"]),
    model: "TS1002_multi_8_scene",
    vendor: "Tuya",
    description: "Wall-mounted 8 scene button panel with backlight",
    fromZigbee: [fzLocal.pk8_action, fzLocal.pk8_brightness, fzLocal.pk8_colortemp],
    toZigbee: [],
    exposes: [
        e.action([
            'scene_1',
            'scene_2',
            'scene_3',
            'scene_4',
            'scene_5',
            'scene_6',
            'scene_7',
            'scene_8',
            'hold_on',
            'hold_off',
            'brightness_up',
            'brightness_down',
            'color_temp_up',
            'color_temp_down',
        ]),
    ],
    configure: async (device, coordinatorEndpoint) => {
        await tuya.configureMagicPacket(device, coordinatorEndpoint);
    },
};

module.exports = device;`

closed 2025-09-01 by github-actions[bot]