Koenkk/zigbee2mqtt #25053: Tuya ZG-101Z/D detected as ERS-10TZBVK-AA

BlaXun, 2024-12-03:
### What happened?

I bought a couple of these radial switches (which work in ZHA in Home Assistant) https://aliexpress.com/item/1005007575864175.html?spm=a2g0o.order_list.order_list_main.65.4a3f5c5fBDiin9

When adding them to zigbee2mqtt they are discovered as [ERS-10TZBVK-AA](https://www.zigbee2mqtt.io/devices/ERS-10TZBVK-AA.html#tuya-ers-10tzbvk-aa)

I tried manually adding support for this device (never did this before) but can't get it to load into "blank state" so I can add support step-by-step.

### What did you expect to happen?

Either correct detection or unsupported with a blank state so I can try and add support by myself

### How to reproduce it (minimal and precise)

Get one of the mentioned switches and try to pair it

### Zigbee2MQTT version

1.42.0

### Adapter firmware version

20210708

### Adapter

SONOFF Zigbee 3.0 USB Dongle plus

### Setup

Home Assistant Add-On

BlaXun, 2024-12-03:
Side note: I discovered the code ZG-101Z/D by opening up the switch and taking a look at its internals

BlaXun, 2024-12-03:
Log from interview to "successful" pairing

info 2024-12-03 22:23:45zh:controller: Interview for '0xa4c138d65b0e06fd' started
info 2024-12-03 22:23:45z2m: Device '0xa4c138d65b0e06fd' joined
info 2024-12-03 22:23:46z2m: Starting interview of '0xa4c138d65b0e06fd'
info 2024-12-03 22:23:46z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c138d65b0e06fd","ieee_address":"0xa4c138d65b0e06fd"},"type":"device_joined"}'
info 2024-12-03 22:23:46z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c138d65b0e06fd","ieee_address":"0xa4c138d65b0e06fd","status":"started"},"type":"device_interview"}'
info 2024-12-03 22:23:46z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c138d65b0e06fd","ieee_address":"0xa4c138d65b0e06fd"},"type":"device_announce"}'
info 2024-12-03 22:23:58zh:controller: Succesfully interviewed '0xa4c138d65b0e06fd'
info 2024-12-03 22:23:58z2m: Successfully interviewed '0xa4c138d65b0e06fd', device has successfully been paired
info 2024-12-03 22:23:58z2m: Device '0xa4c138d65b0e06fd' is supported, identified as: Tuya Smart knob (ERS-10TZBVK-AA)
info 2024-12-03 22:23:58z2m:mqtt: MQTT publish: topic 'homeassistant/sensor/0xa4c138d65b0e06fd/action/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0xa4c138d65b0e06fd"],"manufacturer":"Tuya","model":"Smart knob (ERS-10TZBVK-AA)","name":"0xa4c138d65b0e06fd","sw_version":"0122052017","via_device":"zigbee2mqtt_bridge_0x00124b0030d2eaaf"},"enabled_by_default":true,"entity_category":"diagnostic","icon":"mdi:gesture-double-tap","name":"Action","object_id":"0xa4c138d65b0e06fd_action","origin":{"name":"Zigbee2MQTT","sw":"1.42.0","url":"https://www.zigbee2mqtt.io"},"state_topic":"zigbee2mqtt/0xa4c138d65b0e06fd","unique_id":"0xa4c138d65b0e06fd_action_zigbee2mqtt","value_template":"{{ value_json.action }}"}'
info 2024-12-03 22:23:58z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Smart knob","exposes":[{"access":1,"category":"diagnostic","description":"Triggered action (e.g. a button click)","label":"Action","name":"action","property":"action","type":"enum","values":["toggle","brightness_step_up","brightness_step_down","color_temperature_step_up","color_temperature_step_down","saturation_move","hue_move","hue_stop","single","double","hold","rotate_left","rotate_right"]},{"access":1,"label":"Action step size","name":"action_step_size","property":"action_step_size","type":"numeric","value_max":255,"value_min":0},{"access":1,"label":"Action transition time","name":"action_transition_time","property":"action_transition_time","type":"numeric","unit":"s"},{"access":1,"label":"Action rate","name":"action_rate","property":"action_rate","type":"numeric","value_max":255,"value_min":0},{"access":1,"category":"diagnostic","description":"Remaining battery in %, can take up to 24 hours before reported","label":"Battery","name":"battery","property":"battery","type":"numeric","unit":"%","value_max":100,"value_min":0},{"access":7,"description":"Operation mode: \"command\" - for group control, \"event\" - for clicks","label":"Operation mode","name":"operation_mode","property":"operation_mode","type":"enum","values":["command","event"]},{"access":1,"category":"diagnostic","description":"Link quality (signal strength)","label":"Linkquality","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"ERS-10TZBVK-AA","options":[{"access":2,"description":"Simulate a brightness value. If this device provides a brightness_move_up or brightness_move_down action it is possible to specify the update interval and delta. The action_brightness_delta indicates the delta for each interval. ","features":[{"access":2,"description":"Delta per interval, 20 by default","label":"Delta","name":"delta","property":"delta","type":"numeric","value_min":0},{"access":2,"description":"Interval duration","label":"Interval","name":"interval","property":"interval","type":"numeric","unit":"ms","value_min":0}],"label":"Simulated brightness","name":"simulated_brightness","property":"simulated_brightness","type":"composite"}],"supports_ota":false,"vendor":"Tuya"},"friendly_name":"0xa4c138d65b0e06fd","ieee_address":"0xa4c138d65b0e06fd","status":"successful","supported":true},"type":"device_interview"}'
info 2024-12-03 22:23:58z2m:mqtt: MQTT publish: topic 'homeassistant/sensor/0xa4c138d65b0e06fd/action_step_size/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0xa4c138d65b0e06fd"],"manufacturer":"Tuya","model":"Smart knob (ERS-10TZBVK-AA)","name":"0xa4c138d65b0e06fd","sw_version":"0122052017","via_device":"zigbee2mqtt_bridge_0x00124b0030d2eaaf"},"enabled_by_default":true,"name":"Action step size","object_id":"0xa4c138d65b0e06fd_action_step_size","origin":{"name":"Zigbee2MQTT","sw":"1.42.0","url":"https://www.zigbee2mqtt.io"},"state_topic":"zigbee2mqtt/0xa4c138d65b0e06fd","unique_id":"0xa4c138d65b0e06fd_action_step_size_zigbee2mqtt","value_template":"{{ value_json.action_step_size }}"}'
[... cut ...]
info 2024-12-03 22:24:18z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xa4c138d65b0e06fd', payload '{"action":null,"action_rate":null,"action_step_size":null,"action_transition_time":null,"battery":100,"linkquality":75,"operation_mode":"event","voltage":3000}'
info 2024-12-03 22:24:28z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xa4c138d65b0e06fd', payload '{"action":null,"action_rate":null,"action_step_size":null,"action_transition_time":null,"battery":100,"linkquality":75,"operation_mode":"event","voltage":3000}'

spali, 2024-12-05:
From the look I have exactly the same. But I can't find the code inside. The only letters I found was "IHSENO XUANNIU-V02" on the board.
Anyway I'm almost sure it's the same as #23543  which should be expected to be detected as [ERS-10TZBVK-AA](https://www.zigbee2mqtt.io/devices/ERS-10TZBVK-AA.html#tuya-ers-10tzbvk-aa).
My knobs that look like yours and like the one from the other issue, kind of works, I have still problems with the reaction time and some strange `z2m: EventBus error 'HomeAssistant/publishEntityState': Cannot read properties of null (reading 'toString')` errors. But I'm almost sure this is because I just started completely fresh and have some interference problems and also are on the `dev`  branch (edge addon) to not start with an old to migrate config 😉  Also currently lack other "input" devices to compare with.

BlaXun, 2024-12-05:
For me it also kinda works. When I pair the device to Z2M and then check in HA it does showbup, but the events available for automations are very limited. However, as soon as I open up Z2M interface and perform the different gestures on the device (turn left, turn right, press, double press, hold) they will then be available in HA (not sure if I had to restart z2m). But I also do get some big error messages in Z2M from time to time regarding these switches... so something is not completely correct for sure

spali, 2024-12-05:
as mentioned, newbie here, but I think the different possible events/triggers by device get's learned by HA during use.... so as I did understand, it's also expected to execute everything once so, HA knows them.
So it seems it works, but we both have errors... maybe a specific issue for the errors would make more sense.... I don't as I still not sure if experimental v2.0.0 is the problem or my bad zigbee network.

BlaXun, 2024-12-05:
I wasn't aware it is supposed to learn stuff by simply executing commands on the end-device. That's cool. My zigbee mesh is completely new though

xeoMe015, 2024-12-05:
I have the same issue with my “_TZ3000_abrsvsou” 

i’m new to zigbee2mqtt as well unfortunatly i get the following error in my logs:

z2m: Publish 'get' 'operation_mode' to '0xa4c138e5134fd8f0' failed: 'Error: ZCL command 0xa4c138e5134fd8f0/1 genOnOff.read(["tuyaOperationMode"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Data request failed with error: 'MAC transaction expired' (240))'

spali, 2024-12-05:
> I have the same issue with my “_TZ3000_abrsvsou”
>
> i’m new to zigbee2mqtt as well unfortunatly i get the following error in my logs:
>
> z2m: Publish 'get' 'operation_mode' to '0xa4c138e5134fd8f0' [...]

That's not the same error this issue is about. Looks like your device doesn't provide operation_mode. But probably not even critical... if the device works otherwise.

BlaXun, 2024-12-09:
> These are only available in operation mode "command" as I know. Didn't search a lot, but the only blueprint for this device, I stumbled up was one for the "event" mode. Just search in the HASS community board for "tuya smart knob" or similar.

And, if I am not mistaken, due to the device not being detected correctly it throws an error when trying to swithc to command mode. So, yeah, something is definetely broken here. Oh well...

SaturnusDJ, 2024-12-21:
Upon reconfiguring this ZG-101Z/D device in Z2M 1.42.0:

```
error: 	z2m: Failed to configure 'button_r', attempt 1 (Error: ZCL command 0x****************/1 genBasic.read([4,0,1,5,7,65534], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/devices/tuya.ts:7657:13)

error: 	z2m: Failed to configure 'button_r', attempt 4 (Error: ZCL command 0x****************/1 genBasic.read([4,0,1,5,7,65534], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
    at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:350:28)
    at Endpoint.zclCommand (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:956:26)
    at Endpoint.read (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:446:29)
    at Configure.configure (/app/lib/extension/configure.ts:130:13)
    at Configure.onMQTTMessage (/app/lib/extension/configure.ts:57:21)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:211:17))

info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/configure', payload '{"data":{"id":"button_r"},"error":"Failed to configure (ZCL command 0x****************/1 genBasic.read([4,0,1,5,7,65534], {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"reservedBits\":0,\"writeUndiv\":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE'))","status":"error","transaction":"*******"}'
```

I think some adjustment is needed to get the `action_step_size`, `action_transition_time` and `action_rate` exposes working.

sorz2122, 2024-12-26:
My button only sends a signal every three steps when turning, which is quite annoying for dimming. You have to turn forever. Does anyone else have this problem? Is this normal?

SaturnusDJ, 2024-12-26:
> My button only sends a signal every three steps when turning, which is quite annoying for dimming. You have to turn forever. Does anyone else have this problem? Is this normal?

My understanding from https://youtube.com/watch?v=QSI3C6fHQSI (a must watch for this device) is that this is normal. Things probably work better once we get access to `action_rate`.

SaturnusDJ, 2025-03-01:
Can you help @Koenkk?

LeVraiRoiDHyrule, 2025-03-11:
Hi, I am having the exact same problem.

jief666, 2025-03-15:
> My button only sends a signal every three steps when turning, which is quite annoying for dimming. You have to turn forever. Does anyone else have this problem? Is this normal?

Just got a knob. PCB says "ZG-101Z/D". Knob is seen as "TS004F by _TZ3000_abrsvsou". It sends on event every 3 steps. Annoying. 

Did you find a solution, or is it just a bad copy of the original ERS-10TZBVK-AA ?

SaturnusDJ, 2025-03-16:
> Just got a knob. PCB says "ZG-101Z/D". Knob is seen as "TS004F by _TZ3000_abrsvsou". It sends on event every 3 steps. Annoying.
>
> Did you find a solution, or is it just a bad copy of the original [...]

https://github.com/Koenkk/zigbee2mqtt/issues/25053#issuecomment-2563090023

jief666, 2025-03-16:
> > > My button only sends a signal every three steps when turning, which is quite annoying for dimming. You have to turn forever. Does anyone else have this problem? Is this normal?
> > 
> > 
> > Just got a knob. PCB says "ZG-101Z/D". Knob is seen as "TS004F by _TZ3000_abrsvsou". It sends on event every 3 steps. Annoying.
> > Did you find a solution, or is it just a bad copy of the original ERS-10TZBVK-AA ?
> 
> [#25053 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/25053#issuecomment-2563090023)

I read that comment. But the video is about someone creating a very interesting automation for an original MOES knob. Nothing to do with this "3 steps per event" problem.

SaturnusDJ, 2025-03-16:
Hmmm right, where did I read about it then...

We need a better Z2M integration first.
I expect this white button to have the same electronics as the grey button, because both have command and event mode. Then we can use the same 'exposes' to have a better experience.

jief666, 2025-03-17:
I also wouldn't be surprised that they copied the electronics but couldn't copy the software. On that cheap button, when you triple click, you get an event saying it changed command mode. But then, the Zigbee event you received stayed the same.
I guess they didn't bother to program that part !

SaturnusDJ, 2025-03-22:
Yes that is possible. I don't have the real one for reference. What I assumed earlier is that the whole electronics part including software is the same, except the brand and model name. Seen this before with TuYa, Moes, BlitzWolf and many other brands. All come down to the same core design: TuYa. Search for TuYa button on AliExpress and you will find both the white and the black one.

We need to have a complete configuration in Z2M for this variant before we can conclude anything definitely. Soooo @Koenkk 😬😬 can you please help.

Koenkk, 2025-03-23:
- Could you provide the data/database.db entry of this device?
- What is wrong with the current integration? Is anything not working?

SaturnusDJ, 2025-03-23:
@Koenkk Thanks!

Rotating and pressing this button/rotary device works, but there is no `action_step_size`, `action_transition_time` and `action_rate` while I think there should be.
(https://www.zigbee2mqtt.io/devices/ERS-10TZBVK-AA.html)

Upon pairing or reconfiguring:
```
error: 	z2m: Failed to configure 'button_r', attempt 4 (Error: ZCL command 0xXXXXXXXXXXXXXXXX/1 genBasic.read([4,0,1,5,7,65534], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Timeout - 45307 - 1 - 101 - 0 - 1 after 10000ms)
    at Object.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/utils/waitress.ts:67:23)
    at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:663:47)
    at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:695:28)
    at /app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:443:20
    at Queue.execute (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/utils/queue.ts:36:20)
    at ZStackAdapter.sendZclFrameToEndpoint (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:441:16)
    at Request.send (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:82:20)
    at Endpoint.sendRequest (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:324:20)
    at Endpoint.zclCommand (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:949:28)
    at Endpoint.read (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:445:29))
```
When doing interview again:
```
error: 	z2m: Request 'zigbee2mqtt/bridge/request/device/interview' failed with error: 'interview of 'button_r' (0xXXXXXXXXXXXXXXXX) failed: Error: Interview failed because can not get active endpoints ('0xXXXXXXXXXXXXXXXX')'
```
I thought this means some configuration is missing, or am I wrong?

From the database:
```
{"id":71,"type":"EndDevice","ieeeAddr":"0xXXXXXXXXXXXXXXXX","nwkAddr":45307,"manufId":4742,"manufName":"_TZ3000_abrsvsou","powerSource":"Battery","modelId":"TS004F","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":260,"inClusterList":[0,1,3,4,6,4096],"outClusterList":[25,10,3,4,5,6,8,4096],"clusters":{"genBasic":{"attributes":{"modelId":"TS004F","manufacturerName":"_TZ3000_abrsvsou","powerSource":3,"zclVersion":3,"appVersion":145,"stackVersion":2,"hwVersion":1,"swBuildId":"0122052017"}},"genOnOff":{"attributes":{"tuyaOperationMode":1,"onOff":0}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200,"batteryVoltage":29}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":145,"stackVersion":2,"hwVersion":1,"swBuildId":"0122052017","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1742753627172}
```

SaturnusDJ, 2025-03-26:
@Koenkk 
Thanks.

After force remove: Next to the coordinator it pairs successfully. Then interviewing or reconfiguring gives the same errors. However, seems like last time I missed a second line of logging for both, so I added that to my post above this one.

Debug log when rotating in event mode:
```
:52] info: 	z2m: Successfully changed options
:52] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/options', payload '{"data":{"restart_required":false},"status":"ok","transaction":"ctz3r-1"}'
:52] debug: 	zh:zstack:unpi:parser: <-- [254,24,68,129,0,0,6,0,29,108,1,1,0,255,0,153,186,8,0,0,4,17,13,252,0,29,108,29,246]
:52] debug: 	zh:zstack:unpi:parser: --- parseNext [254,24,68,129,0,0,6,0,29,108,1,1,0,255,0,153,186,8,0,0,4,17,13,252,0,29,108,29,246]
:52] debug: 	zh:zstack:unpi:parser: --> parsed 24 - 2 - 4 - 129 - [0,0,6,0,29,108,1,1,0,255,0,153,186,8,0,0,4,17,13,252,0,29,108,29] - 246
:52] debug: 	zh:zstack:znp: <-- AREQ: AF - incomingMsg - {"groupid":0,"clusterid":6,"srcaddr":27677,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":255,"securityuse":0,"timestamp":572057,"transseqnumber":0,"len":4,"data":{"type":"Buffer","data":[17,13,252,0]}}
:52] debug: 	zh:controller: Received payload: clusterID=6, address=27677, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=255, frame={"header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":13,"commandIdentifier":252},"payload":{"value":0},"command":{"ID":252,"parameters":[{"name":"value","type":32}],"name":"tuyaAction2"}}
:52] debug: 	zh:zstack:unpi:parser: --- parseNext []
:52] debug: 	z2m: Received Zigbee message from '0xXXXXXXXXXXXXXXXX', type 'commandTuyaAction2', cluster 'genOnOff', data '{"value":0}' from endpoint 1 with groupID 0
:52] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xXXXXXXXXXXXXXXXX', payload '{"action":"rotate_right","battery":100,"linkquality":255,"operation_mode":"event","voltage":2900}'
:52] debug: 	zh:zstack:unpi:parser: <-- [254,23,68,129,0,0,6,0,29,108,1,1,0,255,0,114,248,8,0,0,3,17,14,3,29,108,29,171]
:52] debug: 	zh:zstack:unpi:parser: --- parseNext [254,23,68,129,0,0,6,0,29,108,1,1,0,255,0,114,248,8,0,0,3,17,14,3,29,108,29,171]
:52] debug: 	zh:zstack:unpi:parser: --> parsed 23 - 2 - 4 - 129 - [0,0,6,0,29,108,1,1,0,255,0,114,248,8,0,0,3,17,14,3,29,108,29] - 171
:52] debug: 	zh:zstack:znp: <-- AREQ: AF - incomingMsg - {"groupid":0,"clusterid":6,"srcaddr":27677,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":255,"securityuse":0,"timestamp":587890,"transseqnumber":0,"len":3,"data":{"type":"Buffer","data":[17,14,3]}}
:52] debug: 	zh:controller: Failed to parse frame: Error: Cluster 'genOnOff' has no command '3'
:52] debug: 	zh:controller: Received payload: clusterID=6, address=27677, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=255, frame=undefined
:52] debug: 	zh:zstack:unpi:parser: --- parseNext []
:52] debug: 	z2m: Received Zigbee message from '0xXXXXXXXXXXXXXXXX', type 'raw', cluster 'genOnOff', data '{"data":[17,14,3],"type":"Buffer"}' from endpoint 1 with groupID 0
:52] debug: 	z2m: No converter available for 'ERS-10TZBVK-AA' with cluster 'genOnOff' and type 'raw' and data '{"data":[17,14,3],"type":"Buffer"}'
:54] debug: 	z2m:mqtt: Received MQTT message on 'zigbee2mqtt/bridge/request/options' with data '{"options":{"advanced":{"log_level":"info"}},"transaction":"ctz3r-2"}'
:54] info: 	z2m: Successfully changed options
```

Debug log when rotating in command mode:
```
:36] info: 	z2m: Successfully changed options
:36] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/options', payload '{"data":{"restart_required":false},"status":"ok","transaction":"ctz3r-3"}'
:36] debug: 	zh:zstack:unpi:parser: <-- [254,24,68,129,0,0,6,0,29,108,1,1,0,244,0,49,190,39,0,0,4,17,22,252,0,29,108,29,101]
:36] debug: 	zh:zstack:unpi:parser: --- parseNext [254,24,68,129,0,0,6,0,29,108,1,1,0,244,0,49,190,39,0,0,4,17,22,252,0,29,108,29,101]
:36] debug: 	zh:zstack:unpi:parser: --> parsed 24 - 2 - 4 - 129 - [0,0,6,0,29,108,1,1,0,244,0,49,190,39,0,0,4,17,22,252,0,29,108,29] - 101
:36] debug: 	zh:zstack:znp: <-- AREQ: AF - incomingMsg - {"groupid":0,"clusterid":6,"srcaddr":27677,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":244,"securityuse":0,"timestamp":2604593,"transseqnumber":0,"len":4,"data":{"type":"Buffer","data":[17,22,252,0]}}
:36] debug: 	zh:controller: Received payload: clusterID=6, address=27677, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=244, frame={"header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":22,"commandIdentifier":252},"payload":{"value":0},"command":{"ID":252,"parameters":[{"name":"value","type":32}],"name":"tuyaAction2"}}
:36] debug: 	zh:zstack:unpi:parser: --- parseNext []
:36] debug: 	z2m: Received Zigbee message from '0xXXXXXXXXXXXXXXXX', type 'commandTuyaAction2', cluster 'genOnOff', data '{"value":0}' from endpoint 1 with groupID 0
:36] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xXXXXXXXXXXXXXXXX', payload '{"action":"rotate_right","battery":100,"linkquality":244,"operation_mode":"command","voltage":2900}'
:37] debug: 	zh:zstack:unpi:parser: <-- [254,23,68,129,0,0,6,0,29,108,1,1,0,251,0,242,251,39,0,0,3,17,23,3,29,108,29,26]
[... cut ...]
:37] debug: 	z2m: Received Zigbee message from '0xXXXXXXXXXXXXXXXX', type 'raw', cluster 'genOnOff', data '{"data":[17,23,3],"type":"Buffer"}' from endpoint 1 with groupID 0
:37] debug: 	z2m: No converter available for 'ERS-10TZBVK-AA' with cluster 'genOnOff' and type 'raw' and data '{"data":[17,23,3],"type":"Buffer"}'
:37] debug: 	z2m:mqtt: Received MQTT message on 'zigbee2mqtt/bridge/request/options' with data '{"options":{"advanced":{"log_level":"info"}},"transaction":"ctz3r-4"}'
:37] info: 	z2m: Successfully changed options
```

SaturnusDJ, 2025-03-27:
@Koenkk 
Before I do tests... Since the device is recognized as [ERS-10TZBVK-AA](https://www.zigbee2mqtt.io/devices/ERS-10TZBVK-AA.html#tuya-ers-10tzbvk-aa), is it possible that this unknown data is just the same as what `action_step_size`, `action_transition_time` and `action_rate` are for [ERS-10TZBVK-AA](https://www.zigbee2mqtt.io/devices/ERS-10TZBVK-AA.html#tuya-ers-10tzbvk-aa)?

Koenkk, 2025-03-28:
@SaturnusDJ no, because these messages are not of type `raw`

SaturnusDJ, 2025-03-31:
@Koenkk 
Okay 😃 

The debug logs above were fast rotations.

Below are slower rotations. There most likely is debug data from other device as well. The device in question is 0xXXXXXXXXXXXXXXXX. All other addresses and names are not the button.

Right:
```
:18] info: 	z2m: Successfully changed options
:18] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/options', payload '{"data":{"restart_required":false},"status":"ok","transaction":"ph7i0-1"}'
:18] debug: 	zh:zstack:unpi:parser: <-- [254,24,68,129,0,0,6,0,29,108,1,1,0,247,0,39,96,222,0,0,4,17,60,252,0,29,108,29,125]
:18] debug: 	zh:zstack:unpi:parser: --- parseNext [254,24,68,129,0,0,6,0,29,108,1,1,0,247,0,39,96,222,0,0,4,17,60,252,0,29,108,29,125]
:18] debug: 	zh:zstack:unpi:parser: --> parsed 24 - 2 - 4 - 129 - [0,0,6,0,29,108,1,1,0,247,0,39,96,222,0,0,4,17,60,252,0,29,108,29] - 125
:18] debug: 	zh:zstack:znp: <-- AREQ: AF - incomingMsg - {"groupid":0,"clusterid":6,"srcaddr":27677,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":247,"securityuse":0,"timestamp":14573607,"transseqnumber":0,"len":4,"data":{"type":"Buffer","data":[17,60,252,0]}}
:18] debug: 	zh:controller: Received payload: clusterID=6, address=27677, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=247, frame={"header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":60,"commandIdentifier":252},"payload":{"value":0},"command":{"ID":252,"parameters":[{"name":"value","type":32}],"name":"tuyaAction2"}}
:18] debug: 	zh:zstack:unpi:parser: --- parseNext []
:18] debug: 	z2m: Received Zigbee message from '0xXXXXXXXXXXXXXXXX', type 'commandTuyaAction2', cluster 'genOnOff', data '{"value":0}' from endpoint 1 with groupID 0
:18] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xXXXXXXXXXXXXXXXX', payload '{"action":"rotate_right","battery":100,"linkquality":247,"operation_mode":"command","voltage":2900}'
:19] debug: 	zh:zstack:unpi:parser: <-- [254,23,68,129,0,0,6,0,29,108,1,1,0,247,0,168,157,222,0,0,3,17,61,3,29,108,29,249]
:19] debug: 	zh:zstack:unpi:parser: --- parseNext [254,23,68,129,0,0,6,0,29,108,1,1,0,247,0,168,157,222,0,0,3,17,61,3,29,108,29,249]
:19] debug: 	zh:zstack:unpi:parser: --> parsed 23 - 2 - 4 - 129 - [0,0,6,0,29,108,1,1,0,247,0,168,157,222,0,0,3,17,61,3,29,108,29] - 249
:19] debug: 	zh:zstack:znp: <-- AREQ: AF - incomingMsg - {"groupid":0,"clusterid":6,"srcaddr":27677,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":247,"securityuse":0,"timestamp":14589352,"transseqnumber":0,"len":3,"data":{"type":"Buffer","data":[17,61,3]}}
:19] debug: 	zh:controller: Failed to parse frame: Error: Cluster 'genOnOff' has no command '3'
:19] debug: 	zh:controller: Received payload: clusterID=6, address=27677, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=247, frame=undefined
:19] debug: 	zh:zstack:unpi:parser: --- parseNext []
:19] debug: 	z2m: Received Zigbee message from '0xXXXXXXXXXXXXXXXX', type 'raw', cluster 'genOnOff', data '{"data":[17,61,3],"type":"Buffer"}' from endpoint 1 with groupID 0
:19] debug: 	z2m: No converter available for 'ERS-10TZBVK-AA' with cluster 'genOnOff' and type 'raw' and data '{"data":[17,61,3],"type":"Buffer"}'
:19] debug: 	zh:zstack:unpi:parser: <-- [254,24,68,129,0,0,6,0,29,108,1,1,0,251,0,241,236,222,0,0,4,17,62,252,0,29,108,29,41]
:19] debug: 	zh:zstack:unpi:parser: --- parseNext [254,24,68,129,0,0,6,0,29,108,1,1,0,251,0,241,236,222,0,0,4,17,62,252,0,29,108,29,41]
:19] debug: 	zh:zstack:unpi:parser: --> parsed 24 - 2 - 4 - 129 - [0,0,6,0,29,108,1,1,0,251,0,241,236,222,0,0,4,17,62,252,0,29,108,29] - 41
:19] debug: 	zh:zstack:znp: <-- AREQ: AF - incomingMsg - {"groupid":0,"clusterid":6,"srcaddr":27677,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":251,"securityuse":0,"timestamp":14609649,"transseqnumber":0,"len":4,"data":{"type":"Buffer","data":[17,62,252,0]}}
:19] debug: 	zh:controller: Received payload: clusterID=6, address=27677, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=251, frame={"header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":62,"commandIdentifier":252},"payload":{"value":0},"command":{"ID":252,"parameters":[{"name":"value","type":32}],"name":"tuyaAction2"}}
:19] debug: 	zh:zstack:unpi:parser: --- parseNext []
:19] debug: 	z2m: Received Zigbee message from '0xXXXXXXXXXXXXXXXX', type 'commandTuyaAction2', cluster 'genOnOff', data '{"value":0}' from endpoint 1 with groupID 0
:19] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xXXXXXXXXXXXXXXXX', payload '{"action":"rotate_right","battery":100,"linkquality":251,"operation_mode":"command","voltage":2900}'
:19] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/OTHER_DEVICE_1', payload '{"battery":97,"device_temperature":23,"illuminance":56,"linkquality":138,"occupancy":false,"power_outage_count":240,"temperature":31,"voltage":2995}'
:19] debug: 	zh:zstack:unpi:parser: <-- [254,23,68,129,0,0,6,0,29,108,1,1,0,244,0,207,42,223,0,0,3,17,63,3,29,108,29,41]
:19] debug: 	zh:zstack:unpi:parser: --- parseNext [254,23,68,129,0,0,6,0,29,108,1,1,0,244,0,207,42,223,0,0,3,17,63,3,29,108,29,41]
:19] debug: 	zh:zstack:unpi:parser: --> parsed 23 - 2 - 4 - 129 - [0,0,6,0,29,108,1,1,0,244,0,207,42,223,0,0,3,17,63,3,29,108,29] - 41
[... cut ...]
[2025-03-31 12:57:39] debug: 	z2m: Received Zigbee message from '0xXXXXXXXXXXXXXXXX', type 'raw', cluster 'genOnOff', data '{"data":[17,75,4],"type":"Buffer"}' from endpoint 1 with groupID 0
[2025-03-31 12:57:39] debug: 	z2m: No converter available for 'ERS-10TZBVK-AA' with cluster 'genOnOff' and type 'raw' and data '{"data":[17,75,4],"type":"Buffer"}'
[2025-03-31 12:57:41] debug: 	z2m:mqtt: Received MQTT message on 'zigbee2mqtt/bridge/request/options' with data '{"options":{"advanced":{"log_level":"info"}},"transaction":"ph7i0-4"}'
[2025-03-31 12:57:41] info: 	z2m: Successfully changed options
```

Not sure why left is so much shorter. Let me know if you need more.

SaturnusDJ, 2025-03-31:
I just received the dark coloured version of this button I mentioned before. It looks identical to [the button](https://www.zigbee2mqtt.io/devices/ERS-10TZBVK-AA.html) that should work well in Z2M. In fact, it just might be that button.

It works exactly like the documentation, except `action_rate` might be missing.
Example:
`{"action":"brightness_step_down","action_step_size":13,"action_transition_time":0.01,"battery":100,"linkquality":160,"operation_mode":"command","voltage":3000}'`

Reconfiguring next to the coordinator:
```
:52] error: 	z2m: Failed to configure '0xBBBBBBBBBBBBBBBB', attempt 2 (Error: ZCL command 0xBBBBBBBBBBBBBBBB/1 genBasic.read([4,0,1,5,7,65534], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Timeout - 39286 - 1 - 145 - 0 - 1 after 10000ms)
    at Object.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/utils/waitress.ts:67:23)
    at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:663:47)
    at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:695:28)
    at /app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:443:20
    at Queue.execute (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/utils/queue.ts:36:20)
    at ZStackAdapter.sendZclFrameToEndpoint (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:441:16)
    at Request.send (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:82:20)
    at Endpoint.sendRequest (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:324:20)
    at Endpoint.zclCommand (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:949:28)
    at Endpoint.read (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:445:29))
:52] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/configure', payload '{"data":{},"error":"Failed to configure (ZCL command 0xBBBBBBBBBBBBBBBB/1 genBasic.read([4,0,1,5,7,65534], {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"reservedBits\":0,\"writeUndiv\":false}) failed (Timeout - 39286 - 1 - 145 - 0 - 1 after 10000ms))","status":"error","transaction":"j4cxd-2"}'
```

Interview:
```
:51] error: 	z2m: Request 'zigbee2mqtt/bridge/request/device/interview' failed with error: 'interview of '0xBBBBBBBBBBBBBBBB' (0xBBBBBBBBBBBBBBBB) failed: Error: Interview failed because can not get active endpoints ('0xBBBBBBBBBBBBBBBB')'
:51] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/interview', payload '{"data":{},"error":"interview of '0xBBBBBBBBBBBBBBBB' (0xBBBBBBBBBBBBBBBB) failed: Error: Interview failed because can not get active endpoints ('0xBBBBBBBBBBBBBBBB')","status":"error","transaction":"dp033-1"}'
```

I think it's the same story as with that BlitzWolf power socket which was also TuYa. 20 versions of the same model.

Koenkk, 2025-03-31:
@SaturnusDJ could you provide the debug log of the rotate events again after changing the `operation_mode` to `command`?

SaturnusDJ, 2025-03-31:
@Koenkk 
Is it not in command mode?
The logs in the long post says `"operation_mode":"command"`

Koenkk, 2025-03-31:
It looks like it is event, if you put it to event, do the values change? If not, can you try changing it back to command?

SaturnusDJ, 2025-03-31:
Okay, more attempts below. Called the device "BUTTON" everywhere now. Everything else is other devices.
Is there a way how I can remove unrelated debug messages? (Is `AREQ: AF - incomingMsg` maybe the start of a new debug block?)

Event mode, right:
```
:20] info: 	z2m: Successfully changed options
:20] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/options', payload '{"data":{"restart_required":false},"status":"ok","transaction":"yurwy-1"}'
:21] debug: 	zh:zstack:unpi:parser: <-- [254,32,68,129,0,0,2,7,136,121,1,1,0,40,0,54,10,163,0,0,12,8,82,10,0,0,37,135,160,1,0,0]
:21] debug: 	zh:zstack:unpi:parser: --- parseNext [254,32,68,129,0,0,2,7,136,121,1,1,0,40,0,54,10,163,0,0,12,8,82,10,0,0,37,135,160,1,0,0]
:21] debug: 	zh:zstack:unpi:parser: <-- [0,136,121,29,21]
:21] debug: 	zh:zstack:unpi:parser: --- parseNext [254,32,68,129,0,0,2,7,136,121,1,1,0,40,0,54,10,163,0,0,12,8,82,10,0,0,37,135,160,1,0,0,0,136,121,29,21]
:21] debug: 	zh:zstack:unpi:parser: --> parsed 32 - 2 - 4 - 129 - [0,0,2,7,136,121,1,1,0,40,0,54,10,163,0,0,12,8,82,10,0,0,37,135,160,1,0,0,0,136,121,29] - 21
:21] debug: 	zh:zstack:znp: <-- AREQ: AF - incomingMsg - {"groupid":0,"clusterid":1794,"srcaddr":31112,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":40,"securityuse":0,"timestamp":10684982,"transseqnumber":0,"len":12,"data":{"type":"Buffer","data":[8,82,10,0,0,37,135,160,1,0,0,0]}}
:21] debug: 	zh:controller: Received payload: clusterID=1794, address=31112, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=40, frame={"header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":82,"commandIdentifier":10},"payload":[{"attrId":0,"dataType":37,"attrData":106631}],"command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}}
:21] debug: 	zh:controller:endpoint: ZCL command 0xAAAAAAAAAAAAAAAA/1 seMetering.defaultRsp({"cmdId":10,"statusCode":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"reservedBits":0,"transactionSequenceNumber":82,"writeUndiv":false})
:21] debug: 	zh:zstack: sendZclFrameToEndpointInternal 0xAAAAAAAAAAAAAAAA:31112/1 (0,0,1)
:21] debug: 	zh:zstack:znp: --> SREQ: AF - dataRequest - {"dstaddr":31112,"destendpoint":1,"srcendpoint":1,"clusterid":1794,"transid":69,"options":0,"radius":30,"len":5,"data":{"type":"Buffer","data":[24,82,11,10,0]}}
:21] debug: 	zh:zstack:unpi:writer: --> frame [254,15,36,1,136,121,1,1,2,7,69,0,30,5,24,82,11,10,0,203]
:21] debug: 	zh:zstack:unpi:parser: --- parseNext []
:21] debug: 	z2m: Received Zigbee message from 'OTHER_SENSOR', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":106631}' from endpoint 1 with groupID 0
:21] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/OTHER_SENSOR', payload '{"current":0,"energy":1066.31,"indicator_mode":"off/on","linkquality":40,"power":0,"power_outage_memory":"restore","state":"ON","voltage":232}'
:21] debug: 	zh:zstack:unpi:parser: <-- [254,1,100,1,0,100]
:21] debug: 	zh:zstack:unpi:parser: --- parseNext [254,1,100,1,0,100]
:21] debug: 	zh:zstack:unpi:parser: --> parsed 1 - 3 - 4 - 1 - [0] - 100
:21] debug: 	zh:zstack:znp: <-- SRSP: AF - dataRequest - {"status":0}
:21] debug: 	zh:zstack:unpi:parser: --- parseNext []
:21] debug: 	zh:zstack:unpi:parser: <-- [254,3,68,128,0,1,69,131]
:21] debug: 	zh:zstack:unpi:parser: --- parseNext [254,3,68,128,0,1,69,131]
:21] debug: 	zh:zstack:unpi:parser: --> parsed 3 - 2 - 4 - 128 - [0,1,69] - 131
:21] debug: 	zh:zstack:znp: <-- AREQ: AF - dataConfirm - {"status":0,"endpoint":1,"transid":69}
:21] debug: 	zh:zstack:unpi:parser: --- parseNext []
:21] debug: 	zh:zstack:unpi:parser: <-- [254,24,68,129,0,0,6,0,29,108,1,1,0,211,0,123,50,163,0,0,4,17,83,252,0,29,108,29,69]
:21] debug: 	zh:zstack:unpi:parser: --- parseNext [254,24,68,129,0,0,6,0,29,108,1,1,0,211,0,123,50,163,0,0,4,17,83,252,0,29,108,29,69]
:21] debug: 	zh:zstack:unpi:parser: --> parsed 24 - 2 - 4 - 129 - [0,0,6,0,29,108,1,1,0,211,0,123,50,163,0,0,4,17,83,252,0,29,108,29] - 69
:21] debug: 	zh:zstack:znp: <-- AREQ: AF - incomingMsg - {"groupid":0,"clusterid":6,"srcaddr":27677,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":211,"securityuse":0,"timestamp":10695291,"transseqnumber":0,"len":4,"data":{"type":"Buffer","data":[17,83,252,0]}}
:21] debug: 	zh:controller: Received payload: clusterID=6, address=27677, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=211, frame={"header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":83,"commandIdentifier":252},"payload":{"value":0},"command":{"ID":252,"parameters":[{"name":"value","type":32}],"name":"tuyaAction2"}}
:21] debug: 	zh:zstack:unpi:parser: --- parseNext []
:21] debug: 	z2m: Received Zigbee message from 'BUTTON', type 'commandTuyaAction2', cluster 'genOnOff', data '{"value":0}' from endpoint 1 with groupID 0
:21] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/BUTTON', payload '{"action":"rotate_right","battery":100,"linkquality":211,"operation_mode":"event","voltage":2900}'
:22] debug: 	zh:zstack:unpi:parser: <-- [254,23,68,129,0,0,6,0,29,108,1,1,0,189,0,51,112,163,0,0,3,17,84,3,29,108,29,209]
[... cut ...]
:35] debug: 	z2m: Received Zigbee message from 'OTHER_SENSOR', type 'readResponse', cluster 'haElectricalMeasurement', data '{"activePower":60,"rmsCurrent":387,"rmsVoltage":230}' from endpoint 1 with groupID 0
:35] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/OTHER_SENSOR', payload '{"current":0.39,"energy":262.28,"indicator_mode":"off/on","linkquality":58,"power":60,"power_outage_memory":"restore","state":"ON","voltage":230}'
:36] debug: 	z2m:mqtt: Received MQTT message on 'zigbee2mqtt/bridge/request/options' with data '{"options":{"advanced":{"log_level":"info"}},"transaction":"yurwy-8"}'
:36] info: 	z2m: Successfully changed options
```

JiefHA, 2025-04-01:
I had the white button. It doesn't send command events. I am almost sure now that they just copy the Moes button and did not implement all the features.

Koenkk, 2025-04-03:
The logging shows that the command mode doesn't work indeed, I updated the definition.

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.

SaturnusDJ, 2025-04-03:
@Koenkk 
Thanks! So that is it for the white button I guess.

Anything needed to fix/improve the dark coloured button? https://github.com/Koenkk/zigbee2mqtt/issues/25053#issuecomment-2766063264

Koenkk, 2025-04-04:
@SaturnusDJ for the dark one I see the interview fails, can you try re-pairing it close to the coordinator?

SaturnusDJ, 2025-04-04:
@Koenkk This happened next to the coordinator.

vit5041, 2025-04-04:
sorry guys, I don't get, is it possible to do something with that only every third rotation is reported? One click rotation is not visible in logs. Current behavior that command/event mode swap is removed.. All the rest functioning as before.. Hm.

SaturnusDJ, 2025-04-05:
> sorry guys, I don't get, is it possible to do something with that only every third rotation is reported? One click rotation is not visible in logs. Current behavior that command/event mode swap is removed.. All the rest functioning as before.. Hm.

No. The white button only sends out data at every third rotation. The dark button sends out data every rotation. Don't take this as 100% guarantee though, it would not be surprising if there are even more variants.

Koenkk, 2025-04-06:
@SaturnusDJ did you also try pairing it while permitting joining only via the coordinator?

SaturnusDJ, 2025-04-06:
@Koenkk 
```
[2025-04-06 21:35:03] info: 	z2m: Zigbee: allowing new devices to join via Coordinator.
[2025-04-06 21:35:03] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"device":"Coordinator","time":254},"status":"ok","transaction":"5gv9g-4"}'
[2025-04-06 21:35:25] info: 	zh:controller: Interview for '0xXXXXXXXXXXXXXXXX' started
[2025-04-06 21:35:25] info: 	z2m: Device '0xXXXXXXXXXXXXXXXX' joined
[2025-04-06 21:35:25] info: 	z2m: Starting interview of '0xXXXXXXXXXXXXXXXX'
[2025-04-06 21:35:25] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xXXXXXXXXXXXXXXXX","ieee_address":"0xXXXXXXXXXXXXXXXX"},"type":"device_joined"}'
[2025-04-06 21:35:25] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xXXXXXXXXXXXXXXXX","ieee_address":"0xXXXXXXXXXXXXXXXX","status":"started"},"type":"device_interview"}'
[2025-04-06 21:35:25] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xXXXXXXXXXXXXXXXX","ieee_address":"0xXXXXXXXXXXXXXXXX"},"type":"device_announce"}'
[2025-04-06 21:35:28] info: 	zh:controller: Succesfully interviewed '0xXXXXXXXXXXXXXXXX'
[2025-04-06 21:35:28] info: 	z2m: Successfully interviewed '0xXXXXXXXXXXXXXXXX', device has successfully been paired
[2025-04-06 21:35:28] info: 	z2m: Device '0xXXXXXXXXXXXXXXXX' is supported, identified as: Tuya Smart knob (ERS-10TZBVK-AA)
[2025-04-06 21:35:28] info: 	z2m: Configuring '0xXXXXXXXXXXXXXXXX'
[2025-04-06 21:35:28] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Smart knob","exposes":[{"access":1,"category":"diagnostic","description":"Triggered action (e.g. a button click)","label":"Action","name":"action","property":"action","type":"enum","values":["toggle","brightness_step_up","brightness_step_down","color_temperature_step_up","color_temperature_step_down","saturation_move","hue_move","hue_stop","single","double","hold","rotate_left","rotate_right"]},{"access":1,"label":"Action step size","name":"action_step_size","property":"action_step_size","type":"numeric","value_max":255,"value_min":0},{"access":1,"label":"Action transition time","name":"action_transition_time","property":"action_transition_time","type":"numeric","unit":"s"},{"access":1,"label":"Action rate","name":"action_rate","property":"action_rate","type":"numeric","value_max":255,"value_min":0},{"access":1,"category":"diagnostic","description":"Remaining battery in %, can take up to 24 hours before reported","label":"Battery","name":"battery","property":"battery","type":"numeric","unit":"%","value_max":100,"value_min":0},{"access":7,"description":"Operation mode: \"command\" - for group control, \"event\" - for clicks","label":"Operation mode","name":"operation_mode","property":"operation_mode","type":"enum","values":["command","event"]},{"access":1,"category":"diagnostic","description":"Link quality (signal strength)","label":"Linkquality","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"ERS-10TZBVK-AA","options":[{"access":2,"description":"Simulate a brightness value. If this device provides a brightness_move_up or brightness_move_down action it is possible to specify the update interval and delta. The action_brightness_delta indicates the delta for each interval.","features":[{"access":2,"description":"Delta per interval, 20 by default","label":"Delta","name":"delta","property":"delta","type":"numeric","value_min":0},{"access":2,"description":"Interval duration","label":"Interval","name":"interval","property":"interval","type":"numeric","unit":"ms","value_min":0}],"label":"Simulated brightness","name":"simulated_brightness","property":"simulated_brightness","type":"composite"}],"supports_ota":false,"vendor":"Tuya"},"friendly_name":"0xXXXXXXXXXXXXXXXX","ieee_address":"0xXXXXXXXXXXXXXXXX","status":"successful","supported":true},"type":"device_interview"}'
[2025-04-06 21:35:29] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xXXXXXXXXXXXXXXXX', payload '{"battery":100,"linkquality":255,"operation_mode":"command","voltage":3000}'
[2025-04-06 21:35:30] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xXXXXXXXXXXXXXXXX', payload '{"battery":100,"linkquality":255,"operation_mode":"event","voltage":3000}'
[2025-04-06 21:35:30] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xXXXXXXXXXXXXXXXX', payload '{"battery":100,"linkquality":255,"operation_mode":"event","voltage":3000}'
[2025-04-06 21:35:31] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xXXXXXXXXXXXXXXXX', payload '{"battery":100,"linkquality":255,"operation_mode":"event","voltage":3000}'
[2025-04-06 21:35:32] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xXXXXXXXXXXXXXXXX', payload '{"battery":100,"linkquality":255,"operation_mode":"event","voltage":3000}'
[2025-04-06 21:35:35] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xXXXXXXXXXXXXXXXX', payload '{"battery":100,"linkquality":255,"operation_mode":"event","voltage":3000}'
[2025-04-06 21:35:35] info: 	z2m: Successfully configured '0xXXXXXXXXXXXXXXXX'
```
In that case, no errors. 

1. What does this imply? Joining new devices should happen via the coordinator?
2. Afterwards, doing reconfigure or interview via webinterface results in the errors again. Does this have a meaning?
3. `action_rate` (https://www.zigbee2mqtt.io/devices/ERS-10TZBVK-AA.html) is still not in the messages. I guess the device does not have it.

Koenkk, 2025-04-07:
1. Some devices can be very picky during pairing, pairing directly via the coordinator reduces the point of failures
2. Can you provide the debug log again?
3. Can you provide the debug log in `command` mode?

Koenkk, 2025-04-08:
1. Yes
2. You can use [this form](https://form.jotform.com/233385867866072) to share the log with me
3. Looks that this one sends the `action_step_size` correctly (these are the native ZCL commands so it works differently than the white one)

SaturnusDJ, 2025-04-08:
2. Done. `0xBUTTON-DARK` is the one. I made the device leave the network. Then paired with coordinator and left the device next to it. Then did reconfigure, followed by interview.
3. Yes I think this dark one is good, except I have not seen any `action_rate`, but upon reading the [manual](https://www.zigbee2mqtt.io/devices/ERS-10TZBVK-AA.html) again...
> Action_Rate = Keeps "No value" all the time

...maybe that is expected.

gymnae, 2025-04-09:
Hi there, since the last update this smart knob is now correctly recognized, but stopped working with ControllerX, which I used to have gradual changes while turning the knob pressed and de-pressed. This might also be a quirk with ControllerX, because it only supports ERS-10TZBVK-AA. 
I removed and re-paired the knob, but also don't get the command mode to work anymore.
Next up I try to ditch ControllerX and use an automation directly. Fewer things to break.

Koenkk, 2025-04-09:
> Done. 0xBUTTON-DARK is the one. I made the device leave the network. Then paired with coordinator and left the device next to it. Then did reconfigure, followed by interview.

I forgot to mention, can you:
- Force remove the device through the frontend
- Restart z2m (this is what I forgot to mention)
- Pair the device with only joining enabled on the coordinator -> provide the debug log of this

SaturnusDJ, 2025-04-10:
Done, files sent as before.

Ahh indeed. Seen `action_rate` when pushing plus holding the button. 👍🏼

Koenkk, 2025-04-10:
@SaturnusDJ looks that all is OK now!

```
[2025-04-10 13:37:49] info: 	zh:controller: Succesfully interviewed '0xBUTTON-DARK'
...
[2025-04-10 13:37:49] info: 	z2m: Successfully interviewed '0xBUTTON-DARK', device has successfully been paired
...
[2025-04-10 13:37:53] info: 	z2m: Successfully configured '0xBUTTON-DARK'
```

SaturnusDJ, 2025-04-11:
@Koenkk 
Good. I think/assume that the times it failed on re-interview and re-configure (the buttons in the webinterface) was because it either joined on a router before, or it had already hopped from the coordinator to a router.

Should be all okay indeed. Thanks for the help!!! 👌🏼

skdw, 2025-04-11:
Hello! After updating zigbee2mqtt to [v2.2.1](https://github.com/Koenkk/zigbee2mqtt/releases/tag/2.2.1), my two older type [ERS-10TZBVK-AA](https://www.zigbee2mqtt.io/devices/ERS-10TZBVK-AA.html) smart knobs have just stopped working. My deivces are mistakenly recognized as [ZG-101Z/D](https://www.zigbee2mqtt.io/images/devices/ZG-101Z-D.png). Knob action messages are gone now, including `[command, event]` operation mode switch.

I am able to force remove and re-join the device. 
```
info 2025-04-11 22:12:47z2m: Device '0xa4c13828e63ce1bd' is supported, identified as: Tuya Smart knob (ZG-101Z/D)
info 2025-04-11 22:12:47z2m: Configuring '0xa4c13828e63ce1bd'
info 2025-04-11 22:12:47z2m:mqtt: MQTT publish: topic 'homeassistant/sensor/0xa4c13828e63ce1bd/battery/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"hw_version":1,"identifiers":["zigbee2mqtt_0xa4c13828e63ce1bd"],"manufacturer":"Tuya","model":"Smart knob","model_id":"ZG-101Z/D","name":"0xa4c13828e63ce1bd","via_device":"zigbee2mqtt_bridge_0x00124b0029e728f5"},"device_class":"battery","enabled_by_default":true,"entity_category":"diagnostic","object_id":"0xa4c13828e63ce1bd_battery","origin":{"name":"Zigbee2MQTT","sw":"2.2.1","url":"https://www.zigbee2mqtt.io"},"state_class":"measurement","state_topic":"zigbee2mqtt/0xa4c13828e63ce1bd","unique_id":"0xa4c13828e63ce1bd_battery_zigbee2mqtt","unit_of_measurement":"%","value_template":"{{ value_json.battery }}"}'
info 2025-04-11 22:12:47z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Smart knob","exposes":[{"access":5,"category":"diagnostic","description":"Remaining battery in %","label":"Battery","name":"battery","property":"battery","type":"numeric","unit":"%","value_max":100,"value_min":0},{"access":1,"category":"diagnostic","description":"Triggered action (e.g. a button click)","label":"Action","name":"action","property":"action","type":"enum","values":["rotate_left","rotate_right"]},{"access":1,"category":"diagnostic","description":"Link quality (signal strength)","label":"Linkquality","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"ZG-101Z/D","options":[],"supports_ota":false,"vendor":"Tuya"},"friendly_name":"0xa4c13828e63ce1bd","ieee_address":"0xa4c13828e63ce1bd","status":"successful","supported":true},"type":"device_interview"}'
info 2025-04-11 22:12:47z2m:mqtt: MQTT publish: topic 'homeassistant/sensor/0xa4c13828e63ce1bd/linkquality/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"hw_version":1,"identifiers":["zigbee2mqtt_0xa4c13828e63ce1bd"],"manufacturer":"Tuya","model":"Smart knob","model_id":"ZG-101Z/D","name":"0xa4c13828e63ce1bd","via_device":"zigbee2mqtt_bridge_0x00124b0029e728f5"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:signal","name":"Linkquality","object_id":"0xa4c13828e63ce1bd_linkquality","origin":{"name":"Zigbee2MQTT","sw":"2.2.1","url":"https://www.zigbee2mqtt.io"},"state_class":"measurement","state_topic":"zigbee2mqtt/0xa4c13828e63ce1bd","unique_id":"0xa4c13828e63ce1bd_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
info 2025-04-11 22:12:48z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xa4c13828e63ce1bd', payload '{"battery":82,"linkquality":145}'
info 2025-04-11 22:12:48z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xa4c13828e63ce1bd', payload '{"battery":82,"linkquality":149}'
```

Re-interviewing fails with an error:
```
error 2025-04-11 22:03:11z2m: Request 'zigbee2mqtt/bridge/request/device/interview' failed with error: 'interview of '0xa4c13828e63ce1bd' (0xa4c13828e63ce1bd) failed: Error: Interview failed because can not get active endpoints ('0xa4c13828e63ce1bd')'
info 2025-04-11 22:03:11z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/interview', payload '{"data":{},"error":"interview of '0xa4c13828e63ce1bd' (0xa4c13828e63ce1bd) failed: Error: Interview failed because can not get active endpoints ('0xa4c13828e63ce1bd')","status":"error","transaction":"mwe2c-2"}'
```

Re-configuration also ends up a failure:
```
error 2025-04-11 22:00:11z2m: Failed to configure '0xa4c13828e63ce1bd', attempt 2 (Error: Bind 0xa4c13828e63ce1bd/1 genPowerCfg from '0x00124b0029e728f5/1' failed (AREQ - ZDO - bindRsp after 10000ms) at Object.start (/app/node_modules/.pnpm/zigbee-herdsman@3.4.11/node_modules/zigbee-herdsman/src/utils/waitress.ts:67:23) at func (/app/node_modules/.pnpm/zigbee-herdsman@3.4.11/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:423:47) at Queue.execute (/app/node_modules/.pnpm/zigbee-herdsman@3.4.11/node_modules/zigbee-herdsman/src/utils/queue.ts:36:20) at ZStackAdapter.sendZdoInternal (/app/node_modules/.pnpm/zigbee-herdsman@3.4.11/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:428:43) at ZStackAdapter.sendZdo (/app/node_modules/.pnpm/zigbee-herdsman@3.4.11/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:305:16) at Endpoint.bind (/app/node_modules/.pnpm/zigbee-herdsman@3.4.11/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:571:30) at setupAttributes (/app/node_modules/.pnpm/zigbee-herdsman-converters@23.20.1/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:158:13) at configure (/app/node_modules/. [... cut ...] re.configure (/app/lib/extension/configure.ts:124:13))
info 2025-04-11 22:00:11z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/configure', payload '{"data":{},"error":"Failed to configure (Bind 0xa4c13828e63ce1bd/1 genPowerCfg from '0x00124b0029e728f5/1' failed (AREQ - ZDO - bindRsp after 10000ms))","status":"error","transaction":"mwe2c-1"}'
```
I will be grateful for help to recognize my devices again as [ERS-10TZBVK-AA](https://www.zigbee2mqtt.io/devices/ERS-10TZBVK-AA.html).

Koenkk, 2025-04-12:
@skdw could you provide the data/database.db entry of your device?

skdw, 2025-04-12:
@Koenkk sure, there it is: [database.db.txt](https://github.com/user-attachments/files/19719606/database.db.txt)

SaturnusDJ, 2025-04-15:
@Koenkk 
The white one is going crazy now here. Loads and loads of "device announce" bridge events. Already did force remove, Z2M restart and rejoin on coordinator.

Koenkk, 2025-04-16:
@skdw @renatoccosta fixed!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.

@SaturnusDJ `deviceAnnounce` means the device restarts, for battery powered device it can occur if the battery is getting empty.

kbullet, 2025-04-21:
@Koenkk just saw this thread here after creating my PR. Though I'm not entirely sure if _TZ3000_abrsvsou should be detected as ERS-10TZBVK-AA since it is a white knob... 

i've created a PR here to add operation mode for command/event ZG-101Z/D wdyt?

https://github.com/Koenkk/zigbee-herdsman-converters/pull/9217

renatoccosta, 2025-04-21:
@kbullet, the knob I have is this one: https://www.aliexpress.com/item/1005006765315209.html. But it seems to be different from yours. It has the hold and rotate left/right action.

kbullet, 2025-04-22:
> [@kbullet](https://github.com/kbullet), the knob I have is this one: https://www.aliexpress.com/item/1005006765315209.html. But it seems to be different from yours. It has the hold and rotate left/right action.

Ah yes just had another glance your version should be identified as ERS-10TZBVK-AA as per @Koenkk earlier fix. For those using the above version (_TZ3000_abrsvsou, applicationVersion: 145) will be identified as ZG-101Z/D which now has command/event modes added back.

JensvanZutphen, 2025-05-02:
is this knob's functionality fixed now in 2.3.0?

gymnae, 2025-05-02:
> > [@kbullet](https://github.com/kbullet), the knob I have is this one: https://www.aliexpress.com/item/1005006765315209.html. But it seems to be different from yours. It has the hold and rotate left/right action.
> 
> Ah yes just had another glance your version should be identified as ERS-10TZBVK-AA (appVersion 66) as per [@Koenkk](https://github.com/Koenkk) earlier fix. For those using the above version (_TZ3000_abrsvsou, applicationVersion: 145) will be identified as ZG-101Z/D which now has command/event modes added back.
> 
> edit: for those with knob above (appVersion 145) you can try to use my blueprint [here](https://my.home-assistant.io/redirect/blueprint_import/?blueprint_url=https%3A%2F%2Fraw.githubusercontent.com%2Fkbullet%2FHomeAssistant%2Frefs%2Fheads%2Fmain%2Fblueprints%2FZG-101ZD.yaml) to use with the knob once the latest fix is released.

I have the same product, _TZ3000_abrsvsou, and as long as it was recognized as ERS-10TZBVK-AA, I was able to have press and hold rotate actions via zigbee2mqtt and ControllerX. But ControllerX doesn't support ZG-101Z/D.

Since it's my smart knobs are recognized as ZG-101Z/D, your blueprint helped getting back the basic functionality, but they're lacking press and hold actions, which were nice to change colors etc.

dottedmag, 2025-05-03:
Hey folks. I've got the white knob, and captured communication between the knob and Tuya gateway.

I can confirm that after it is connected to Tuya it sends up/down events on every click (not on every third), and it also sends events on rotation while pressed.

I can attach wireshark capture if that's useful.

SaturnusDJ, 2025-05-03:
> [@SaturnusDJ](https://github.com/SaturnusDJ) `deviceAnnounce` means the device restarts, for battery powered device it can occur if the battery is getting empty.
After reinstalling (the same) battery two times it is more stable now.

Confirming that in 2.3.0 the white one utilizes both command and event mode again. (The dark one is still superior.)

[3 more replies]

closed 2025-08-27 by github-actions[bot]