Koenkk/zigbee2mqtt #25057: Aqara opple switches (WXCJKG13LM) does not publish button_x_hold events

Zolli, 2024-12-04:
### What happened?

I tested the newly introduced home assistant event entity support and everything works expected, except this one, this is a 6 button switch and support single, double, triple, release and hold for all of its buttons. But sadly the hold action not defined for all of its buttons just a common one called `hold` while `button_X_release` is exist, and in the past, when the action sensors still available this events get pusblished as `button_X_hold`.

### What did you expect to happen?

`button_X_hold` should be available for all of the switch buttons

### How to reproduce it (minimal and precise)

Pair a device and hold a button for 2 seconds, and you will see a simpele `hold` event got published.

### Zigbee2MQTT version

1.42.0

### Adapter firmware version

20240710

### Adapter

LAUNCHXL-CC1352-P2

### Setup

Standalone docker container

mrin, 2024-12-17:
I have the same issue. Trying switch to use events with disabled legacy options.

Zolli, 2024-12-17:
I somewhat solved to this myself in my NR flows and combine the 2 property where I need, so it's not a high priority bug for me at the moment.

chernyshaw, 2025-02-16:
Same for my "Opple wireless switch (triple band) (WXCJKG13LM)"
In MQTT Explorer I can see: 
```json
"action": "button_1_hold",
"action": "button_1_release"
```
But in Home Assitant I can see 
```
hold
button_1_release
```
So it's `hold` for any button.

nununo, 2025-04-12:
I created a PR on this Blueprint to effectively allow for `button_X_hold` taking into consideration the provided `button_type` attribute: https://github.com/remb0/blueprints/pull/7

closed 2025-06-20 by github-actions[bot]

brunoabele, 2025-08-09:
Found this with the same problem, and wanted to give details on the workaround:

I have the same OPPLE switch, and I react on the events: First for a single click, then for a hold (checking both hold and the button). 
I also switched the OPPLE to event type operations in Z2M, not sure if that is required.

```
triggers:
  - trigger: state
    entity_id:
      - event.0x54efff10004c19d9_action
conditions: []
actions:
  - choose:
      - conditions:
          - condition: state
            entity_id: event.0x54efff10004c19d9_action
            attribute: event_type
            state: button_1_single
        sequence: []
      - conditions:
          - condition: state
            entity_id: event.0x54efff10004c19d9_action
            attribute: event_type
            state: hold
          - condition: state
            entity_id: event.0x54efff10004c19d9_action
            attribute: button
            state: button_1
        sequence []

```

Hope this helps!

Btw: In the log and in device actions, button_1_hold is logged and available, but not in events.