Koenkk/zigbee2mqtt #25060: [Wrong device]: Danfoss Ally 014G2480 detected as  Tuya Model TS0201

Cubixzno1, 2024-12-04:
### Link

https://www.danfoss.com/da-dk/about-danfoss/news/dhs/new-room-sensor-for-ally/

### Model

014G2480

### Description

Room Sensor

### Vendor

Danfoss

### Picture (link)

https://www.danfoss.com/media/d1dlu332/1120x747-danfoss-ally-roomsensor.jpg?anchor=center&mode=crop&width=520

### Database entry

{"id":65,"type":"EndDevice","ieeeAddr":"0xa4c13849ae478738","nwkAddr":27314,"manufId":4417,"manufName":"_TZ3000_mxzo5rhf","powerSource":"Battery","modelId":"TS0201","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":770,"inClusterList":[1,3,1026,1029,0],"outClusterList":[3,25,10],"clusters":{"genBasic":{"attributes":{"65503":"\u0006\u0000\u0000\u0000\u0012\t\u0000\u0000\u0000\u0012","65506":51,"65508":1,"65534":0,"modelId":"TS0201","manufacturerName":"_TZ3000_mxzo5rhf","powerSource":3,"zclVersion":3,"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200,"batteryVoltage":30}},"msTemperatureMeasurement":{"attributes":{"measuredValue":2081}},"msRelativeHumidity":{"attributes":{"measuredValue":4455}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":217189513},"lastSeen":1733344412817}

### Notes

Everything seems to work and it integrates easily to HA

Cubixzno1, 2024-12-05:
Yes i know. I just thought it would be nice to see the "correct info" in z2m. :) If i'm wrong just close it.

chris-1243, 2024-12-05:
It is not possible to do the difference as the detection is based on the modelID. I let you close your issue unless you have other question.

You may create your own external converter with the information required (brands, model, etc).

closed 2025-02-11 by github-actions[bot]