Koenkk/zigbee2mqtt #25062: Zigbee2MQTT 2.0 is not starting with a TCP coordinator

rohankapoorcom, 2024-12-04:
### What happened?

I have a docker install of Zigbee2MQTT - I switched from `image: koenkk/zigbee2mqtt:1.42.0` to `image: koenkk/zigbee2mqtt:latest-dev` which should include all of the 2.0 breaking changes. I was not using any of the legacy features.

Migration file shows the following:

```
[ADDITION] Migrated settings to version 2

[REMOVAL] Removed configuring group members through configuration.yaml (groups.xyz.devices setting). This will not impact current group members; however, you will no longer be able to add or remove devices from a group through the configuration.yaml.

For more details, see https://github.com/Koenkk/zigbee2mqtt/discussions/24198
```
My configuration.yaml includes the following for the serial port section:

```
serial:
  port: tcp://zigbee-radio.REDACTED_DOMAIN.com:6638
  adapter: zstack
 ```
 
 Zigbee2MQTT is unable to start with the following repeating in the logs:
 
 ```
 Using '/app/data' as data directory

Starting Zigbee2MQTT without watchdog.

[2024-12-04 12:58:20] info: 	z2m: Logging to console, file (filename: log.log)

[2024-12-04 12:58:20] info: 	z2m: Starting Zigbee2MQTT version 1.42.0-dev (commit #53208ae06f22e4bbe42762ea8268224d9f355034)

[2024-12-04 12:58:20] info: 	z2m: Starting zigbee-herdsman (3.0.0)

[2024-12-04 12:58:20] error: 	z2m: Error while starting zigbee-herdsman

[2024-12-04 12:58:20] error: 	z2m: Failed to start zigbee-herdsman

[2024-12-04 12:58:20] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions

[2024-12-04 12:58:20] error: 	z2m: Exiting...

[2024-12-04 12:58:20] error: 	z2m: Error: Invalid TCP path, expected format: tcp://<host>:<port>

    at findTCPAdapter (/app/node_modules/.pnpm/zigbee-herdsman@3.0.0/node_modules/zigbee-herdsman/src/adapter/adapterDiscovery.ts:466:15)

    at discoverAdapter (/app/node_modules/.pnpm/zigbee-herdsman@3.0.0/node_modules/zigbee-herdsman/src/adapter/adapterDiscovery.ts:496:26)

    at Function.create (/app/node_modules/.pnpm/zigbee-herdsman@3.0.0/node_modules/zigbee-herdsman/src/adapter/adapter.ts:68:54)

    at Controller.start (/app/node_modules/.pnpm/zigbee-herdsman@3.0.0/node_modules/zigbee-herdsman/src/controller/controller.ts:133:24)

    at Zigbee.start (/app/lib/zigbee.ts:69:27)

    at Controller.start (/app/lib/controller.ts:139:13)

    at start (/app/index.js:161:5)

```

I restored my backup of the data folder and switched back to `image: koenkk/zigbee2mqtt:1.42.0` and Zigbee2MQTT started up again.

### What did you expect to happen?

I expected Zigbee2MQTT to start up as it did before upgrading to the development image and connect to my TCP coordinator

### Zigbee2MQTT version

1.42.0-dev

### Adapter firmware version

20240710

### Adapter

TubesZB CC2652 P7 Based Zigbee to PoE Coordinator 2023

### Setup

Docker on x86-64

### Debug log

 ```
 Using '/app/data' as data directory

Starting Zigbee2MQTT without watchdog.

[2024-12-04 12:58:20] info: 	z2m: Logging to console, file (filename: log.log)

[2024-12-04 12:58:20] info: 	z2m: Starting Zigbee2MQTT version 1.42.0-dev (commit #53208ae06f22e4bbe42762ea8268224d9f355034)

[2024-12-04 12:58:20] info: 	z2m: Starting zigbee-herdsman (3.0.0)

[2024-12-04 12:58:20] error: 	z2m: Error while starting zigbee-herdsman

[2024-12-04 12:58:20] error: 	z2m: Failed to start zigbee-herdsman

[2024-12-04 12:58:20] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions

[2024-12-04 12:58:20] error: 	z2m: Exiting...

[2024-12-04 12:58:20] error: 	z2m: Error: Invalid TCP path, expected format: tcp://<host>:<port>

    at findTCPAdapter (/app/node_modules/.pnpm/zigbee-herdsman@3.0.0/node_modules/zigbee-herdsman/src/adapter/adapterDiscovery.ts:466:15)

    at discoverAdapter (/app/node_modules/.pnpm/zigbee-herdsman@3.0.0/node_modules/zigbee-herdsman/src/adapter/adapterDiscovery.ts:496:26)

    at Function.create (/app/node_modules/.pnpm/zigbee-herdsman@3.0.0/node_modules/zigbee-herdsman/src/adapter/adapter.ts:68:54)

    at Controller.start (/app/node_modules/.pnpm/zigbee-herdsman@3.0.0/node_modules/zigbee-herdsman/src/controller/controller.ts:133:24)

    at Zigbee.start (/app/lib/zigbee.ts:69:27)

    at Controller.start (/app/lib/controller.ts:139:13)

    at start (/app/index.js:161:5)

```

lux73, 2024-12-04:
try using plain IP Adress

Koenkk, 2024-12-05:
Fixed!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now. Note that the dev branch currently contains **breaking changes**, see https://github.com/Koenkk/zigbee2mqtt/discussions/24198

closed 2024-12-05 by Koenkk

rohankapoorcom, 2024-12-05:
Thanks @Koenkk! Confirmed that it's working.